MichaIng/DietPi #5737: Build image

symbios24, 2022-09-09:
Hello there i have a Orange Pi One and I want to build an updated Image for my sbc are there any instructions anyware?

Thank you

Joulinar, 2022-09-10:
Maybe following could help to build your own image https://dietpi.com/docs/hardware/#make-your-own-distribution

symbios24, 2022-09-10:
Thanks a lot 😊

symbios24, 2022-09-10:
Sorry one last thing

If i only want to build the wifi hotspot from dietpi in a clean debian image instead the whole dietpi ,how can i do that?

Thanks

Joulinar, 2022-09-10:
Theoretically you could try to follow the install steps we do to install HotSpot https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L8381

symbios24, 2022-09-14:
I managed to build an image successfully for the orange pi one
The only issue is that when i put the root/dietpi password itt takes me for a glimpse to the main after-login screen and it goes back again to login screen asking the username/password again this can go on forever,,only the dietpi/dietpi login works correct but i don't have root access like this
Is there a fix for the root login issue ??

I ussed the latest Bullseye Image from Armbian

Armbian_22.08.1_Orangepione_bullseye_current_5.15.63

Thanks

MichaIng, 2022-09-18:
I guess this can be closed, with the login issue fixed in the other issue? Feel free to reopen if required.

closed 2022-09-18 by MichaIng