MichaIng/DietPi #5739: Error while Owncloud-Installation

DrMoshSince1988, 2022-09-11:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bullseye 0~
- Kernel version | Linux homeserver 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
- SBC model | RPi4
- Power supply used | 3A
- SD card used | (SanDisk ultra 200GB

#### Additional Information (if applicable)
- Software title | Owncloud
- Was the software title installed freshly or updated/migrated?  fresh
- Can this issue be replicated on a fresh installation of DietPi? yes
  6b46220e-aee8-4679-827b-c48cdf58d390

#### Steps to reproduce

1. start Owncloud installation
2. ...

#### Expected behaviour
  installation without errors
- ...

#### Actual behaviour
 
- Command: tar xf owncloud-latest.tar.bz2 --one-top-level=/var/www                                                  │ 
                                   │  - Exit code: 2                                                                                                      │ 
                                   │  - DietPi version: v8.8.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                   │ 
                                   │  - Image creator: DietPi Core Team                                                                                   │ 
                                   │  - Pre-image: from scratch                                                                                           │ 
                                   │  - Error log:                                                                                                        │ 
                                   │ bzip2: (stdin) is not a bzip2 file.                                                                                  │ 
                                   │ tar: Child returned status 2                                                                                         │ 
                                   │ tar: Error is not recoverable: exiting now  
- ...

#### Extra details

Joulinar, 2022-09-11:
Looks like something chnaged on OwnCloud side and the directory we try to download from did not exist anymore. As workaround you could try following

1. start install again and wait for the error to happen
2. just leave the install as is, don't exit or do something else on this screen
3. open another SSH session and do following

```
cd /tmp/DietPi-Software/
rm owncloud-latest.tar.bz2
curl -sSfL https://download.owncloud.com/server/stable/owncloud-latest.tar.bz2 -o owncloud-latest.tar.bz2
```
4. go back to first SSH session and hit retry

MichaIng, 2022-09-12:
One can also do the steps from the error handler menu, choosng "Open subshell". Thanks for reporting 🙂.

Joulinar, 2022-09-12:
Ahh you are right, forget about it 🤣

MichaIng, 2022-09-15:
Fix merged for upcoming DietPi v8.9: #5740

closed 2022-09-15 by MichaIng