MichaIng/DietPi #5742: Let's encrypt not auto renewing

eikaramba, 2022-09-12:
### Creating a bug report/issue

#### Required Information
- DietPi version: **8.7.1**
- Distro version **bullseye**
- Kernel version **Linux HC1Server 4.14.241+ SMP PREEMPT Wed Jul 28 16:55:16 UTC 2021 armv7l GNU/Linux**
- SBC model **Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)**
- Power supply used **official raspberry power supply**
- SD card used **sandisk ultra 32gb**

#### Additional Information (if applicable)
- Software title **Let's Encrypt**

#### Steps to reproduce
I am using **Lighttpd** and noticed that auto renewal for let's encrypt does not work correctly. The certification is not renewed automatically. However it seems that the actual renewal is taking place and a new certificat is fetched but not installed.

because if i start dietpi-letsencrypt and just click "ok" to fetch a new certificate and then option 1 (because the certification is not scheduled for renewal) afterwards everything is working and the new certificate is in place.

Apart from that let's encrypt works exactly like it should. it's really just the auto renewal which is not happening automatically.

```
DietPi-LetsEncrypt
─────────────────────────────────────────────────────
 Mode: Running Certbot

[  OK  ] DietPi-LetsEncrypt | Lighttpd webserver detected
[  OK  ] DietPi-LetsEncrypt | systemctl start lighttpd
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Cert not yet due for renewal

You have an existing certificate that has exactly the same domains or certificate name you requested and isn't close to expiry.
(ref: /etc/letsencrypt/renewal/xxxxxx.conf)

What would you like to do?
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
1: Keep the existing certificate for now
2: Renew & replace the certificate (may be subject to CA rate limits)
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Select the appropriate number [1-2] then [enter] (press 'c' to cancel): 1
Keeping the existing certificate

- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
Certificate not yet due for renewal; no action taken.
- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
[  OK  ] DietPi-LetsEncrypt | Desired setting in /etc/lighttpd/lighttpd.conf was already set:   "mod_setenv",
[ INFO ] DietPi-LetsEncrypt | Checking for required APT packages: lighttpd-mod-openssl
[  OK  ] DietPi-LetsEncrypt | sed -i s/, "Options" => "-SessionTicket"// /etc/lighttpd/conf-available/50-dietpi-https.conf
[  OK  ] DietPi-LetsEncrypt | Desired setting in /etc/lighttpd/lighttpd.conf was already set:   "mod_redirect",
[ INFO ] DietPi-LetsEncrypt | Applying HTTPS domain to known web application configs
[  OK  ] DietPi-LetsEncrypt | sed -i s|'http://localhost/nextcloud'|'https://xxxxxx/nextcloud'| /var/www/nextcloud/config/config.php
[  OK  ] DietPi-LetsEncrypt | systemctl restart lighttpd
[  OK  ] DietPi-LetsEncrypt | eval echo -e '#!/bin/dash\nsystemctl reload lighttpd' > /etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh
[  OK  ] DietPi-LetsEncrypt | chmod +x /etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh

Press any key to return to the DietPi-LetsEncrypt menu ...
```

#### Expected behaviour
automatic renewal and installation of new certificate

#### Actual behaviour
not using new certification

#### Extra details

- **systemctl status certbot**

```
● certbot.service - Certbot
     Loaded: loaded (/lib/systemd/system/certbot.service; static)
     Active: inactive (dead) since Mon 2022-09-12 02:32:54 CEST; 12h ago
TriggeredBy: ● certbot.timer
       Docs: file:///usr/share/doc/python-certbot-doc/html/index.html
             https://certbot.eff.org/docs
    Process: 10695 ExecStart=/usr/bin/certbot -q renew (code=exited, status=0/SUCCESS)
   Main PID: 10695 (code=exited, status=0/SUCCESS)

Sep 12 02:32:52 HC1Server systemd[1]: Starting Certbot...
Sep 12 02:32:54 HC1Server systemd[1]: certbot.service: Succeeded.
Sep 12 02:32:54 HC1Server systemd[1]: Finished Certbot.
```

- **/var/log/letsencrypt/letsencrypt.log** is actually empty (strange)

- **/etc/letsencrypt/renewal/xxxxxxxx.conf**

```
# renew_before_expiry = 30 days
version = 1.12.0
archive_dir = /etc/letsencrypt/archive/xxxxxxxxxxxx
cert = /etc/letsencrypt/live/xxxxxxxxxxxx/cert.pem
privkey = /etc/letsencrypt/live/xxxxxxxxxxxx/privkey.pem
chain = /etc/letsencrypt/live/xxxxxxxxxxxx/chain.pem
fullchain = /etc/letsencrypt/live/xxxxxxxxxxxx/fullchain.pem

# Options used in the renewal process
[renewalparams]
account = XXXXXXXXXXXXXX
rsa_key_size = 4096
authenticator = webroot
webroot_path = /var/www,
server = https://acme-v02.api.letsencrypt.org/directory
[[webroot_map]]
app.xxxxxxxxx.xyz = /var/www
cloud.xxxxxxxxx.xyz = /var/www
```

- the **/etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh** file is in place with correct content

Any help is appreciated, also let me know if i should provide more logs or infos.

Joulinar, 2022-09-12:
For Lighttpd it might be needed to restart the web server to take effect. At least something we have seen 2-3 times in the past.

MichaIng, 2022-09-12:
Ah right the service wasn't reloaded some versions ago, now since you did rerun `dietpi-letsencrypt`, `/etc/letsencrypt/renewal-hooks/deploy/dietpi-lighttpd.sh` should contain the reload, done after any actual cert deployment.  You can test it with:
```sh
certbot renew --force-renewal
```

eikaramba, 2022-09-13:
yeahhh that command actually worked as expected and the certificate was renewed and installed automatically. final test is to see the regular renewal work in a few weeks or so. but i will close this ticket for now. should i experience any more troubel i will reopen it. thank you.

closed 2022-09-13 by eikaramba