MichaIng/DietPi #5743: Raspberry Pi Pico W

symbios24, 2022-09-13:
Hello

Is it possible to make an image for the raspberry pi pico w??

I know that most of the software will not be supported by pico w but software like hotspot i think it should work

In general add only the supported software

Thanks

ravenclaw900, 2022-09-14:
Hi,
Building an image for the Pico is probably impossible. It only has 264 KB of RAM, while DietPi requires at least a few MB to boot. In addition, there's probably not even a Linux kernel that can run on the RP2040 chip used by the Pico. The Pico is not designed to run a complete operating system, it is meant to run a program written for bare metal. Finding software written like that (or writing your own) is probably your best bet for using the Pico.

MichaIng, 2022-09-15:
Jep, the Pico is microcontroller, no SBC. It runs native code, like MycroPython, but cannot run a regular OS like Debian or DietPi. You'll notice that the Raspberry Pi OS cannot be used on RPi Pico either.

I'll hence mark this as closed.

closed 2022-09-15 by MichaIng