MichaIng/DietPi #5745: Image Request for Orange Pi i96

symbios24, 2022-09-15:
Hello

Is it possible to make an image for the:

Orange Pi i96 ??

Thanks

Joulinar, 2022-09-15:
I'm not sure if there is an Armbian base image available that could be used as starting point. At least I did not found one.

symbios24, 2022-09-15:
There is a bullseye image here:

https://github.com/TheRemote/Legendary-OrangePi-i96/releases/download/1.12/Legendary_OrangePi_i96_debian_bullseye_server_v1.12.tar.xz

There also an official debian image here

https://drive.google.com/drive/mobile/folders/1tYs1pARc7XNYLaF8zdPqjDOQ8VOrrFJt?usp=drive_open

Joulinar, 2022-09-15:
In this case, you could try it and install one of these images. If they are running, have a look to our install script. It might be able to install DietPi on top https://dietpi.com/docs/hardware/#make-your-own-distribution

MichaIng, 2022-09-15:
Sadly both images use Linux 3.10, which isn't really supported anymore by Debian Bullseye. Many things won't work, some only with workarounds, hence we cannot produce a reasonable image for broad usage for this SBC anymore. The SoC itself also is from a manufacturer I don't even know, so zero change that there is any kernel development done upstream for it.

I'll hence mark this as closed. For your personal usage, feel free to try the `dietpi-installer` as suggested by Joulinar.

closed 2022-09-15 by MichaIng