MichaIng/DietPi #5749: Orange Pi One login issue

symbios24, 2022-09-17:
Hello there

I made a custom image for the orange pi one by using the quide here and selecting the generic H3 from the boards section

I have one issue when i put the root/dietpi password it takes me for a sec to the main after-login screen and then it goes back again to login screen asking the username/password again this can go on forever,,only the dietpi/dietpi login works correct but i don't have root access like this
Is there a fix for the root login issue ??

I used the latest Bullseye Image from Armbian

Armbian_22.08.1_Orangepione_bullseye_current_5.15.63

Thanks

MichaIng, 2022-09-17:
Many thanks for your report.

When you login as `dietpi`, does it show the same behaviour when you run `sudo -s` and/or `sudo -i`, starting an interactive resp. login root subshell?

symbios24, 2022-09-17:
You mean if i login with the dietpi/dietpi account if it shows the same screen after with the ip/temp etc? infos ??

Joulinar, 2022-09-17:
if logged in with user `dietpi`, is `sudo -s` and/or `sudo -i` working to switch to user `root`

symbios24, 2022-09-17:
sudo -s is working is reloading the main screen and im in root axcount

sudo -i is not working it says: sudo /usr/bin/zsh command not found

MichaIng, 2022-09-17:
Ahh, `zsh` is configured as default shell. Did you select it when logging into the Armbian image the first time? Looks like we should reset it to `bash` in `dietpi-installer`.

symbios24, 2022-09-17:
Yes i selected the zsh in the first login in armbian

symbios24, 2022-09-17:
How can i fix that manual now?
Or should i wait until the next dietpi update to re-install it?

MichaIng, 2022-09-17:
```sh
sudo usermod -s /bin/bash root
```

MichaIng, 2022-09-17:
Done now as well by `dietpi-installer`: https://github.com/MichaIng/DietPi/commit/3e94f09

symbios24, 2022-09-17:
Thanks a lot

MichaIng, 2022-09-17:
Marking as done.

closed 2022-09-17 by MichaIng

symbios24, 2022-09-19:
Hi i have one question i see dietpi releases of bookworm also is this in beta/testing stage or its official released?

Joulinar, 2022-09-19:
It is testing, same as bookworm itself.

MichaIng, 2022-09-19:
As long as you can live with breaking changes being shipped with package upgrades, and that the one or the other software install might not work yet, you are welcome to help testing 🙂: https://github.com/MichaIng/DietPi/wiki/Debian-Bookworm-testing

MichaIng, 2022-09-23:
Can you link the guide or show the steps you did? Usually it should work via device tree overlay on Armbian and DietPi just the same way.

symbios24, 2022-09-25:
How can i de-activate the swap as is never been used and taking half my cards space,i cannot do it from the drive manager,any other way??

Joulinar, 2022-09-25:
What is the issue using drive manager? There you should be able to set a value of 0 to disable SWAP.

symbios24, 2022-09-25:
I use a small card 4gb and the swap is set to 1.5gb when i select it is says its not possible to set bigger size i think,can i set it al least 100mb?
As i only use dietpi as a hotspot.

Joulinar, 2022-09-25:
I guess you like to disable? Did you tried 0 as value?

symbios24, 2022-09-25:
I cannot enter inside the swap menu to configure it,it says something about the minimum size

MichaIng, 2022-09-25:
A bug in the drive manager indeed. Fixed with: https://github.com/MichaIng/DietPi/commit/ab9523f
It didn't allow to change the swap file size at all, if not at least 2 GiB free space were available. This of course doesn't make sense when it's about disabling the swap file. Also, if there is a swap file on that drive currently, then it's size needs to be taken into account as well. And last but not least, this 2 GiB free space requirement doesn't make so much sense at all, e.g. when external drives are used for the swap file only and generally on non-system drives. Only on the root filesystem a certain free space is important, but this can be, depending on the setup, much less than 2 GiB of course.

MichaIng, 2022-09-25:
Ah, to remove the swap file on your system:
```sh
/boot/dietpi/func/dietpi-set_swapfile 0
```

symbios24, 2022-09-26:
> Can you link the guide or show the steps you did? Usually it should work via device tree overlay on Armbian and DietPi just the same way
.
I usually do this to enable the usb otg port in Armbian:

cd /boot/dtb
sudo cp sun8i-h3-orangepi-one.dtb sun8i-h3-orangepi-one.dtb.old
sudo dtc -I dtb -O dts -o sun8i-h3-orangepi-one.dts sun8i-h3-orangepi-one.dtb
sudo sed -i -e 's/dr_mode = "otg";/dr_mode = "host";/g' sun8i-h3-orangepi-one.dts
sudo dtc -I dts -O dtb -o sun8i-h3-orangepi-one.dtb sun8i-h3-orangepi-one.dts

But after installing dietpi this method is not working anymore.

MichaIng, 2022-09-30:
```sh
sudo sed -i -e 's/dr_mode = "otg";/dr_mode = "host";/g' sun8i-h3-orangepi-one.dts
```
But this disables OTG mode and enables host mode. The two words need to be exchanged.

There is no related device tree overlay in:
```sh
ls -l /boot/dtb/rockchip/overlay
```