MichaIng/DietPi #5750: dietpi-imager not working!

DrCWO, 2022-09-17:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye 0

- Kernel version | `uname -a`
Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
5V 3A

- SD card used | (EG: SanDisk ultra)
SanDisk ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dietpi-imager via this commend:
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"

- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

#### Steps to reproduce

1. insert SD-Card in SD-Card reader
2. bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
3. select /dev/sda
4. select /dev/sda1
5. select dd
6. enter "backup" (filename)
7. start imager
8. I get this:
 - Command: ln -s /tmp/DietPi-Imager_mnt/etc/systemd/system/dietpi-fs_partition_resize.service                       │
                │ /tmp/DietPi-Imager_mnt/etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service                   │
                │  - Exit code: 1                                                                                                      │
                │  - DietPi version: v8.8.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                   │
                │  - Image creator: DietPi Core Team                                                                                   │
                │  - Pre-image: from scratch                                                                                           │
                │  - Error log:                                                                                                        │
                │ ln: failed to create symbolic link                                                                                   │
                │ '/tmp/DietPi-Imager_mnt/etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service': No such file  

#### Expected behaviour

- Should create the image file...

#### Actual behaviour

- Stops with the error code above

#### Extra details

- none

MichaIng, 2022-09-17:
Many thanks for your report. Indeed the directory does not necessarily exist as long as it's not a freshly created image (`dietpi-installer`). I pushed a fix to create it, if missing: https://github.com/MichaIng/DietPi/commit/a2f96dd
Please try:
```sh
bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"
```

MichaIng, 2022-09-17:
Marking this as solved.

closed 2022-09-17 by MichaIng