MichaIng/DietPi #5753: Open Beta v8.9 | Please help testing and hardening our upcoming release

MichaIng, 2022-09-17:
RC version | v8.9.0
---------- | -------
v8.9.0 PR | #5746
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2022-09-24**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.9
____
### v8.9.0
_(2022-09-17)_

#### Enhancements
- Raspberry Pi | The raspberrypi-sys-mods package has been replaced with an own one, using only the small subset of the contained tweaks, which are useful on DietPi as well. A recent update of the package added a large number of dependencies, required for rpi-imager setup steps on Raspberry Pi OS only: https://github.com/MichaIng/DietPi/issues/5734
- DietPi-Installer | The root user home directory and login shell are now reset to /root and /bin/bash, as expected by the fresh DietPi system. Many thanks to @symbios24 for reporting a case where root logins did fail because zsh was the login shell, but missing on a fresh DietPi system: https://github.com/MichaIng/DietPi/issues/5749
- DietPi-Drive_Manager | mergerfs fstab entries are now preserved. Many thanks to @paradix for suggesting this enhancement: https://github.com/MichaIng/DietPi/issues/5622
- DietPi-Software | Plex Media Server: The "plex" system group is now removed on uninstall. This was not done automatically on package removal, since we change the primary group of the "plex" system user to "dietpi". Many thanks to @mail2rst for reporting this issue: https://dietpi.com/forum/t/plex-installation-broken-via-dietpi-software-after-uninstallation-of-docker-docker-compose/14130
- DietPi-Software | Go: GOPATH is not set anymore to /mnt/dietpi_userdata/go on fresh installs and will default to ~/go instead. This especially makes multi-user setups easier where usually every user wants its own Go workspace. Many thanks to @tlgs for bringing this to our attention: https://github.com/MichaIng/DietPi/issues/5735
- DietPi-Software | HAProxy: Bumped version to latest stable v2.6.5, update via reinstall: dietpi-software reinstall 98
- DietPi-Software | phpBB: Bumped version for new installs to latest stable v3.3.8, update it by following the instructions on the web UI ACP.
- DietPi-Software | NoMachine: Bumped version to latest stable v7.10.1_1, update via reinstall: dietpi-software reinstall 30

#### Bug fixes
- DietPi-Imager | Resolved an issue where the script could have failed because of a missing directory. Many thanks to @DrCWO for reporting this issue: https://github.com/MichaIng/DietPi/issues/5750
- DietPi-Drive_Manager | Resolved an issue with the reserved blocks option where the entered value was accidentally applied as number of blocks instead of as MiB, leading to a much smaller reserved space as intended.
- DietPi-Software | ownCloud: Resolved an issue where the install failed because of a changed download URL. Many thanks to @DrMoshSince1988 for reporting this issue: https://github.com/MichaIng/DietPi/issues/5739
- DietPi-Software | Amiberry: Resolved an issue where mouse movement did not work because of change in the latest libSDL2 version. Many thanks to @lilltonka for reporting this issue and @midwan for the fix in upcoming Amiberry v5.4: https://github.com/MichaIng/DietPi/issues/5730, https://github.com/BlitterStudio/amiberry/issues/995

Joulinar, 2022-09-20:
Patch for `raspberrypi-sys-mods` seems to be working fine. Unwanted packages got removed automatically. 👍 

```
[  OK  ] DietPi-Patch | Patched to DietPi v8.9
[ INFO ] DietPi-Update | APT autopurge, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be REMOVED:
  alsa-utils* cpio* gettext-base* initramfs-tools* initramfs-tools-core*
  klibc-utils* libasound2* libasound2-data* libatopology2* libfftw3-single3*
  libgomp1* libklibc* libsamplerate0* linux-base* lua5.1* raspi-config*
0 upgraded, 0 newly installed, 16 to remove and 0 not upgraded.
                                                               After this operation, 9692 kB disk space will be freed.
(Reading database ... 22992 files and directories currently installed.)                                               (Reading database ...
Removing raspi-config (20220915) ...
Removing alsa-utils (1.2.4-1+rpt1) ...
Removing initramfs-tools (0.140) ...
Removing initramfs-tools-core (0.140) ...
Removing cpio (2.13+dfsg-4) ...
Removing gettext-base (0.21-4) ...
Removing klibc-utils (2.0.8-6.1+rpi1) ...
Removing libatopology2:armhf (1.2.4-1.1+rpt2) ...
Removing libasound2:armhf (1.2.4-1.1+rpt2) ...
Removing libasound2-data (1.2.4-1.1+rpt2) ...
Removing libfftw3-single3:armhf (3.3.8-2) ...
Removing libgomp1:armhf (10.2.1-6+rpi1) ...
Removing libklibc:armhf (2.0.8-6.1+rpi1) ...
Removing libsamplerate0:armhf (0.2.1+ds0-1) ...
Removing linux-base (4.6) ...
Removing lua5.1 (5.1.5-8.1+b2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...
(Reading database ... 22529 files and directories currently installed.)
Purging configuration files for linux-base (4.6) ...
Purging configuration files for initramfs-tools (0.140) ...
Purging configuration files for alsa-utils (1.2.4-1+rpt1) ...
Purging configuration files for initramfs-tools-core (0.140) ...
Purging configuration files for raspi-config (20220915) ...
[  OK  ] DietPi-Update | APT autopurge

```

MichaIng, 2022-09-22:
#### Beta v8.9.1 has been merged: #5754
- DietPi-Survey/DietPi-BugReport | From Debian Bookworm on, for survey and bug report uploads to DietPi servers, the deprecated RSA host key algorithm has been replaced by the modern Ed25519 algorithm. The known hosts entry has been additionally hardened to allow only connecting to a specific non-default port.

MichaIng, 2022-09-24:
#### DietPi v8.9 has been released: #5760
Many thanks to all testers ❤️ !

closed 2022-09-24 by MichaIng