MichaIng/DietPi #5755: [FAILED] DietPi-TimeSync | systemctl start systemd-timesyncd

p4cm4n, 2022-09-18:
#### Details:
- Date           | Sun Sep 18 19:09:56 CEST 2022
- DietPi version | v8.8.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 1
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service is masked.
```

MichaIng, 2022-09-18:
```
systemd-timesyncd.service is masked.
```
=>
```sh
systemctl unmask systemd-timesyncd
/boot/dietpi/func/run_ntpd 1
```

p4cm4n, 2022-09-18:
I tried it but got the same error.

I already uploaded a bug report e316e724-8d1f-4cb5-ae98-462918aa8df2

MichaIng, 2022-09-18:
What was the output of `systemctl unmask systemd-timesyncd`? Can you also show:
```sh
ls -l /etc/systemd/system/systemd-timesyncd.service
systemctl status systemd-timesyncd
```

Joulinar, 2022-09-19:
did you install something like OMV on your system? Or some other app that might have installed an own NPT service? Or did you have done an upgrade Buster > Bullseye?

MichaIng, 2022-09-19:
Ah, then probably the package is missing:
```sh
apt install systemd-tmesyncd
/boot/dietpi/func/run_ntpd 1
```

MichaIng, 2022-09-20:
It seems you installed a later systemd version from backports. Try this:
```sh
apt install systemd-timesyncd/bullseye-backports
/boot/dietpi/func/run_ntpd 1
```

p4cm4n, 2022-09-20:
Worked like a charm! :D

Thank you once again for your awesome work! :)

closed 2022-09-20 by p4cm4n