MichaIng/DietPi #5756: G_OBTAIN_CPU_TEMP from python ?

jalsco, 2022-09-19:
Hello,  hope someone can give some quick - howto - advice 

Just am trying to use :
`G_OBTAIN_CPU_TEMP`
`G_OBTAIN_CPU_USAGE`
from within a small python script.

Usually I use :  `subprocess.Popen`
I also tried using  `os.Popen`

Both of these I use successfully in other scripts - but python cannot find these 2 DietPi commands 🤷🏻‍♂️

MichaIng, 2022-09-19:
Those are no executables but shell functions. So you'd need to load `dietpi-globals` first:
```py
    now_cpu = subprocess.run(['bash -c ". /boot/dietpi/func/dietpi-globals; G_OBTAIN_CPU_USAGE"'],
                             stdout=subprocess.PIPE,
                             stderr=subprocess.PIPE,
                             text=True,
                             shell=True)
```

jalsco, 2022-09-19:
thank you so much !! 🙏🏻🙏🏻🙏🏻😁

closed 2022-09-19 by jalsco