MichaIng/DietPi #5758: Boot error message

mixpc, 2022-09-22:
#### Details:
- Date           | Thu Sep 22 16:14:36 BST 2022
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -6nc 1 -W 10 2620:fe::fe`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ... flash sd card, place it in the Rasberry Pi 3 sd card reader
2. ... power on, ssh with boot default user root and password and after login there's this message to post an issue in GitHub
#### Expected behaviour:

- ... Simple first boot, as always
#### Actual behaviour:

- ... This error message is showing in the screen
#### Extra details:

- ... I cannot say much more. DietPie alwayed booted flawlessly. It's the first time I notice this message. But running ping www.google.com works! So i don't know what could go wrong.
#### Additional logs:
```
PING 2620:fe::fe(2620:fe::fe) 56 data bytes
From fe80::ce32:e5ff:fe6e:6a84%eth0 icmp_seq=1 Destination unreachable: No route

--- 2620:fe::fe ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms
```

Joulinar, 2022-09-22:
Looks like IPv6 is not working on your network. In the error screen, you should have a possibility to disable IPv6 directly.

mixpc, 2022-09-22:
@Joulinar Excellent! Indeed, IPv6 was disabled. I have chosen do disable it in the pop-up menu (CONFIG_ENABLE_IPV6=0) and no issues at all. Thank you!

```
[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[FAILED] DietPi-Update | Checking IPv6 network connectivity
 - Command: ping -6nc 1 -W 10 2620:fe::fe
[ SUB1 ] DietPi-Set_hardware > enableipv6 (disable)
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_ENABLE_IPV6=0
[  OK  ] enableipv6 disable | Completed
[  OK  ] DietPi-Update | Checking IPv6 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
```

closed 2022-09-22 by mixpc