MichaIng/DietPi #5759: MineOS can't log in

maybaxstv, 2022-09-22:
### Creating a bug report/issue

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
- SBC model | RPi4
- Power supply used | 5V 3A
- SD card used | SanDisk ultra 64 gb

#### Additional Information (if applicable)
- Software title | MineOS
- The software was  installed freshly 
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Install MineOS  from dietpi-software

#### Expected behaviour

- go to https://<your.IP>:8443
- enter your credentials and log in to the app

#### Actual behaviour

- When I try to login in app, usssing Username: root,  Password: The same as your root login password, I can't log in

#### Extra details

- ...

MichaIng, 2022-09-22:
Many thanks for your report.

Ah, it is still the missing yescrypt support: https://github.com/hexparrot/mineos-node/issues/441

Still not sure how to deal with this, disabling MineOS for Bullseye or offering the yescrypt => SHA512 change with a bug warning 🤔.

MichaIng, 2022-09-22:
Ah, it seems to be possible to set the password for a single user with SHA512 explicitly:
```sh
chpasswd --crypt-method SHA512 <<< 'user:password'
```
In combination with #4551 this would be a solution.

Ast3risk-ops, 2022-09-25:
Same. MineOS is severely out of date

Ast3risk-ops, 2022-09-25:
With npm

MichaIng, 2022-09-25:
> MineOS is severely out of date

Not sure what you mean, we pull/update to the latest MineOS and dependency versions. It is latest MineOS which does not support the default UNIX user password encryption algorithm since Debian Bullseye, hence an upstream issue, as linked above.

Ast3risk-ops, 2022-09-25:
@MichaIng Well, npm goes insane when it is set up, login fails, the OS that has it is old, the site is old, I just feel that it kinda sucks an pterodactyl is better as it is easier to use.

MichaIng, 2022-09-25:
> the OS that has it is old, the site is old

What do you mean with "OS is old"? The Node.js implementation of MineOS is [actively developed](https://github.com/hexparrot/mineos-node/pulse), just nobody finds time to implement support for yescrypt, that's all.

Also not sure what you mean by "npm goes insane"? It is common that Node modules get updated when you do a reinstall/update after a while, but that is perfectly fine and does not cause any issues.

Feel free to open a software request for pterodactyl, but let's not mix it with this MineOS issue.

Ast3risk-ops, 2022-09-25:
@MichaIng I guess I will. MineOS sucks compared to that

MichaIng, 2024-11-09:
PR up to create again a non-root login user. Seems to work all well: #7276
Its password will be stored with SHA512 hash, so login works OOTB.

MichaIng, 2024-11-09:
Fixed for next release. Can be done manually like this:
```sh
sudo useradd -rMU -d /mnt/dietpi_userdata/mineos/serverdata -s /usr/sbin/nologin mineos
sudo chpasswd --crypt-method SHA512 <<< 'mineos:dietpi'
sudo chown -R mineos /mnt/dietpi_userdata/mineos/serverdata
sudo systemctl restart mineos
```

mikedebian, 2024-11-09:
I would also like to note that with newer minecraft versions, the jdk that mineos uses is severely outdated and wont start any server. 

For x64 I use curl -O https://download.bell-sw.com/java/22.0.2+11/bellsoft-jdk22.0.2+11-linux-amd64.deb

For the pi I use curl -O https://download.bell-sw.com/java/22.0.2+11/bellsoft-jdk22.0.2+11-linux-aarch64.deb

MichaIng, 2024-11-09:
Right, I recognised the same. Nasty that it does not show any error, but the server does just not start, service logs containing the start request, but nothing after that. Do you know where to find logs from the actual Minecraft servers?

However, I added a note about that to our docs: https://github.com/MichaIng/DietPi-Docs/pull/1034/files

mikedebian, 2024-11-10:
I find logs in /mnt/dietpi_userdata/mineos/serverdata/servers/ServerName/logs and in /var/log/mineos.log

MichaIng, 2024-11-11:
Ah I think the problem was/is that the Minecraft server does not even start any logging if the Java version is too low. Somewhat a gap in logging, as the initial Java executable output is not logged to journal and `/var/log/mineos.log`, and not done to the Minecraft server logs either, as those likely require the server to be actually running, loading some Java logging system/library.

closed 2024-12-25 by MichaIng