MichaIng/DietPi #5764: systemctl restart mariadb fails

princp69, 2022-09-24:
#### Details:
- Date           | Sat Sep 24 18:35:31 BST 2022
- DietPi version | v6.34.0 (MichaIng/stretch)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux`
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | `systemctl restart mariadb`
- Exit code      | 5
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Failed to restart mariadb.service: Unit mariadb.service not found.
```

Joulinar, 2022-09-24:
Looks like MariaDB is not installed. Can you check available packages 

```
dietpi-software list | grep "ID 88"
dpkg -l mariadb*

```

MichaIng, 2022-09-24:
Many thanks for your report.

Before we start to investigate outdated issues, please update your DietPi to latest version:
```sh
dietpi-update 1
```

You should also consider to upgrade to a supported Debian version, since Stretch isn't supported by DietPi anymore, hence stays at v7.9 (instead of latest 8.8). Also on Debian itself it is oldoldstable and not supported by the Debian security team anymore, but only minimally by LTS team: https://dietpi.com/blog/?p=1001

princp69, 2022-09-25:
Thanks for the rapid reply.

I went ahead to upgrade to 8.9.2 onto a new card. However, I now cannot get my Hifiberry DAC+ Pro working. I’ve used dietpi-config and dietpi-software to update and install, but aplay -l returns no soundcards found.

Any help appreciated.

MichaIng, 2022-09-25:
You did do a reboot after changing the sound card in `dietpi-config`? Can you verify that the device tree overlay is enabled?
```sh
grep 'dtoverlay' /boot/config.txt
```
If so, please show the output of:
```sh
lsusb
dmesg -l emerg,alert,crit,err
```

princp69, 2022-09-25:
Again, thanks for such a quick reply:

root@DietPi:~# grep 'dtoverlay' /boot/config.txt
dtoverlay=disable-wifi
dtoverlay=hifiberry-dacplus
**This looks good!**

root@DietPi:~# lsusb
Bus 001 Device 004: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC) 
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@DietPi:~# dmesg -l emerg,alert,crit,err
[    8.804690] pcm512x 1-004d: Failed to reset device: -121
[   10.174210] hwmon hwmon1: Undervoltage detected!
[   11.040913] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped

princp69, 2022-09-25:
And, yes, multiple reboots.

MichaIng, 2022-09-25:
Ah, it's not a USB DAC 😅. Did you try the `hifiberry-dacplusadcpro` overlay instead of `hifiberry-dacplus`?

But probably the issue is undervoltage:
```
[ 8.804690] pcm512x 1-004d: Failed to reset device: -121
[ 10.174210] hwmon hwmon1: Undervoltage detected!
```
The 5.x kernel indeed is known to have higher peaks, also different p-states, at least I remember a few cases where undervoltage issues stated with Linux 5.4 where non were before. Do you have a different PSU or USB-cable to test?

You can also try to disable initial turbo via `dietpi-config` or:
```sh
sed -i '/^initial_turbo=/d' /boot/config.txt
```

princp69, 2022-09-25:
No, not USB but a HAT board, Hifiberry DAC+ Pro.
Seem to have the right driver; the same as on the old install that worked fine until I tried updating.

I've disabled turbo. No impact.
Also tried another PSU, but again no luck, still no sound card.

root@DietPi:~# dmesg -l emerg,alert,crit,err
[    8.244673] pcm512x 1-004d: Failed to reset device: -121
[   21.178482] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped

MichaIng, 2022-09-30:
Do you have I2C enabled or disabled? Could you try to toggle it? Can be done in `dietpi-config` advanced options.

princp69, 2022-10-01:
Yes, I2C is enabled. I've tried it on and off, so difference.
Not sure what you mean about the device driver.

princp69, 2022-10-01:
I've tried a completely fresh install with a new card.
Same problem.
dmesg -l emerg,alert,crit,err
[    4.895295] pcm512x 1-004d: Failed to reset device: -121

MichaIng, 2022-10-02:
And did you try the `hifiberry-dacplusadcpro` overlay?

Otherwise I think we'd need to check on RPi forum. Must be related to the kernel upgrade, but I think if it was a kernel bug affecting all HiFiBerry DAC+ Pro, it would have been reported at the RPi repo already already.

princp69, 2022-10-10:
It does seem there are similar problems for others. How do I proceed now?
Replied here too: https://github.com/raspberrypi/linux/pull/4904#issuecomment-1273809160.

princp69, 2022-10-11:
Thanks for the rapid reply.

I went ahead to upgrade to 8.9.2 onto a new card. However, I now cannot get my Hifiberry DAC+ Pro working. I’ve used dietpi-config and dietpi-software to update and install, but play -l returns no soundcards found.

FYI the MariaDB error was odd as it wasn’t installed.
___________
Philipp Prince
07966 008892

MichaIng, 2022-10-11:
Just to assure, that HAT did work before, right (as of potential hardware incompatibility asked in the other issue, although RPi 4 only)? `force_eeprom_read=0``was added as an attempt to solve the issue, right?

Let's hope j-schambacher can help debugging the case, as I'm out of ideas.

princp69, 2022-10-16:
I've managed to get it working. Multiple installs. Re-sitting the board. Checking everything. Something was amiss. Now it works! Thanks for your help.

Joulinar, 2022-10-16:
I guess we could close the issue in this case?

closed 2022-10-23 by MichaIng