MichaIng/DietPi #5765: Repository 'https://dl.ubnt.com/unifi/debian stable InRelease' changed its 'Codename' value from 'unifi-7.1' to 'unifi-7.2'

massimopisani, 2022-09-24:
#### Details:
- Date           | Sat Sep 24 21:16:29 CEST 2022
- Bug report     | 0783811e-106f-43ec-a39e-ed4fea05a6f9
- DietPi version | v7.8.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux RPi200 5.10.63-v7+ #1488 SMP Thu Nov 18 16:14:44 GMT 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 http://archive.raspbian.org/raspbian stretch InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Get:1 https://dl.ubnt.com/unifi/debian stable InRelease [3038 B]
Reading package lists...
E: Repository 'https://dl.ubnt.com/unifi/debian stable InRelease' changed its 'Codename' value from 'unifi-7.1' to 'unifi-7.2'
```

Joulinar, 2022-09-24:
Did you add the repository `https://dl.ubnt.com` yourself?

Try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

MichaIng, 2022-09-25:
This is a repository for the UniFi OS and applications: https://www.ui.com/
However, it is not part of our software portfolio, hence this must have been added manually or by a 3rd party installer.

APT allows a suite change for a given codename, i.e. if `unifi-7.2` changed from `unstable` to `stable`, or `bookworm` from `testing` to `stable`, without interactive, but not the other way round. Running `apt update` however should give you a prompt to accept the codename change.

closed 2022-10-23 by MichaIng