MichaIng/DietPi #5766: Software-Request | WebWormhole

ingoiot, 2022-09-25:
### Creating a software request

#### Formal software information
- Software title | WebWormhole 
- Short description | WebWormhole creates ephemeral pipes between computers to send files
or other data.

- Official URL (if available) | https://webwormhole.io/
- GitHub URL (if available) | https://github.com/saljam/webwormhole
- (Official) install documentation | https://github.com/saljam/webwormhole

#### Are there similar/alternative software titles available with DietPi-Software?
no
- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
 download and compile and start server on a free port.

Ast3risk-ops, 2022-09-25:
Uhhhh, you mean wormhole, correct?

Ast3risk-ops, 2022-09-25:
The file sharing thingy? This looks like a web interface for it