MichaIng/DietPi #5771: DiePi own distribution script - no bash

HolgerTB, 2022-09-26:
After creating a custom diet distribution on a Hetzner Cloud (bas Debian 11 installation) server using the script

bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"

the bash is not starting automatically.

I installed dietpi last week on a couple of cloud instances without any issue. Today I had this issue.

After starting bash manually on the console everything is working fine

Joulinar, 2022-09-26:
Hi Holger, you mean there is no output while executing the command? You are using root user, right? I just tested it on my R5S without issues. 🤔

HolgerTB, 2022-09-26:
After login as root only # is prompted. after entering bash the shell is loaded and the configuration starts.

entering chsh -s /bin/bash fixed the problem

Joulinar, 2022-09-26:
this is right after the install script finished and system rebooted??

HolgerTB, 2022-09-26:
Yes. Correct. 

And when I start bash and finish installation and reboot there is no bash

Joulinar, 2022-09-26:
how does it behave on user dietpi? Maybe this change could be the issue https://github.com/MichaIng/DietPi/commit/3e94f098657c6f343390bd5f7d803e2a93556923

HolgerTB, 2022-09-26:
i will setup a new instance and post the outcome. stand by

Joulinar, 2022-09-26:
but you should be able to login as user `dietpi` on the running system??

HolgerTB, 2022-09-26:
I did the first login after installer script and reboot as dietpi. the prompt is 

[FAILED] Root login required                                                           
To finish DietPi first run setup, root permissions are required
Please login again as user "root" with password "dietpi", respectively the one you chose via "dietpi.txt".

HolgerTB, 2022-09-26:
I finished installation and I can logon as dietpi. As root I don't have a bash

Joulinar, 2022-09-26:
@MichaIng can you have a look to the latest changes you did on the installer?

HolgerTB, 2022-09-26:
i tested it on Friday. It worked without any issues.

Joulinar, 2022-09-26:
yes, the new installer has been released on weekend. And there is a change to fix/clear the shell 🙄 

```
usermod -d /root -s /bin/dash root
```

but it should be:

```
usermod -d /root -s /bin/bash root
```

This should fix it as well

HolgerTB, 2022-09-26:
great.

Joulinar, 2022-09-26:
did you tried running following? `usermod -d /root -s /bin/bash root`. After new login, it should be fixed.

HolgerTB, 2022-09-26:
yes. This fixed it

closed 2022-10-22 by MichaIng