MichaIng/DietPi #5776: Out of memory warning

barukh27, 2022-09-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | DietPi v8.9.2 
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | bullseye 1
- Kernel version |  5.15.61-v7l+ #1579
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | (EG: 5V 1A RAVpower)

#### Steps to reproduce
The other day I updated the OS to the new version. I reinstalled Kodi too (I wanted to get rid of the libva error). Out of the blue the system started to show some "out of memory" warnings and some killed processes. Even if I reboot the system the processes are killed.

It also says: context.c:56 warning mpde_setiminalloc ignoring a request to set MPD_MINALLOC a second time.

free -m reads as the following:

root@DietPi:~# free -m
               total        used        free      shared  buff/cache   available
Mem:            3595        1114         788          89        1692        2349
Swap:              0           0           0

What should I do?

Thanks in advance.

Joulinar, 2022-09-28:
can you check for kernel error messages once proccesses got killed

barukh27, 2022-09-28:
Thank you for your time and help.

root@DietPi:~# dmesg -l err,crit,alert,emerg
[14403.139870] Out of memory: Killed process 24804 (jellyfin) total-vm:537444kB, anon-rss:120748kB, file-rss:77292kB, shmem-rss:0kB, UID:111 pgtables:672kB oom_score_adj:0
[14437.679306] Out of memory: Killed process 25294 (jellyfin) total-vm:530092kB, anon-rss:118056kB, file-rss:76896kB, shmem-rss:0kB, UID:111 pgtables:668kB oom_score_adj:0
[14468.344083] Out of memory: Killed process 25540 (jellyfin) total-vm:529040kB, anon-rss:117944kB, file-rss:76856kB, shmem-rss:0kB, UID:111 pgtables:676kB oom_score_adj:0
[14512.113123] Out of memory: Killed process 25860 (jellyfin) total-vm:553744kB, anon-rss:118468kB, file-rss:76848kB, shmem-rss:0kB, UID:111 pgtables:680kB oom_score_adj:0
[14556.911477] Out of memory: Killed process 26166 (jellyfin) total-vm:537260kB, anon-rss:118392kB, file-rss:76968kB, shmem-rss:0kB, UID:111 pgtables:680kB oom_score_adj:0
[14608.615603] Out of memory: Killed process 26596 (jellyfin) total-vm:570200kB, anon-rss:117964kB, file-rss:76920kB, shmem-rss:0kB, UID:111 pgtables:684kB oom_score_adj:0
[14627.261034] Out of memory: Killed process 26990 (jellyfin) total-vm:484428kB, anon-rss:135284kB, file-rss:67316kB, shmem-rss:0kB, UID:111 pgtables:656kB oom_score_adj:0

Joulinar, 2022-09-28:
you could try to add some SWAP file to see how it behave. 

```
/boot/dietpi/func/dietpi-set_swapfile 2048
```

MichaIng, 2022-09-28:
You can check `htop` for which process is consuming so much RAM.

barukh27, 2022-09-28:
The problem persists.

[image]

Radarr and Jellyfin appear to consume the most.

MichaIng, 2022-09-28:
You can scroll down or sort by `RES` (resident memory consumption) to find the ones which consume most. However, currently less than 1 GiB is used. Probably it's during peak times, when downloads are done, Jellyfin and/or Kodi do video transcoding? You can check Kodi and Jellyfin memory consumption while playing/streaming videos.

Joulinar, 2022-09-28:
yap keep `htop` open and watch how memory consumption is doing whil using Jellyfin and/or Kodi.

barukh27, 2022-09-29:
Some of the processes are down again. Htop shows:

[image]

Joulinar, 2022-09-29:
interesting would be information from `htop` exactly the moment your services get killed. Is it `jellyfin` the only one getting killed or are there other es well?

barukh27, 2022-09-29:
There are others too.

Radarr, mono, qbitorrent...

barukh27, 2022-09-29:
Should I perform a fresh start?

MichaIng, 2022-09-29:
Can you click on `RES` to sort by RAM consumption. Then monitor it for a while and see whether one of the processes raises much higher. At the moment, memory consumption is very fine.

barukh27, 2022-09-29:
I've restarted the processes. After a while this is what htop shows.

[image]

Thank you for your help and interest.

MichaIng, 2022-09-29:
Please open `htop`, then on the green line, click with `RES` field with your mouse (through PuTTY), so that the process list is sorted accordingly. Otherwise we do not see all memory consuming processes.

What are those Docker container, you run on the machine? It looks like there are a bunch of zombie containers, which do not run any process. Only one runs a `java` process, but we do not see which application it actually is.

Joulinar, 2022-09-29:
btw did you reboot the entire system in meantime? If not, give it a try.

barukh27, 2022-09-29:
> Please open `htop`, then on the green line, click the `RES` field with your mouse (through PuTTY), so that the process list is sorted accordingly. Otherwise we do not see all memory consuming processes.

[image]

> What are those Docker container, you run on the machine? It looks like there are a bunch of zombie containers, which do not run any process. Only one runs a `java` process, but we do not see which application it actually is.

It may be pyload or jdownloader.

barukh27, 2022-09-29:
> btw did you reboot the entire system in meantime? If not, give it a try.

Yes, I did so several times.

Joulinar, 2022-09-29:
if possible, continue watching memory consumption using `htop`. It seems to be growing slowly.

barukh27, 2022-09-30:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

MichaIng, 2022-09-30:
When you hit F5 key while htop is open, you get the list with the individual processes without patent>child tree view. This could make monitoring easier without scrolling.

Also, are there large files somewhere in /tmp?
```sh
G_TREEVIEW /tmp
```

ravenclaw900, 2022-09-30:
Try `G_TREESIZE` instead:
```
G_TREESIZE /tmp
```

barukh27, 2022-09-30:
> Try `G_TREESIZE` instead:
> 
> ```
> G_TREESIZE /tmp
> ```

Thanks.

This is the result:

12.0 KiB /tmp
4.0 KiB /tmp/systemd-private-ee4910ede78149d6bb07223757e40e0f-jackett.service-Qz0iji
4.0 KiB /tmp/.vnc-0
0.0 bytes /tmp/www-data-temp-aspnet-0
0.0 bytes /tmp/systemd-private-ee4910ede78149d6bb07223757e40e0f-upower.service-yNekng
0.0 bytes /tmp/systemd-private-ee4910ede78149d6bb07223757e40e0f-systemd-logind.service-V5f0bj
0.0 bytes /tmp/systemd-private-ee4910ede78149d6bb07223757e40e0f-bazarr.service-wDTohf
0.0 bytes /tmp/ssh-yJlR2nIXIpFy
0.0 bytes /tmp/ssh-PHuVY2MIWCW6
0.0 bytes /tmp/.vncserver-license
0.0 bytes /tmp/.font-unix
0.0 bytes /tmp/.XIM-unix
0.0 bytes /tmp/.X11-unix
0.0 bytes /tmp/.Test-unix
0.0 bytes /tmp/.ICE-unix

MichaIng, 2022-09-30:
Right, sorry wrong function name 😅.

In your htop I'm confused where all the system processes are, like system, the SSH server and all those 🤔. Since there is space below the Jackett process, I'd expect them to show up there.

One of the containers is jDownloader indeed, but not sure what the others are, since they have no child process, like zombies. You did a reboot since they showed up the first time, ~two days ago (so they reappeared), right?

Memory usage peaked to 1.44 GiB in between, but there are no processes visible using more then before and afterwards. Files in some tmpfs could be a reason, but your `/tmp` at least is empty.

However, currently RAM usage is all fine. Probably this helps to find the culprit:
```bash
while $(( $(free -m | mawk '/^Mem:/{print $7}') > 500 )); do sleep 1; done
ps -eo cmd,%mem,%cpu --sort=-%mem | head | tee /root/ps_output.txt
```
This sleeps until available memory falls below 500 MiB, then it prints the top memory using processes to console and into `/root/ps_output.txt` for reviewing in case of a system or SSH crash. You can run this on an extra SSH session or as a background script to not disturb work.

barukh27, 2022-10-01:
I stopped some containers: heimdall, pyload

I left the terminal running for the night. The console doesn't show anything but the line.

The ps_output document reads as:

CMD                         %MEM %CPU
/usr/lib/arm-linux-gnueabih  9.1 46.3
/usr/bin/jellyfin --webdir=  8.0 20.9
/usr/bin/mono /usr/lib/sona  4.5 21.0
/app/prowlarr/bin/Prowlarr   3.5 11.5
java -Dsun.jnu.encoding=UTF  3.5 16.2
/opt/jackett/jackett --NoRe  3.1  7.0
/usr/bin/python3 -u /opt/ba  2.5  4.8
python3 /app/calibre-web/cp  2.0  5.1
/usr/sbin/mariadbd           1.7  0.1

Joulinar, 2022-10-01:
there is not that much memory consumption in total but available memory was below 500MB, otherwise the scrit would not have printed the lines. @MichaIng could it be that `/tmp` is filling up during runtime? To test further, we could move `/tmp` out of `tmpfs`. Probably this improves the situation. 🤔

MichaIng, 2022-10-02:
Looka like `ps` is truncating the process names/paths. Any idea what `/usr/lib/arm-linux-gnueabih` is, taking ~300 MiB of RAM?

Let's check what /tmp may contain:
```bash
while (( $(free -m | mawk '/^Mem:/{print $7}') > 500 )); do sleep 1; done
ps -eo cmd,%mem,%cpu --sort=-%mem | head | tee /root/ps_output.txt
df | grep tmpfs
G_FREESIZE /tmp
```

CircuitC, 2022-10-04:
Did the fresh install resolve the issue?

barukh27, 2022-10-04:
> Did the fresh install resolve the issue?

Hi there!

I have everything up again. Not issues so far.

Thank you.

closed 2022-10-23 by MichaIng