MichaIng/DietPi #5779: Problem update version v8.7.1 to 8.9.2

jonrox, 2022-09-29:
#### Details:
- Date           | Thu Sep 29 21:35:01 CEST 2022
- Bug report     | 80d1a637-702d-4aad-b571-d940f640952a
- DietPi version | v8.7.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -y upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. when i run dietpi-update
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up rbfeeder (1:1.0.8+deb10) ...
Checking if service is running and restart, if needed...
usermod: user rbfeeder is currently used by process 5868
dpkg: error processing package rbfeeder (--configure):
 installed rbfeeder package post-installation script subprocess returned error exit status 8
Errors were encountered while processing:
 rbfeeder
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2022-09-29:
Looks like you have an app installed called `rbfeeder`. This one tries to update, but did not work as the running service prevents the system from updating the user

```
Checking if service is running and restart, if needed...
usermod: user rbfeeder is currently used by process 5868
```

Try to stop that service and run `apt update && apt upgrade`

jonrox, 2022-09-30:
thanks, I haven't close the service with `sudo systemctl status rbfeeder ` before updating

closed 2022-09-30 by jonrox