MichaIng/DietPi #5780: AP Wifi config

marcosdvpereira, 2022-10-02:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I'm using DietPi on a NanoPi NeoAir and the first boot was done through a Serial/Bluetooth dongle, which is ok for the first NanoPi but, have to solder the pin-headers just to have access to ssh through serial is a pain.

#### Describe the solution you'd like:

- Automatically enable an AP when no network on the device is detected, could have a default ssid and password, also protect confi web page with a username/password could be a plus.
- Default settings could be changed afterwards on /boot/dietpi.txt
- After network is detected (a ping to a local ip, configured on dietpi.txt, ping to a pulic webserver...) the AP is disabled. 

#### Describe alternatives you've considered:

- Similar to [this project](https://github.com/sabhiram/raspberry-wifi-conf)

#### Additional context

- In my perspective should be a out of the box feature, if the device has a wifi interface, enable this auto ap feature.
- Thank you for this great distribution

Joulinar, 2022-10-02:
Without valid network connection, you are not able to complete DietPi first run setup at all. Therefore creating an AP would not make that much sense as it still would not allow to run initial setup. 

On the other hand side, if there is WiFi available as wlan0 device, you could connect to a WiFi network automatically on first boot.

MichaIng, 2022-10-02:
What do you mean by "default SSID and password"? If/as you need to enter it anyway (`dietpi-wifi.txt`), the additional step to enable WiFi in `dietpi.txt` isn't so large?

Or you do not mean a WiFi connection to an AP but enabling a WiFi hotspot (with web interface?) on the DietPi system? Indeed a client connected to this AP could get SSH access and setup the Internet connection. But it's probably a bit insecure with a publicly known default SSID and password being setup automatically, as this is very uncommon and unexpected by most users (conpared to SSH).

marcosdvpereira, 2022-10-02:
dietpi(-wifi).txt seems easier indeed. Also, specify a ssid and password in it for when no WiFi connection detected, enables the AP to reconfigure the WiFi with that settings. By doing that it's a personal decision by making the AP secure or not.