MichaIng/DietPi #5781: Error message

bsoener, 2022-10-03:
Why do I get this error message. Please help

```
#### Details:
- Date           | Mon Oct  3 07:54:15 CEST 2022
- Bug report     | b4dc8b0d-67b3-450f-a401-b48aa6d1ab62
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | From scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y update`
- Exit code      | 100
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ...
2. ...
#### Expected behaviour:
<!-- What SHOULD happen? -->
- ...
#### Actual behaviour:
<!-- What IS happening? -->
- ...
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://downloads.plex.tv/repo/deb public InRelease
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
E: Release file for https://deb.debian.org/debian-security/dists/bullseye-security/InRelease is not valid yet (invalid for another 28min 23s). Updates for this repository will not be applied.
```

Joulinar, 2022-10-03:
You system time doesn't seems to be in sync. Try to synchronise manually.

```
/boot/dietpi/func/run_ntpd 1
apt update
apt upgrade
```

bsoener, 2022-10-03:
wow. thanks for the quick reply. and it works

closed 2022-10-03 by bsoener