MichaIng/DietPi #5782: odroid c4 installation issue

Power-onoff, 2022-10-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Unable to verify
- 
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Unable to verify
- 
- Kernel version | `uname -a`
- Unable to verify
- 
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Odroid C4
- 
- Power supply used | (EG: 5V 1A RAVpower)
- 12V 2A SMPS
- 
- SD card used | (EG: SanDisk ultra)
- Axxen microSDHC SK10 16GB TLC CLASS10 UHS-3

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Unable to verify
- 
- Was the software title installed freshly or updated/migrated?
- Unable to verify
- 
- Can this issue be replicated on a fresh installation of DietPi?
- Unable to verify
- 

- Bug report ID | `echo $G_HW_UUID`
- Unable to verify

#### Steps to reproduce

1. Download the dietpi image (DATE: Mon Aug 29 17:41:03 UTC 2022)
2. Flash to MicroSD using "valenaEtcher-Portable-1.7.9"
3. Install MicroSD to odriod c4
4. Connect the HDMI terminal
5. Connect the 12 volt power supply

#### Expected behaviour

1. The kernel must be loaded and the device must operate normally

#### Actual behaviour

1. The bootloader and kernel are not working and the entire device is dead

#### Extra details

In order to remove the existing Debian buster version and install the bullseye version, I formatted the MicroSD card and installed it using balena Etcher
And turn the power to remove all connectors, except the power and hdmi odroid c 4 does not do anything and still.
Odroid wiki in Ubuntu use images to operate normally by installing, using a etcher
Please help me with the installation of odroid c4 dietpi

MichaIng, 2022-10-07:
Has the SPI flash been erased? AFAIK there is a button at the downside of the device to force booting via MMC bootloader. Can you test this?

Power-onoff, 2022-10-07:
As far as I know, Odroid C4 doesn't have a SPI flash

Board Layout : https://wiki.odroid.com/odroid-c4/hardware/hardware#board_layout

https://forum.odroid.com/viewtopic.php?p=297857&sid=4a77b65d43f28f2b978ed55c8c9e6c15#p297857

MichaIng, 2022-10-07:
Ah, seems like only the HC4 has it. I didn't know that. Could you test the Armbian Bullseye image? Should be identical regarding bootloader and kernel: https://www.armbian.com/odroid-c4/

Power-onoff, 2022-10-07:
I'll check and let you know right away

Power-onoff, 2022-10-07:
Verified that there is no problem booting the Ambian Bullseye image

MichaIng, 2022-10-07:
I just compared again both, they should be 100% identical regarding anything relevant for early boot stage. However, just yesterday I moved new builds in place, which contain a more recent kernel and bootloader (also newer than on Armbian image). Can you please test that one?

Power-onoff, 2022-10-07:
Yes, you have a USB-UART adapter.

I'm ready for the test.

MichaIng, 2022-10-23:
Just attach it when trying to boot so you see the early boot logs which should show where and why it fails.

Power-onoff, 2022-11-06:
> Just attach it when trying to boot so you see the early boot logs which should show where and why it fails.

SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 241406

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==106 ps 9
R0_TxDqDly_Margi==106 ps 9

R1_RxClkDly_Margin==0 ps 0
R1_TxDqDly_Margi==0 ps 0

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0001

soc_vref_reg_value 0x 00000050 0000004e 0000004d 0000004e 0000004d 0000004d 0000004b 0000004e 0000004d 0000004f 0000004e 0000004e 0000004d 0000004e 00000051 0000004d 0000004d 0000004e 0000004e 0000004d 0000004d 0000004c 0000004f 0000004f 0000004e 0000004e 0000004d 0000004e 0000004e 0000004e 0000004c 0000004f dram_vref_reg_value 0x 00000020
2D training succeed
aml_ddr_fw_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:43
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00700024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass

non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 435
result report
boot times 0Enable ddr reg access
Load FIP HDR from SD, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from SD, src: 0x00078200, des: 0x01768000, size: 0x0009c000, part: 0
bl2z: ptr: 05129330, size: 00001e40
0.0;M3 CHK:0;cm4_sp_mode 0
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12a_v1.1.3386-3b31431 2019-05-21 10:41:54 luan.yuan@droid15-sz]
OPS=0x10
ring efuse init
2b 0c 10 00 01 04 24 00 00 0d 36 30 43 57 50 50
[0.017355 Inits done]
secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):4fc40b1
NOTICE:  BL31: Built : 15:57:33, May 22 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

<debug_uart>

U-Boot 2022.07-armbian (Oct 18 2022 - 07:35:04 +0000) odroid-c4/hc4

Model: Hardkernel ODROID-C4
SoC:   Amlogic Meson SM1 (S905X3) Revision 2b:c (10:2)
DRAM:  3.8 GiB
Core:  388 devices, 27 uclasses, devicetree: separate
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Board variant: c4
Net:   eth0: ethernet@ff3f0000
Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
MMC Device 2 not found
no mmc device at slot 2
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
ethernet@ff3f0000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
Could not initialize PHY ethernet@ff3f0000
missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/01-00-1e-06-48-0f-ab
ethernet@ff3f0000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
Could not initialize PHY ethernet@ff3f0000
Retrieving file: pxelinux.cfg/00000000
ethernet@ff3f0000 Waiting for PHY auto negotiation to complete......... TIMEOUT !

MichaIng, 2022-11-07:
U-Boot loads correctly but then it seems to try reading from eMMC or fails to read from SD card (from where it loaded U-Boot successfully before), not sure:
```
Card did not respond to voltage select! : -110
MMC Device 2 not found
```
Is there an eMMC module attached? If so, could you try to detach it and then boot?

Power-onoff, 2022-11-11:
> U-Boot loads correctly but then it seems to try reading from eMMC or fails to read from SD card (from where it loaded U-Boot successfully before), not sure:
>
> ```
> Card did not respond to voltage [...]

The result I mentioned is the result of the test with the eMMC module removed, that is, only the sdcard is connected

MichaIng, 2022-11-12:
Hmm, currently there is no general boot issue reported with Odroid C4, but only that warn `reboot` does not work with latest U-Boot, an issue plaguing many of the meson64 boards with Armbian kernel. I'll check whether we can test an alternative U-Boot.

usual-user, 2022-11-13:
I just attached my 2022.10 GA odroid-c4/hc4 firmware composition here. The mainline u-boot configuration seems to be derived from the n2 and also applies to the hc4.`CONFIG_IDENT_STRING=" odroid-c4/hc4`The TF-A components are selected according to the odroid-c4. Because I don't have a suitable device available, I can't check the functionality myself. So just run the "blank microSD card with firmware only" attempt and see what happens.
[u-boot-meson.zip](https://github.com/MichaIng/DietPi/files/9996975/u-boot-meson.zip)

MichaIng, 2022-11-13:
Many thanks for providing this, let's see whether it boots more reliable than the current Armbian U-Boot build. I can imagine that it's similar to the Odroid N2 boot issues with some SD card and eMMC module models.

MichaIng, 2022-12-04:
Since there was a U-Boot update in the meantime, @Power-onoff could you try [this](https://dietpi.com/downloads/images/testing/) image?

If it still doesn't work for you, I'll generate one with usual-user's U-Boot build provided above. This worked well for Odroid N2 already.

Power-onoff, 2022-12-05:
> Since there was a U-Boot update in the meantime, @Power-onoff could you try [this](https://dietpi.com/downloads/images/testing/) image?
> 
> If it still doesn't work for you, I'll generate one with usual-user's U-Boot build provided above. This worked well for Odroid N2 already.

Sadly, the boot still failed.

```
Retrieving file: pxelinux.cfg/default-arm-meson-odroid-n2
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (76 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (202 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 250580

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
[... 131 lines cut ...]
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (77 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (202 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>
=>
=>
```

Power-onoff, 2022-12-06:
> New images are ready: https://dietpi.com/downloads/images/testing/

I flashed it using balenaEtcher
But it still failed to boot
Did I make a mistake or miss anything?

DietPi_OdroidC4-ARMv8-Bullseye.img
```
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 256758

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

[... 269 lines cut ...]
Retrieving file: pxelinux.cfg/default-arm-meson-odroid-n2
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (81 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (189 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>
```

MichaIng, 2022-12-06:
Btw, no need to test the Bookworm image, if the Bullseye image does not boot. They share the exact same bootloader and kernel 😉.

Is it right that you have 3 USB devices attached? If so, does it make a difference when you detach all USB devices (apart from the UART adapter)?

Let's wait for @usual-user, whether the bootloader was flashed correctly, or whether first stage bootloader needs to be flashed. However, second stage is reached, based on output, I'd say.

Power-onoff, 2022-12-07:
> Is it right that you have 3 USB devices attached? If so, does it make a difference when you detach all USB devices (apart from the UART adapter)?

All tests were conducted with only power, LAN, and UART connected
I didn't connect any USB

usual-user, 2022-12-07:
First of all, thank you very much for confirming that my firmware composition
 works. I am pleased that it works right away because after all I built the
 firmware only according to theoretical assumptions and on-target with a
 completely different tool-chain. And can't even verify the result myself.
The logs indicate that everything is working as expected.
> Let's wait for @usual-user, whether the bootloader was [flashed correctly](https://github.com/MichaIng/DietPi/issues/5782#issuecomment-1339540019), or whether first stage bootloader needs to be flashed. However, second stage is reached, based on output, I'd say.

The firmware consists of several components than just the payload (u-boot). U-boot (BL33) is only the last component in the boot chain. The other boot artifacts (BL2, BL3X) are only available by Amlogic as binary-only. Therefore, all FIPs consist of the same TF-A components, which is also recognizable by the identical log messages up to the u-boot version message.
Regarding the way the firmware is brought into place, I can only say that my firmware composition (FIP) only provides what the MASKROM code of the SOC requires and is placed at the entry point that the MASKROM code uses. But because now the TF-A artifacts are identical, and thus also their functionality, I can not say which code Armbian places in the MBR. However, it is not something that the MASKROM code of the SOC, TF-A or u-boot requires. Probably a legacy code artifact.
But now to the real issue:
> Card did not respond to voltage select! : -110

From the specification reference from the second post by @Power-onoff, I see:
> Storage<----->1x Micro SD slot (DS/HS mode up to UHS-I SDR104)

From the initial post by @Power-onoff, I see:
> Axxen microSDHC SK10 16GB TLC CLASS10 UHS-3

Here is my analysis:
I don't know if there is a hardware limitation for the maximum UHS-I class or if only the necessary software support is missing, but the error message reflects exactly the facts. The fact that the firmware can be loaded from the card is due to the fact that the MASKROM uses the lowest common mode to use the greatest possible compatibility. Subsequent firmware may then try to use higher modes, which may not work. I would therefore recommend trying a maximum UHS-I class microSD card.

Power-onoff, 2022-12-07:
tested on the Silicon Power 64GB MicroSDXC UHS-1 Class10 microSD card and confirmed that the login screen appears after booting.
<details>
<summary>more / less</summary>
<div markdown="1">

```
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 765032

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==106 ps 9
R0_TxDqDly_Margi==106 ps 9

R1_RxClkDly_Margin==0 ps 0
R1_TxDqDly_Margi==0 ps 0

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0001

soc_vref_reg_value 0x 00000050 0000004e 0000004d 0000004e 0000004d 0000004d 0000004b 0000004d 0000004d 0000004f 0000004d 0000004d 0000004d 0000004e 00000050 0000004d 0000004d 0000004e 0000004f 0000004e 0000004e 0000004d 0000004f 0000004f 0000004e 0000004e 0000004e 0000004e 0000004e 0000004f 0000004d 00000050 dram_vref_reg_value 0x 00000021
2D training succeed
aml_ddr_fw_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:43
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00700024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass

non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 430
result report
boot times 0Enable ddr reg access
Load FIP HDR from SD, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from SD, src: 0x00078200, des: 0x01768000, size: 0x0009c000, part: 0
bl2z: ptr: 05129330, size: 00001e40
0.0;M3 CHK:0;cm4_sp_mode 0
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12a_v1.1.3386-3b31431 2019-05-21 10:41:54 luan.yuan@droid15-sz]
OPS=0x10
ring efuse init
2b 0c 10 00 01 04 24 00 00 0d 36 30 43 57 50 50
[0.017354 Inits done]
secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):4fc40b1
NOTICE:  BL31: Built : 15:57:33, May 22 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

<debug_uart>

U-Boot 2022.10 (Nov 13 2022 - 00:00:00 +0000) odroid-c4/hc4

Model: Hardkernel ODROID-C4
SoC:   Amlogic Meson SM1 (S905X3) Revision 2b:c (10:2)
DRAM:  3.8 GiB
Core:  387 devices, 27 uclasses, devicetree: separate
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Board variant: c4
Net:   eth0: ethernet@ff3f0000
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 3 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2847 bytes read in 2 ms (1.4 MiB/s)
## Executing script at 08000000
312 bytes read in 2 ms (152.3 KiB/s)
27214336 bytes read in 1156 ms (22.5 MiB/s)
12349432 bytes read in 525 ms (22.4 MiB/s)
74543 bytes read in 7 ms (10.2 MiB/s)
## Loading init Ramdisk from Legacy Image at 13000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    12349368 Bytes = 11.8 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 04080000
   Booting using the fdt blob at 0x4080000
   Loading Ramdisk to 3f439000, end 3fffffb8 ... OK
   Loading Device Tree to 000000003f423000, end 000000003f43832e ... OK

Starting kernel ...

[    5.277697] panfrost ffe40000.gpu: error -ENODEV: dev_pm_opp_set_regulators: no regulator (mali) found

Debian GNU/Linux 11 DietPi ttyAML0

DietPi login:
```
</div>
</details>

MichaIng, 2022-12-07:
So the Odroid C4 is picky on SD cards. Not sure whether this is a hardware limitation or may be solved bootloader wise.

Now it would be interesting whether [old C4 image](https://dietpi.com/downloads/images/) boots on those two UHS-I cards as well, so whether #5959 does bring any benefit or not. Since you are the first to report boot issues with C4 (recently), I guess we can stick with the Armbian build. But good to have an alternative at hand, since it wouldn't be the first time that an Armbian U-Boot release broke booting, or made it less reliable, like recently on NanoPi R1.
________
> I can not say which code Armbian places in the MBR. something that the MASKROM code of the SOC, TF-A or u-boot requires.

The first 446 bytes are traditionally used for the first stage bootloader, but not sure whether it is required for those meson SoCs, which are indeed the only ones where those bytes are flashed. Probably for ancient compatibility indeed.

Btw, I just compared both binaries, and there is a larger identical pattern in their code with a few KiB offset. And the offset of this pattern is exactly 512 bytes smaller in your case. So indeed Armbian's build does contain the MBR part (for whatever reason), while yours does not.

Power-onoff, 2022-12-07:
FILE:   DietPi_OdroidC4-ARMv8-Bullseye.img
DATE:   Sun Nov 20 21:18:55 UTC 2022
MD5:    b04895fe49136b5b172eca4b91fc9bf0
SHA1:   c5ec31de0d164a5d087eee4b544a0c76f6fbfdc5
SHA256: ce7dac9ebffda21891e75d7f70f33fcfed2ece49e4f4ed472969ee93e217a21f

microSD card: Sandisk 16GB MicroSDHC UHS-1 Class10

confirmed that the login screen appears after booting.
<details>
<summary>more / less</summary>
<div markdown="1">

```
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 432815

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==106 ps 9
R0_TxDqDly_Margi==106 ps 9

R1_RxClkDly_Margin==0 ps 0
R1_TxDqDly_Margi==0 ps 0

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0001

soc_vref_reg_value 0x 00000050 0000004e 0000004d 0000004e 0000004e 0000004d 0000004c 0000004e 0000004d 0000004f 0000004d 0000004d 0000004d 0000004e 00000051 0000004d 0000004d 0000004e 0000004f 0000004e 0000004e 0000004d 0000004f 0000004f 0000004f 0000004e 0000004e 0000004e 0000004f 0000004e 0000004d 00000050 dram_vref_reg_value 0x 00000021
2D training succeed
aml_ddr_fw_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:43
auto size-- 65535DDR cs0 size: 2048MB
DDR cs1 size: 2048MB
DMC_DDR_CTRL: 00700024DDR size: 3928MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass

non-sec scramble use zero key
ddr scramble enabled

100bdlr_step_size ps== 435
result report
boot times 0Enable ddr reg access
Load FIP HDR from SD, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from SD, src: 0x00078200, des: 0x01768000, size: 0x0009c000, part: 0
bl2z: ptr: 05129330, size: 00001e40
0.0;M3 CHK:0;cm4_sp_mode 0
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12a_v1.1.3386-3b31431 2019-05-21 10:41:54 luan.yuan@droid15-sz]
OPS=0x10
ring efuse init
2b 0c 10 00 01 04 24 00 00 0d 36 30 43 57 50 50
[0.017320 Inits done]
secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):4fc40b1
NOTICE:  BL31: Built : 15:57:33, May 22 2019
NOTICE:  BL31: G12A normal boot!
NOTICE:  BL31: BL33 decompress pass
ERROR:   Error initializing runtime service opteed_fast

<debug_uart>

U-Boot 2022.07-armbian (Oct 18 2022 - 07:35:04 +0000) odroid-c4/hc4

Model: Hardkernel ODROID-C4
SoC:   Amlogic Meson SM1 (S905X3) Revision 2b:c (10:2)
DRAM:  3.8 GiB
Core:  388 devices, 27 uclasses, devicetree: separate
MMC:   sd@ffe05000: 0, mmc@ffe07000: 1
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Board variant: c4
Net:   eth0: ethernet@ff3f0000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2847 bytes read in 3 ms (926.8 KiB/s)
## Executing script at 08000000
312 bytes read in 2 ms (152.3 KiB/s)
26530304 bytes read in 1151 ms (22 MiB/s)
12084624 bytes read in 523 ms (22 MiB/s)
73415 bytes read in 9 ms (7.8 MiB/s)
## Loading init Ramdisk from Legacy Image at 13000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    12084560 Bytes = 11.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 04080000
   Booting using the fdt blob at 0x4080000
   Loading Ramdisk to 3f479000, end 3ffff550 ... OK
   Loading Device Tree to 000000003f464000, end 000000003f478ec6 ... OK

Starting kernel ...

[    1.214952] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    1.215792] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517
[    1.215896] reg-fixed-voltage regulator-vcc_3v3: Failed to register regulator: -517
[    1.216115] reg-fixed-voltage regulator-vddao_1v8: Failed to register regulator: -517
[    2.003238] thermal thermal_zone0: binding zone cpu-thermal with cdev thermal-cpufreq-0 failed:-17
[    2.040456] reg-fixed-voltage regulator-flash_1v8: Failed to register regulator: -517
[    2.043011] reg-fixed-voltage regulator-vcc_1v8: Failed to register regulator: -517
[    4.743334] panfrost ffe40000.gpu: dev_pm_opp_set_regulators: no regulator (mali) found: -19

Debian GNU/Linux 11 DietPi ttyAML0

DietPi login:
```
</div>
</details>

Power-onoff, 2022-12-07:
One of the things I mentioned was a mistake
Correct UHS-3 to UHS-1(U3)
"Axxen microSDHC SK10 16GB TLC CLASS10 UHS-3" >> "Axxen microSDHC SK10 16GB TLC CLASS10 UHS-1(U3)"

> Here is my analysis:
> I don't know if there is a hardware limitation for the maximum UHS-I class or if only the necessary software support is missing, but the error message reflects exactly the [...]

Power-onoff, 2022-12-07:
FILE: DietPi_OdroidC4-ARMv8-Bullseye.img
DATE: Sun Nov 20 21:18:55 UTC 2022
MD5: b04895fe49136b5b172eca4b91fc9bf0
SHA1: c5ec31de0d164a5d087eee4b544a0c76f6fbfdc5
SHA256: ce7dac9ebffda21891e75d7f70f33fcfed2ece49e4f4ed472969ee93e217a21f

microSD card: Axxen microSDHC SK10 16GB TLC CLASS10 UHS-1(U3)

Strangely, boot fails only on certain microSD cards
<details>
<summary>more / less</summary>
<div markdown="1">

```
SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:1;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:2;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:3;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:4;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:5;EMMC:800;NAND:81;SD?:20000;USB:8;LOOP:6;EMMC:800;NAND:81;SD?:20000;USB:8;SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 231322

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
[... 115 lines cut ...]
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (81 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (178 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>

```
</div>
</details>

MichaIng, 2022-12-07:
Okay, so it fails to boot from Axxen microSDHC SK10 16GB TLC CLASS10 UHS-1(U3) with any image and succeeds to boot from any other SD card you tested, right?

Then for now I don't see a reason to switch U-Boot, but good to have it in mind and available, in case we find cases where one boots but the other not.

MichaIng, 2022-12-07:
Many thanks for all your tests btw! Since I don't have a C4 myself, this is very valueable!

Power-onoff, 2022-12-07:
FILE: DietPi_OdroidC4-ARMv8-Bullseye.img
DATE: Sun Nov 20 21:18:55 UTC 2022
MD5: b04895fe49136b5b172eca4b91fc9bf0
SHA1: c5ec31de0d164a5d087eee4b544a0c76f6fbfdc5
SHA256: ce7dac9ebffda21891e75d7f70f33fcfed2ece49e4f4ed472969ee93e217a21f

microSD card: Axxen microSDHC SK10 16GB TLC CLASS10 UHS-1(U3)

Executed a command in the dietpi virtualbox
dd bs=512 seek=1 conv=notrunc,fsync if=u-boot-meson.bin of=/dev/sdc

SM1:BL:511f6b:81ca2f;FEAT:A0F83180:20282000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.0;CHK:0;
bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:00008067
L3:15000020
S1:00000000
B2:20282000
B1:a0f83180

TE: 255998

BL2 Built : 20:29:41, Jun 18 2019. g12a ga659aac - luan.yuan@droid15-sz

Board ID = 1
Set cpu clk to 24M
Set clk81 to 24M
Use GP1_pll as DSU clk.
DSU clk: 1200 Mhz
CPU clk: 1200 MHz
Set clk81 to 166.6M
DDR driver_vesion: LPDDR4_PHY_V_0_1_15 build time: Jun 18 2019 20:29:37
board id: 1
Load FIP HDR from SD, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from SD, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from SD, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from SD, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0001
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from SD, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
[... 108 lines cut ...]
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-meson-odroid-n2
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm-meson
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default-arm
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Retrieving file: pxelinux.cfg/default
Speed: 1000, full duplex
*** ERROR: `serverip' not set
Config file not found
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (103 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
Speed: 1000, full duplex
BOOTP broadcast 1
DHCP client bound to address 192.168.0.5 (188 ms)
*** ERROR: `serverip' not set
Cannot autoload with TFTPGET
=>

Power-onoff, 2022-12-07:
> The images which succeed are all with the same vendor bootloader. So it's a limitation of the mainline U-Boot. As always, sad that manufacturers do not commit their necessary bootloader (same with kernel) support upstream in the first place, but keep maintaining an own fork, usually without ever updating (just patching) it 😞.
> 
> In the meantime, Meveric switched to mainline kernel and U-Boot as well. I guess his newer C4 image fail the same way: https://oph.mdrjr.net/meveric/images/Bullseye/

Tested the image file "Debian-Bullsey64-1.0-20221003-C4.img.xz" and successfully booted.
<details>
<summary>more / less</summary>
<div markdown="1">

```
[    2.867738@0] ffd24000.serial: ttyS1 at MMIO 0xffd24000 (irq = 42, base_baud = 1500000) is a meson_uart
[    2.878584@0] amlogic-new-usb2-v2 ffe09000.usb2phy: USB2 phy probe:phy_mem:0xffe09000, iomap phy_base:0xffffff80084fd000
[    2.888500@0] amlogic-new-usb3-v2 ffe09080.usb3phy: USB3 phy probe:phy_mem:0xffe09080, iomap phy_base:0xffffff800862b080
[    2.900268@0] aml_dma ff63e000.aml_dma: Aml dma
[    2.905497@0] aml_aes_dma ff63e000.aml_dma:aml_aes: Aml AES_dma
[    2.912183@0] aml_sha_dma ff63e000.aml_dma:aml_sha: Aml SHA1/SHA224/SHA256 dma
[    2.916524@0] gpio-keypad: probe of ff800000.gpio_keypad failed with error -22
[    2.923819@0] efusekeynum: 1
[    2.926394@0] efusekeyname:            uuid  offset:     0   size:    32
[    2.933422@0] efuse efuse: probe OK!
[    2.942472@0] ion_dev soc:ion_dev: assigned reserved memory node linux,ion-dev
[    2.945832@0] ge2d: ge2d_init_module
[    2.947816@0] ge2d: ge2d_dev major:236
[    2.951551@0] ge2d: clock source clk_ge2d_gate ffffffc0ca13aa80
[    2.957259@0] ge2d: clock clk_ge2d source ffffffc0ca13a3c0
[    2.962711@0] ge2d: clock source clk_vapb_0 ffffffc0ca13a0c0
[    2.968326@0] ge2d: ge2d init clock is 500000000 HZ, VPU clock is 666666656 HZ
[    2.975619@0] ge2d: ge2d clock is 499 MHZ
[    2.979512@0] ge2d: find address resource
[    2.983504@0] ge2d: map io source 0x00000000ff940000,size=65536 to 0xffffff8008690000
[    2.991291@0] ge2d: reserved mem init failed
[    2.995542@0] ge2d: ge2d: pdev=ffffffc0ca1c1800, irq=47, clk=ffffffc0ca13aa80
[    3.002675@0] ge2d: ge2d start monitor
[    3.007249@0] [tsync_pcr_init]init success.
[    3.007413@1] ge2d: ge2d workqueue monitor start
[    3.015303@0] amvideom vsync irq: 48
[    3.018702@0] create_ge2d_work_queue video task ok
[    3.023937@0] fb: osd_init_module
[    3.027114@0] fb: viu vsync irq: 48
[    3.030201@0] fb: viu2 vsync irq: 64
[    3.033888@0] 0x000000db:Y=db,U=0,V=0
[    3.037399@0] 0x000000dc:Y=dc,U=0,V=0
[    3.041036@0] 0x000000dd:Y=dd,U=0,V=0
[    3.044675@0] 0x000000de:Y=de,U=0,V=0
[    3.048348@0] 0x000000df:Y=df,U=0,V=0
[    3.051957@0] 0x000000e0:Y=e0,U=0,V=0
[    3.055890@0] fb: osd_rdma_init: rdma_table p=0xcf608000,op=0xcf608000 , v=0xffffff8008637000
[    3.064123@0] rdma_register, rdma_table_addr ffffff8008639000 rdma_table_addr_phy cf609000 reg_buf ffffffc0ca039000
[    3.074529@0] rdma_register success, handle 2 table_size 4096
[    3.080213@0] fb: osd_rdma_init:osd rdma handle = 2.
[    3.085180@0] fb: mem_size: 0x800000
[    3.088705@0] fb: mem_size: 0x4b80000
[    3.092342@0] fb: mem_size: 0x100000
[    3.095895@0] fb: mem_size: 0x100000
[    3.099478@0] fb: mem_size: 0x800000
[    3.103007@0] fb: failed to init reserved memory
[    3.107609@0] fb: fb def : 1920 1080 1920 2160 32
[    3.112277@0] fb: init fbdev bpp is:32
[    3.116504@0] fb: malloc_osd_memory, cma:ffffff800a8aeb58
[    3.116506@0] fb: malloc_osd_memory, 1231, base:0x00000000cf400000, size:8388608
[    3.116523@0] fb: use ion buffer for fb memory, fb_index=0
[    3.116525@0] fb: OSD0 as afbcd mode,afbc_type=2
[    3.137104@0] meson-fb meson-fb: create ion_client ffffffc059eb9540, handle=ffffffc059e08e40
[    3.137109@0] meson-fb meson-fb: ion memory(0): created fb at 0x00000000e5800000, size 75 MiB
[    3.137112@0] fb:  0, phy: 0x00000000e5800000, vir:0xffffff800a95a000, size=77312K
[    3.137112@0]
[    3.137116@0] fb: Frame buffer memory assigned at
[    3.137117@0] fb:  0, phy: 0x00000000e5800000, vir:0xffffff800a95a000, size=77312K
[    3.137117@0]
[    3.137118@0] fb:  0, phy: 0x00000000e5800000, vir:0xffffff800a95a000, size=77312K
[... 1728 lines cut ...]
         Starting Permit User Sessions...
[   12.958814] rc.local[2144]: setterm: terminal vt220 does not support --blank
[   12.972549] rc.local[2144]: setterm: cannot (un)set powersave mode: Inappropriate ioctl for device
[  OK  ] Started Permit User Sessions.
[   12.992677] rc.local[2144]: resize2fs 1.44.5 (15-Dec-2018)
[   13.091671@2] EXT4-fs (mmcblk1p2): resizing filesystem from 196608 to 3825405 blocks
[  OK  ] Started OpenBSD Secure Shell server.
[   13.392375@0] EXT4-fs (mmcblk1p2): resized filesystem to 3825405
[   13.551625] rc.local[2144]: Filesystem at /dev/mmcblk1p2 is mounted on /; on-line resizing required
[   13.568362] rc.local[2144]: old_desc_blocks = 1, new_desc_blocks = 1
[   13.584312] [rc.local  OK  [2144]: ] The filesystem on /dev/mmcblk1p2 is now 3825405 (4k) blocks long.Started /etc/rc.local Compatibility.

[  OK  ] Started Serial Getty on ttyS0.
[  OK  ] Started Getty on tty1.
[  OK  ] Reached target Login Prompts.
[   18.642878@1] meson_uart ff803000.serial: ttyS0 use xtal(24M) 24000000 change 115200 to 115200

Debian GNU/Linux 10 odroid-buster64 ttyS0

eth0: 192.168.0.5
odroid-buster64 login: [   64.522081@2] fb: mem_free_work, free memory: addr:800000

```
</div>
</details>

MichaIng, 2022-12-07:
> U-Boot 2015.01 (Aug 07 2020 - 08:55:43)

Ah, only mainline kernel but vendor U-Boot it seems.

usual-user, 2022-12-07:
OK, had found [this](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/arch/arm64/boot/dts/amlogic/meson-sm1-odroid.dtsi?id=323ca765bfe9d637fa774373baec0bc41e51fcfa) fix that was applied to mainline DT but has not yet been incorporated into mainline u-boot. Yes, I dubblechecked that it's really included in my second test build, but it doesn't seem to make any difference. An incorrectly wired IO line for regulator control would have been a good explanation as the cause of the error. Unfortunately, I can't really help under these debugging conditions (no corresponding device for locale investigations) any further. So we have to wait until a mainline solution becomes available. Of course I can then provide a new firmware build as an early adoption.

MichaIng, 2022-12-10:
@usual-user 
Many thanks for all your help and builds. Sooner or later I will have to setup U-Boot and kernel build workflows as well, I guess.

I will ask Hardkernel for developer samples. Not sure if they will still send a C4, but probably an M1 or future SBC ones in pipeline.

Until then or generally, we should bring this up in Odroid forum: https://forum.odroid.com/viewforum.php?f=203

usual-user, 2022-12-11:
> Many thanks for all your help and builds.

You're welcome. It's also a bit self-serving, it confirms to me that fedora also works out-of-the-box on devices I don't own, as long as there's a firmware that can boot the operating system. And fedora is not even designed for a specific device, but only generically compiled for the aarch64 architecture. So one image fits everyone.
> I will ask Hardkernel for developer samples. Not sure if they will still send a C4, but probably an M1 or future SBC ones in pipeline.

This is not a good choice. The HC4 is better, because as I noticed in the meantime, there is a separate `odroid-hc4_defconfig` with additional storage media enabled. The C4 u-boot should also work on the HC4, but cannot read the boot configuration from the additional storage media offered by the HC4. However, this makes the HC4 suitable for checking the function of both firmwares.
The M1 is currently only recommended for developers who want to contribute to mainline support. With kernel 6.2.0-rc1 all code available for the M1 will be available out-of-the-box. Further support code has yet to be written. Userspace support is also available with current mainline releases. The only exception is FFmpeg, where v4l2-request support is still missing. But it's not too bad at the moment, because the rkvdec2 kernel driver is still missing.
All in all, the M1 is already very well supported out-of-the-box with a current mainline operating system. Some older devices still need a lot of out-of-tree patches for equivalent support.
But when it comes to u-boot, so far only the legacy BSP fork from 2015 is available.
TF-A [support](https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/16952) is progressing slowly. And for u-boot no mainline code has been posted yet. There are only a few incomplete out-of-tree hacks for other rk3568 devices.
I'm using an Armbian build as boot trampoline, but as the BSP version gets probably no development and feature backports, no reason to build it myself. So it will take some time, but something is always ;-)
> Until then or generally, we should bring this up in Odroid forum

In my experience, Hardkernel doesn't have much interest in mainline contributions and only supports lagacy outdated BSP implementations. But maybe constant dripping hollows the stone?
Pine64 does a much better job, at least they document the [current mainline support](https://wiki.pine64.org/wiki/Quartz64_Development) very carefully and I am very grateful to the developers for contributing their code to mainline. Without their work, the support of the Odroid-M1 by simply adding an M1 board DTS would not have been possible. And this was not even led by Hardkernel, but a nice example of what little effort (board DTS) is necessary to support a new device if there is sufficient mainline SOC support.

usual-user, 2022-12-11:
> The downside of HC4 is that it's headless and hence media features cannot be tested.

If I don't misread the [specification](https://wiki.odroid.com/odroid-hc4/hardware/hardware), the HC4 is also equipped with an HDMI port, so what are you missing?
> Exactly that M1 isn't so well supported yet is a good reason to have the board to test images with while finding proper builds and configs, even without doing own bootloader and kernel builds.

The bigger problem for Debian based systems is the use of outdated software versions for "stability reasons". The aarch64 SOC architecture is bleeding edge development and cannot be efficiently driven by software designed for the x86 IBM architecture.
Users must be willing to learn new things and use others as the familiar ones from the x86 architecture. But who can teach old dogs new tricks. Just look at how desperately they try to emulate a x86 boot schema, although better is possible. For example, there's no need for a boot partion or a /boot directory, this is a x86 BIOS constraint. But they seem to love how it breaks with conflicting file locations every now and then. Or how they want to use hardware video acceleration via the GPU. A WEB browser designed with this in mind will never work efficiently on a SOC architecture with separate VPUs. Yes SOCs can have several VPUs. The rk356x currently lacks rkvdec2 kernel support, but when it becomes available, it will be on par with many other SOCs and ready for the media desktop. But everything else is already possible today with an up-to-date desktop environment.

MichaIng, 2022-12-12:
Oh you're right, the HC4 has HDMI, how could I miss that.

While with `x86_64`, things are indeed still easier IMO in the meantime, `aarch64` has established quite well and is taken into account by most software developers. About e.g. GPU usage efficiency I cannot say something, but on modern mid- to high class SBCs Kodi and Chromium alike works quite well and generally an ARM server is overall more efficient compared to an `x86_64` machine. However focus of DietPi is anyway a little more on server applications than on GUI applications.