MichaIng/DietPi #5784: Error running dietpi-imager

ger3295, 2022-10-04:
Hello, I am trying to make a smaller image of a 32 Gb SD Card where I have Dietpi Buster installed (I have a python script running with that version that gives me features that the most current ones don't allow me, and it works correctly). This Dietpi runs on Raspberry Pi Zero W.

But when I run the dietpi-imager script, an error occurs. I copy the information below.

Could you help me to know what I am doing wrong? The script should I run it on the same hardware that will have the target memory (in this case Raspberry Pi Zero W)?

Thank you!

#### Details:
- Date           | Tue Oct  4 14:22:33 -03 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPi 5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt`
- Exit code      | 1
- Software title | DietPi-Imager
#### Steps to reproduce:

1. Execute: bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
2. Please select how the input OS is stored:  DRIVE
3. Please select the drive you wish to create the image from: /dev/sdb : 29.7G Mass-Storage Generic_Mass-Storage-0:0
4. Please select the OS root partition: /dev/sdb2 : 29.5G ext4
5. Please select which cloning tool to use: dd
6. Please enter a name for the new image: name
7. Start

- ...
#### Additional logs:
When dietpi-imager starts:

[  OK  ] DietPi-Imager | cd /
bash: line 71: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
bash: line 72: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Reading package lists...
[  OK  ] DietPi-Imager | APT update
[ INFO ] DietPi-Imager | APT install for: zerofree, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  zerofree
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 8432 B of archives.
After this operation, 25.6 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main amd64 zerofree amd64 1.1.1-1 [8432 B]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 8432 B in 0s (27.1 kB/s)
                                Selecting previously unselected package zerofree.
(Reading database ... 69119 files and directories currently installed.)
Preparing to unpack .../zerofree_1.1.1-1_amd64.deb ...
Unpacking zerofree (1.1.1-1) ...
Setting up zerofree (1.1.1-1) ...
[  OK  ] DietPi-Imager | APT install for: zerofree
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sdb ...
[ INFO ] DietPi-Imager | MBR partition table detected
[ INFO ] DietPi-Imager | e2fsck -fp /dev/sdb2, please wait...
rootfs: Inode 37628 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: 40454/320784 files (1.2% non-contiguous), 1006681/1325824 blocks
[  OK  ] DietPi-Imager | e2fsck -fp /dev/sdb2
[  OK  ] DietPi-Imager | mkdir /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | mount /dev/sdb2 /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | rm -f /tmp/DietPi-Imager_mnt/root/.bash_history /tmp/DietPi-Imager_mnt/home/*/.bash_history /tmp/DietPi-Imager_mnt/var/lib/dhcp/*.leases
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | partprobe /dev/sdb
[  OK  ] DietPi-Imager | partx -u /dev/sdb
[ INFO ] DietPi-Imager | e2fsck -fp /dev/sdb2, please wait...
rootfs: Inode 37628 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: 40454/320784 files (1.2% non-contiguous), 1006681/1325824 blocks
[  OK  ] DietPi-Imager | e2fsck -fp /dev/sdb2
[  OK  ] DietPi-Imager | tune2fs -O ^has_journal /dev/sdb2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | tune2fs -O has_journal /dev/sdb2
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[ INFO ] DietPi-Imager | Shrinking last filesystem to minimum size...
resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sdb2 to 1324802 (4k) blocks.
The filesystem on /dev/sdb2 is now 1324802 (4k) blocks long.

resize2fs 1.46.2 (28-Feb-2021)
Resizing the filesystem on /dev/sdb2 to 1324800 (4k) blocks.
The filesystem on /dev/sdb2 is now 1324800 (4k) blocks long.

resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 1324800 (4k) blocks long.  Nothing to do!

[  OK  ] DietPi-Imager | Leaving last filesystem size at 5180 MiB
[  OK  ] DietPi-Imager | resize2fs /dev/sdb2 1325824
[ INFO ] DietPi-Imager | e2fsck -fp /dev/sdb2, please wait...
rootfs: Inode 37628 extent tree (at level 1) could be narrower.  IGNORED.
rootfs: 40454/320784 files (1.2% non-contiguous), 1006681/1325824 blocks
[  OK  ] DietPi-Imager | e2fsck -fp /dev/sdb2
[ INFO ] DietPi-Imager | Overriding filesystems free space with zeros to purge removed data and allow better compression...
[  OK  ] DietPi-Imager | mount /dev/sdb1 /tmp/DietPi-Imager_mnt
[ INFO ] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt, please wait...
**[FAILED]** DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt

**[FAILED]** DietPi-Imager | Unable to continue, DietPi-Imager will now terminate.
[  OK  ] DietPi-Imager | sync
[  OK  ] DietPi-Imager | sleep 1
[  OK  ] DietPi-Imager | umount -R /tmp/DietPi-Imager_mnt
[  OK  ] DietPi-Imager | rmdir /tmp/DietPi-Imager_mnt

root@DietPi:/# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
[  OK  ] DietPi-Imager | cd /
bash: line 71: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
bash: line 72: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sdb ...
root@DietPi:/# bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager')"
[  OK  ] DietPi-Imager | cd /
bash: line 71: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
bash: line 72: ((: == 1 : syntax error: operand expected (error token is "== 1 ")
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sdb ...

### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
Bullseye

- Kernel version | `uname -a`
Linux DietPi 5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
SBC where 32Gb is installed: Raspberry Pi Zero W
PC where i run dietpi-imager: Native PC (x86_64)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2A

- SD card used | (EG: SanDisk ultra)
SanDisk Ultra 32Gb

MichaIng, 2022-10-04:
Many thanks for your report.

What kind of drive is it? Looks like it does not support fstrim and the newly added `quiet-unsupported` does not prevent the command from failing.

Can you mount the drive to `/mnt/sdb1` and show the output of:
```sh
fstrim -v /mnt/sdb1
```
This should have been part of the error prompt and bug report template output as well.

The two syntax errors are not related and visual only. I'll fix them.

MichaIng, 2022-11-09:
Worked around with: https://github.com/MichaIng/DietPi/commit/8e11fc4
And the error messages: https://github.com/MichaIng/DietPi/commit/1a34ccf

closed 2022-11-09 by MichaIng