MichaIng/DietPi #5786: HAproxy install fails, "you have held broken packages"

tkrusterholz, 2022-10-05:
#### Details:
- Date           | Wed Oct  5 00:15:14 CDT 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux pi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -y --allow-change-held-packages install make gcc libpcre3-dev libssl-dev zlib1g-dev libsystemd-dev`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. run dietpi-software
2. select haproxy
3. execute
#### Expected behaviour:

- dietpi magic and the software is installed
#### Actual behaviour:

- install fails with error shown below
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libpcre3-dev : Depends: libpcre3 (= 2:8.39-12) but 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad is to be installed
                Depends: libpcrecpp0v5 (= 2:8.39-12) but 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad is to be installed
E: Unable to correct problems, you have held broken packages.

Joulinar, 2022-10-05:
Quite interesting, you are running Debian Buster but somehow your system is trying to install Debian Stretch packages. 

```
Depends: libpcre3 (= 2:8.39-12) but 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad is to be installed
Depends: libpcrecpp0v5 (= 2:8.39-12) but 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad is to be installed
```

Even more strange, the version number 2:8.44 is higher as the one shown on Debian package overview https://packages.debian.org/search?suite=all&searchon=names&keywords=libpcre3

You might have a source configured that is offering other packages. Can you show following

```
apt update
apt show libpcre3
```

MichaIng, 2022-10-05:
Looks like remains from the PHP repository used on Stretch systems. So those packages need to be downgraded to match the regular Buster repo.

Joulinar, 2022-10-05:
Removing the PHP repository should be sufficient as the packages are not yet installed.

MichaIng, 2022-10-05:
I think they are installed, since otherwise `libpcre3-dev` would be pulled from the PHP repo as well. But let's see what your suggested commands output is.

tkrusterholz, 2022-10-05:
> Quite interesting, you are running Debian Buster but somehow your system is trying to install Debian Stretch packages.
>
> ```
> Depends: libpcre3 (= 2:8.39-12) but [...]

Here's the output of the commands, the last one said there was more to be seen with the -a switch so I included that as well:

root@pi:~# apt update
Hit:1 http://ftp.nl.debian.org/debian buster InRelease
Get:2 http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian buster InRelease [15.0 kB]
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Fetched 15.0 kB in 2s (8,238 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
root@pi:~# apt show libpcre3
Package: libpcre3
Version: 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad
Status: install ok installed
Priority: important
Section: libs
Source: pcre3
Maintainer: Perl 5 Compatible Regular Expression Library <pcre3@packages.debian.org>
Installed-Size: 544 kB
Depends: libc6 (>= 2.4)
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: Perl 5 Compatible Regular Expression Library - runtime files
 This is a library of functions to support regular expressions whose syntax
 and semantics are as close as possible to those of the Perl 5 language.
 .
 This package contains the runtime libraries.

N: There are 2 additional records. Please use the '-a' switch to see them.
root@pi:~# apt show -a libpcre3
Package: libpcre3
Version: 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad
Status: install ok installed
Priority: important
Section: libs
Source: pcre3
Maintainer: Perl 5 Compatible Regular Expression Library <pcre3@packages.debian.org>
Installed-Size: 544 kB
Depends: libc6 (>= 2.4)
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: Perl 5 Compatible Regular Expression Library - runtime files
 This is a library of functions to support regular expressions whose syntax
 and semantics are as close as possible to those of the Perl 5 language.
 .
 This package contains the runtime libraries.

Package: libpcre3
Version: 2:8.39-12
Priority: important
Section: libs
Source: pcre3
Maintainer: Matthew Vernon <matthew@debian.org>
Installed-Size: 636 kB
Depends: libc6 (>= 2.4)
Conflicts: libpcre3-dev (<= 4.3-3)
Breaks: approx (<< 4.4-1~), cduce (<< 0.5.3-2~), cmigrep (<< 1.5-7~), galax (<< 1.1-7~), libpcre-ocaml (<< 6.0.1~), liquidsoap (<< 0.9.2-3~), ocsigen (<< 1.3.3-1~)
Download-Size: 313 kB
APT-Sources: http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian buster/main armhf Packages
Description: Old Perl 5 Compatible Regular Expression Library - runtime files
 This is a library of functions to support regular expressions whose syntax
 and semantics are as close as possible to those of the Perl 5 language.
 .
 New packages should use the newer pcre2 packages, and existing
 packages should migrate to pcre2.
 .
 This package contains the runtime libraries.

Package: libpcre3
Version: 2:8.39-12
Priority: optional
Section: libs
Source: pcre3
Maintainer: Matthew Vernon <matthew@debian.org>
Installed-Size: 525 kB
Depends: libc6 (>= 2.4)
Conflicts: libpcre3-dev (<= 4.3-3)
Breaks: approx (<< 4.4-1~), cduce (<< 0.5.3-2~), cmigrep (<< 1.5-7~), galax (<< 1.1-7~), libpcre-ocaml (<< 6.0.1~), liquidsoap (<< 0.9.2-3~), ocsigen (<< 1.3.3-1~)
Tag: role::shared-lib
Download-Size: 317 kB
APT-Sources: http://ftp.nl.debian.org/debian buster/main armhf Packages
Description: Old Perl 5 Compatible Regular Expression Library - runtime files
 This is a library of functions to support regular expressions whose syntax
 and semantics are as close as possible to those of the Perl 5 language.
 .
 New packages should use the newer pcre2 packages, and existing
 packages should migrate to pcre2.
 .
 This package contains the runtime libraries.

root@pi:~#

Joulinar, 2022-10-06:
@MichaIng was correct, the package is already installed

```
Package: libpcre3
Version: 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad
Status: install ok installed
```

let's check what is still left over from Debian Stretch

```
dpkg -l | grep debian9
```

tkrusterholz, 2022-10-10:
> let's check what is still left over from Debian Stretch

root@pi:~# dpkg -l | grep debian9
ii  idn2                                 2.2.0-2+0~20200302.4+debian9~1.gbpf85c2e        armhf        Internationalized domain names (IDNA2008/TR46) command line tool
ii  libargon2-1:armhf                    0~20190702-0.1+0~20190710.3+debian9~1.gbp2fb167 armhf        memory-hard hashing function - runtime library
ii  libgd3:armhf                         2.2.5-5.2+0~20190808.4+debian9~1.gbp6d9343      armhf        GD Graphics Library
ii  libicu65:armhf                       65.1-1+0~20200223.8+debian9~1.gbp519cf3         armhf        International Components for Unicode
ii  libidn2-0:armhf                      2.2.0-2+0~20200302.4+debian9~1.gbpf85c2e        armhf        Internationalized domain names (IDNA2008/TR46) library
ii  libpcre2-8-0:armhf                   10.39-2+0~20211122.14+debian9~1.gbp0d570b       armhf        New Perl Compatible Regular Expression Library- 8 bit runtime files
ii  libpcre3:armhf                       2:8.44-2+0~20210301.9+debian9~1.gbpa278ad       armhf        Perl 5 Compatible Regular Expression Library - runtime files
ii  libpcrecpp0v5:armhf                  2:8.44-2+0~20210301.9+debian9~1.gbpa278ad       armhf        Perl 5 Compatible Regular Expression Library - C++ runtime files
ii  libsodium23:armhf                    1.0.18-1+0~20191009.1+debian9~1.gbpb6823f       armhf        Network communication, cryptography and signaturing library
ii  libxml2:armhf                        2.9.12+dfsg-0+0~20210621.9+debian9~1.gbp43e861  armhf        GNOME XML library
ii  libzip4:armhf                        1.7.3-1+0~20210114.10+debian9~1.gbp4c125d       armhf        library for reading, creating, and modifying zip archives (runtime)

MichaIng, 2022-10-10:
Okay, let's do it like this:
```sh
dietpi-services stop
mapfile -t packages < <(dpkg -l | mawk '/debian9/{print $2}')
dpkg -r --force-depends "${packages[@]}"
apt -y install "${packages[@]}"
unset -v packages
dietpi-services start
```

tkrusterholz, 2022-10-10:
Hmm:

```
root@pi:~# dietpi-services stop

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
root@pi:~# mapfile -t packages < <(dpkg -l | mawk '/debian9/{print $2}')
root@pi:~# dpkg -r --force-depends "${packages[@]}"
(Reading database ... 40729 files and directories currently installed.)
Removing idn2 (2.2.0-2+0~20200302.4+debian9~1.gbpf85c2e) ...
dpkg: libpcre2-8-0:armhf: dependency problems, but removing anyway as you requested:
 wget depends on libpcre2-8-0 (>= 10.32).
 php7.3-fpm depends on libpcre2-8-0 (>= 10.32).
 php7.3-cli depends on libpcre2-8-0 (>= 10.32).
 git depends on libpcre2-8-0 (>= 10.32).

Removing libpcre2-8-0:armhf (10.39-2+0~20211122.14+debian9~1.gbp0d570b) ...
dpkg: libpcre3:armhf: dependency problems, but removing anyway as you requested:
 mongodb-server depends on libpcre3.
 mongodb-clients depends on libpcre3.
 lighttpd depends on libpcre3.
 libselinux1:armhf depends on libpcre3.
 libpcrecpp0v5:armhf depends on libpcre3 (>= 2:8.41).
 libglib2.0-0:armhf depends on libpcre3.
 grep depends on libpcre3.

Removing libpcre3:armhf (2:8.44-2+0~20210301.9+debian9~1.gbpa278ad) ...
dpkg: libpcrecpp0v5:armhf: dependency problems, but removing anyway as you requested:
 mongodb-server depends on libpcrecpp0v5 (>= 7.7); however:
  Package libpcrecpp0v5:armhf is to be removed.
 mongodb-clients depends on libpcrecpp0v5 (>= 7.7); however:
  Package libpcrecpp0v5:armhf is to be removed.

Removing libpcrecpp0v5:armhf (2:8.44-2+0~20210301.9+debian9~1.gbpa278ad) ...
dpkg: libsodium23:armhf: dependency problems, but removing anyway as you requested:
 php7.3-fpm depends on libsodium23 (>= 1.0.14); however:
  Package libsodium23:armhf is to be removed.
 php7.3-cli depends on libsodium23 (>= 1.0.14); however:
  Package libsodium23:armhf is to be removed.

Removing libsodium23:armhf (1.0.18-1+0~20191009.1+debian9~1.gbpb6823f) ...
dpkg: libxml2:armhf: dependency problems, but removing anyway as you requested:
 php7.3-xml depends on libxml2 (>= 2.9.0).
 php7.3-fpm depends on libxml2 (>= 2.8.0); however:
  Package libxml2:armhf is to be removed.
 php7.3-cli depends on libxml2 (>= 2.8.0); however:
  Package libxml2:armhf is to be removed.
 libxslt1.1:armhf depends on libxml2 (>= 2.9.0).
 liblwres161:armhf depends on libxml2 (>= 2.6.27).
 libisccfg163:armhf depends on libxml2 (>= 2.6.27).
 libisccc161:armhf depends on libxml2 (>= 2.6.27).
 libisc1100:armhf depends on libxml2 (>= 2.7.4).
 libirs161:armhf depends on libxml2 (>= 2.6.27).
 libdns1104:armhf depends on libxml2 (>= 2.7.4).
 libbind9-161:armhf depends on libxml2 (>= 2.6.27).
 libarchive13:armhf depends on libxml2 (>= 2.7.4).
 dnsutils depends on libxml2 (>= 2.6.27).
 bind9-host depends on libxml2 (>= 2.6.27).

Removing libxml2:armhf (2.9.12+dfsg-0+0~20210621.9+debian9~1.gbp43e861) ...
dpkg: libzip4:armhf: dependency problems, but removing anyway as you requested:
 php7.3-zip depends on libzip4 (>= 1.3.2).

Removing libzip4:armhf (1.7.3-1+0~20210114.10+debian9~1.gbp4c125d) ...
dpkg: libargon2-1:armhf: dependency problems, but removing anyway as you requested:
 php7.3-fpm depends on libargon2-1 (>= 0~20171227).
 php7.3-cli depends on libargon2-1 (>= 0~20171227).
 libcryptsetup12:armhf depends on libargon2-1 (>= 0~20171227).

Removing libargon2-1:armhf (0~20190702-0.1+0~20190710.3+debian9~1.gbp2fb167) ...
dpkg: libgd3:armhf: dependency problems, but removing anyway as you requested:
 php7.3-gd depends on libgd3 (>= 2.1.1).

Removing libgd3:armhf (2.2.5-5.2+0~20190808.4+debian9~1.gbp6d9343) ...
Removing libicu65:armhf (65.1-1+0~20200223.8+debian9~1.gbp519cf3) ...
dpkg: libidn2-0:armhf: dependency problems, but removing anyway as you requested:
 wget depends on libidn2-0 (>= 0.6).
 libpsl5:armhf depends on libidn2-0 (>= 0.16).
 libgnutls30:armhf depends on libidn2-0 (>= 2.0.0).
 libcurl4:armhf depends on libidn2-0 (>= 0.6).
 libcurl3-gnutls:armhf depends on libidn2-0 (>= 0.6).
 iputils-ping depends on libidn2-0 (>= 0.6).
 dnsutils depends on libidn2-0 (>= 2.0.0).
 bind9-host depends on libidn2-0 (>= 2.0.0).

Removing libidn2-0:armhf (2.2.0-2+0~20200302.4+debian9~1.gbpf85c2e) ...
Processing triggers for libc-bin (2.28-10+rpt2+rpi1+deb10u1) ...
root@pi:~# apt -y install "${packages[@]}"
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libicu65:armhf
root@pi:~# unset -v packages
root@pi:~# dietpi-services start
mkdir: error while loading shared libraries: libpcre.so.3: cannot open shared object file: No such file or directory
[FAILED] DietPi-Services | Failed to create or enter scripts working directory: /tmp/DietPi-Services
[ INFO ] DietPi-Services | Entered users home directory: /root

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[FAILED] DietPi-Services | start : php7.3-fpm
[FAILED] DietPi-Services | start : lighttpd
[FAILED] DietPi-Services | start : cron
root@pi:~#
```

MichaIng, 2022-10-10:
Ah sorry one package name has changed:
```sh
apt -f install
```
Since all are dependencies, this should pull the required ones automatically. But please show the output, so I can verify this 😉.

tkrusterholz, 2022-10-11:
Well, my main computer that I was connected from had an unexpected power off during the ssh session, and after reboot I could no longer connect to my Pi via ssh.  And after rebooting the Pi now it suffers a kernel panic.  I may just take this opportunity to wipe the SD card and reinstall diet-pi.  As it stands I cannot boot the Pi.

MichaIng, 2022-10-11:
Ah dammit, bad timing, as we uninstalled quite some system libraries (while APT and SSH stay functional, according to my tests) 🙈. The kernel panic happens during early boot stage? Shouldn't be related to the missing libraries. Do you have another Linux system where you can mount the SD cards ext4 partition?

Flashing a fresh DietPi system at least has the benefit that you have a clean Bullseye system then. Depends on how much work it would be to set everything up again.

tkrusterholz, 2022-10-11:
I have a laptop that I've temp booted into Linux Mint and attached the SD card with a USB adapter.  I'm a bit out of my depth with devices on Linux, however from what I can tell from an fdisk -l command I'm not sure I see the card.  The laptop plays a noise when I connect the adapter so it's definitely detecting something, but there are no disks in the fdisk command output that aren't otherwise accounted for (laptop drive, boot USB, etc.)  Any suggestions what I should try to detect and mount the card?

MichaIng, 2022-10-11:
Um, if the disk is not detected at all, there is not much that can be done. Is the USB adapter detected? `lsusb`
Or are there kernel errors showing that the driver fails to load for some reason? `dmesg`

Like SD cards themselves, also SD-USB adapters often have bad quality and break/wear fast. In case of the adapters, same with native SD card sockets and SD-microSD adapters, they are often simply wearing out so that the physical connection between SD card and socket connectors is not tight anymore. If there is a way to give some soft pressure and/or cleaning the connectors, that might help already.

tkrusterholz, 2022-10-11:
I have another SD card that when swapped into the adapter instantly mounts.  And when swapping back to the dietpi card, does not mount.  I think that proves out the adapter.  Unless you have other ideas I think this card is unsalvageable.

tkrusterholz, 2022-10-11:
Ah ha!  I do seem to be able to view the disk from a Windows system.

MichaIng, 2022-10-11:
> Is there something you'd like me to pull off it if/when I can make it available again?

Nothing required from my end. The conflicts with the packages from the PHP repo used on Stretch is clear, the SD card now dying coincidentally 😢.

tkrusterholz, 2022-10-11:
Sounds good.  Thanks for your attention to this and trying to dig into it.  I've replaced the SD card for my Pi and have started rebuilding.  Fortunately it didn't do much other than pihole with a few custom entries in the hosts file.

closed 2022-10-11 by tkrusterholz