MichaIng/DietPi #5789: DietPi v8.9.2 raspberrypi-sys-mods breaks loading of some drivers?

f-laurens, 2022-10-05:
For some unknown reason , after updating from DietPi v8.8.1 to v8.9.2, some drivers do not load anymore on my Raspberry Pi
(no kernel error messages, the drivers just do not load).
I have kept same kernel version (5.15.32) so this is not a driver upgrade/rebuild issue.
My guess is that this might be related to the new DietPi specific version of `raspberrypi-sys-mods`...

What would be the way to reinstall the standard (20220901) version, in order to validate this?

MichaIng, 2022-10-05:
Which drivers do you mean in particular? We have a report that some LCD driver may not load correctly, leading to backlight sysfs entry (so backlight cannot be toggled).

Yes please test whether downgrading the package solves it again.

f-laurens, 2022-10-05:
> Which drivers do you mean in particular? 

The driver that fails to load  in my case is this specific [NFC driver](https://github.com/pguyot/st25r391x).

> Yes please test whether downgrading the package solves it again.

What would be the procedure (I'm no dpkg/apt guru ;-) to downgrade `raspberrypi-sys-mods`?

f-laurens, 2022-10-06:
I downgraded package `raspberrypi-sys-mods `to the standard (non DietPi) 20220915 version thru:
```
sudo apt-get remove raspberrypi-sys-mods
sudo apt-get autoremove
sudo apt-get install raspberrypi-sys-mods=20220915
```
and this indeed solves the problem for me: after rebooting, the NFC driver now loads correctly.

MichaIng, 2022-10-06:
I think I found it, please test:
```sh
cd /tmp
curl -O https://dietpi.com/downloads/binaries/raspberrypi-sys-mods.deb
dpkg -i raspberrypi-sys-mods.deb
rm raspberrypi-sys-mods.deb
```

f-laurens, 2022-10-06:
> I think I found it, please test:

This new version _2:20220915-dietpi2_ indeed fixes the problem.
Thank!

MichaIng, 2022-10-07:
Many thanks for the feedback 👍.

closed 2022-10-15 by MichaIng