MichaIng/DietPi #5791: DietPi-Software | Squeezelite: Rebuild required on Bookworm

MichaIng, 2022-10-06:
with `libavformat59` instead `libavformat58`, else install fails on Raspbian/ARMv6.

aposcic, 2022-10-13:
Hi @MichaIng,

Any chance you could also bump up the SqueezeLite version while rebuilding?

The latest version available at [ralph-irving/squeezelite](https://github.com/ralph-irving/squeezelite) is 1.9.9-1403 (main branch tip).

Many thanks!

Antonio

MichaIng, 2022-10-13:
Jep, since there are no release tags or other branches, the package will be always based on current master branch.

MichaIng, 2022-10-13:
New packages up for testing:
```sh
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/testing/shairport-sync_$G_HW_ARCH_NAME.deb
apt install ./shairport-sync_$G_HW_ARCH_NAME.deb
rm shairport-sync_$G_HW_ARCH_NAME.deb
```

aposcic, 2022-10-13:
Thanks, @MichaIng, this is great! And thanks for providing the script/recipe that you use to build SqueezeLite for DietPi, that's even more useful!

Again, many thanks!

P.S. Just a heads-up that you linked/referenced the `shairport-sync` package instead of `squeezelite` in the comment above. :-) `https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/testing/squeezelite_$G_HW_ARCH_NAME.deb` is the correct URL, of course.

MichaIng, 2022-10-13:
Ah thanks, I fixed it above.

closed 2022-10-15 by MichaIng

FabioLolix, 2022-10-20:
> since there are no release tags or other branches, the package will be always based on current master branch.

While it don't use git tags it is versioned in squeezelite.h, i.e. v1.9.9.1403 https://github.com/ralph-irving/squeezelite/commit/bc72c0de3fff771540a2a45aaafafed539387b3c

MichaIng, 2022-10-20:
Your packages are at v1.9.9.1411 already 🙂.

FabioLolix, 2022-10-20:
> at v1.9.9.1411 already

it was an example because in v1.9.9.1403 the commit is very brief and versioning is spotted immediately

> Your packages

Not sure what you mean with 'Your packages'

MichaIng, 2022-10-20:
We packaged the latest versions downloaded in `dietpi-software` at `https://dietpi.com/downloads/binaries/<distro>/testing/squeezelite_<arch>.deb`. On rebuild via GitHub actions, the latest master commit will be always pulled, and the version for the package indeed taken from `squeezelite.h`.

MichaIng, 2022-10-20:
> Not sure what you mean with 'Your packages'

Ah lol, I meant "Our" packages, typo, edited above 😅.