MichaIng/DietPi #5792: Help needed to resolve curl: (6) Couldn't resolve host '[raw.githubusercontent.com]

gururajkashikar, 2022-10-07:
Hi Michalng,
Good day. Further to our discussion, creating this issue.

During the first startup, while upgrading the dietpi, it fails with the following error

curl: (6) Couldn't resolve host '[raw.githubusercontent.com](http://raw.githubusercontent.com/)'

then pops up following message
-------------------------------
[Failed] Unknown install state/First run setup failed
An error has occured either during first run update or installs.

First run setup will now attempt to re-apply the last step, forced as interactive run.
If this repeatedly fails, please collect all the terminal output and the content of /var/tmp/dietpi/logs/dietpi-firstrun-setup.log if available and report the issue to: > https://github.com/MichaIng/DietPi/issues

-------------------------------

Solutions tried

1. Solution-1 : as mentioned in this [link](https://github.com/MichaIng/DietPi/issues/5426)
2. I tried changing the dns-nameservers to 8.8.8.8 8.8.4.4
3. Space is not an issue as in this [case](https://dietpi.com/forum/t/dietpi-first-run-setup-failed/5033/28). 

Note : When I use curl [www.google.com](http://www.google.com/), I get the html page of google on dietpi box itself. 

Similarly if I do curl  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt it throws curl: (6) Couldn't resolve host '[raw.githubusercontent.com](http://raw.githubusercontent.com/)'

Even on my windows laptop, if I try this URL in the Browser it says 

------------------------
This site can’t be reached
Check if there is a typo in [raw.githubusercontent.com](http://raw.githubusercontent.com/).

If spelling is correct, try running Windows Network Diagnostics.
DNS_PROBE_FINISHED_NXDOMAIN
------------------------
Same error occurs if I try to install any other software like LXQt or MotionEye.

This is a fresh installation and does not have any Ad-blocker or PiHole installed.

Please help to resolve this issue.

Thanks in advance.

Regards,
Guru

MichaIng, 2022-10-07:
Many thanks for opening the issue.

I often see this because public ad block lists sometimes contain raw.githubusercontent.com for some reason. That also your laptop is affected is another indication for this. If you use Pi-hole, AdGuard Home or another DNS-based ad blocker, please try to whitelist this domain.

Joulinar, 2022-10-07:
But this is not working on DHCP used. Isn't it? There DNS server is provided by DHCP server?

gururajkashikar, 2022-10-07:
> Regarding your email: The DNS blocker could run somewhere else on your network or be invoked online by your router. Please try the following:
> 
> ```shell
> echo 'nameserver 9.9.9.9' > /etc/resolv.conf
> curl -fI 'https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt'
> ```

I tried this. No luck. 
Infact my /etc/resolv.conf already had nameserver  8.8.8.8 and nameserver 8.8.4.4 as I have added this in my /etc/resolvconf/resolv.conf.d/base.

Joulinar, 2022-10-07:
Could you try following

`apt install dnsutils`

once done we could check how DNS resolution is going

gururajkashikar, 2022-10-07:
Ok I installed dnsutils and then did nslookup raw.githubusercontent.com. Following is the output
Server : 192.168.2.251
Address : 192.168.2.251#53

** server can't find raw.githubusercontent.com : NXDOMAIN.

Note 192.168.2.251 is the default DNS on my /etc/resolv.conf

Then I tried echo 'nameserver 9.9.9.9' > /etc/resolv.conf and tried nslookup raw.githubusercontent.com
;; connection time out; no servers could be reached.

Even after increasing G_CHECK_URL Timeout to 60 secs and G_CHECK_URL Attempts to 5 still connection time out is popping up.

Joulinar, 2022-10-07:
Do you use static IP or DHCP?

gururajkashikar, 2022-10-07:
I use DHCP.  
Please see the attached screen shot for the output of dig command.
[image]

Joulinar, 2022-10-07:
Pls switch to STATIC IP. Otherwise you are not able to change DNS server.

Btw, do you know who this IP 192.168.2.251 address is?

gururajkashikar, 2022-10-07:
I tried switching to Static. But Networking is not coming up. I'll give it a try once again.

192.168.2.251 is the default DNS obtained automatically from my router.

I'll take a break and try this tomorrow. Need to go to bed now.

Joulinar, 2022-10-07:
Once switching to Static IP, did you adjust the IP configuration? Because the default one is not fitting your network environment. If course this will break your system. On the network configuration page, there should be a button to copy DHCP into STATIC. This way you will keep current IP address assigned. And you will be able to adjust DNS settings.

MichaIng, 2022-10-07:
> Otherwise you are not able to change DNS server.

It is possible via e.g. `echo 'nameserver 9.9.9.9' > /etc/resolv.conf` to change the DNS temporarily, until the DHCP lease is renewed. Only the content of `/etc/resolv.conf` is relevant for the system resolver. Just strange that it timed out with Quad9 DNS.

@gururajkashikar 
How long did the `dig` command take? Default timeout for the system resolver is 5 seconds. As you used `/etc/resolvconf/resolv.conf.d/base` before to add a custom DNS (which should work as well), do you have the `resolvconf` package installed?

Joulinar, 2022-10-07:
@MichaIng query time is showing on the `dig` screen shot. It was 168 msec.

gururajkashikar, 2022-10-07:
@MichaIng, yes i have installed resolvconf package.
But i am seeing one issue. Even though there are three nameserver in /etc/resolv.conf always default one (192.168.2.251) is picked up for name resolution..

I just learnt we need to use dnsmasq to manage this kind of scenario or use Static IP as Joulinar suggest.
Can also see how to use option rotate in resolv.config if that helps.

MichaIng, 2022-10-07:
Not sure what you want to use `dnsmasq` for. It has an effect only if you add it as a fourth DNS `127.0.0.1`, and the system could still use `192.168.2.251`.

The order in which the `nameserver` entries occur can have an effect. Try to add your custom one via `/etc/resolvconf/resolv.conf.d/head` instead.

I never heard of a router which denied access when the static IP does just match the dynamic IP that was previously applied via DHCP. To rule out that the same IP is assigned to another device, you can make it a reserved IP on the router. That would make thing much easier and you don't need `resolvconf` either then.

gururajkashikar, 2022-10-08:
Atlast we resolved it.
Adding nameserver 9.9.9.9 into /etc/resolvconf/resolv.conf.d/head and restarting appened the same to /etc/resolv.conf at the first line, above the default nameserver.
When curl is able to resolve the host and whole process is completed.

Thank you so much for the help, @MichaIng  ans @Joulinar.

Can we make this resolution part of first boot script?

Joulinar, 2022-10-08:
Honestly I'm not 100% sure what exactly you mean. What should be part of our first boot script? Package `resolvconf` is not part of our images and usually it is not needed. As well we don't know what DNS server the user like to use. On our configuration file we offer the user to select between STATIC IP and DHCP.

gururajkashikar, 2022-10-08:
Though the solution was to get nameserver 9.9.9.9 added to resolvconf, I spent almost 2 days debugging this. My thought was, if any one hit this issue, either we add nameserver to resolvconf or atleast provide a message saying the above steps can resolve the issue will surely help.
Anyways now that this issue is here, people will get idea of how to resolve this curl issue if they hit one.

Joulinar, 2022-10-08:
As stated above, our images don't contain `resolvconf` package. And it is not that common that local DNS server blocking access to global GitHub site.

MichaIng, 2022-10-08:
It is good that you found a solution for your particular case, but it is more a workaround. A better workaround would be to apply a static IP, a real solution would be to find out why/how to router blocks resolving the essential `raw.githubusercontent.com`, reporting it to whoever is responsible sabotaging important parts of the open source world with a wrong block list entry. It is better to find and report such issues than working around and ignoring them 😉.

closed 2022-10-23 by MichaIng