MichaIng/DietPi #5794: Update DietPi from v8.8.1 to v8.9.2 fails

joshimosh, 2022-10-09:
#### Details:
- Date           | Sun Oct  9 09:06:01 CEST 2022
- DietPi version | v8.8.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | `Linux RPiHB 5.15.61+ #1579 Fri Aug 26 11:08:59 BST 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `cp -a DietPi-master/rootfs/. /`
- Exit code      | 1
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
cp: cannot stat '/./etc/sysctl.d/dietpi.conf': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-cloudshell.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-firstboot.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-fs_partition_resize.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-kill_ssh.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-postboot.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-preboot.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-vpn.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/dietpi-wifi-monitor.service': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/ifup@.service.d': Structure needs cleaning
cp: cannot stat '/./etc/systemd/system/ifupdown-pre.service.d': Structure needs cleaning

Joulinar, 2022-10-09:
"Structure needs cleaning" usually indicates a filesystem errors. Please try to clean it on reboot:

```sh
> /forcefsck
reboot
# after reboot check logs
journalctl -t systemd-fsck
```

joshimosh, 2022-10-09:
Thank you for the advice. Unfortunately the cleanup left a system that refused to boot.
I will install a backup and try again.

nicgravel, 2022-10-11:
I have the same problem. Error on update and system dead on reboot.
[image]

Joulinar, 2022-10-11:
Looks your SD card has some file system corruption.

MichaIng, 2022-10-11:
Is it an RPi or other device?

nicgravel, 2022-10-11:
rpi4

MichaIng, 2022-10-11:
So no fsck via initramfs possible. Do you have another Linux system where you can mount this SD card (or USB drive) to run fsck on? Because systemd itself is broken so that it cannot even boot util a stage where it might be able to repair itself.

nicgravel, 2022-10-11:
i will try to repair it on another linux system tonight...

joshimosh, 2022-10-12:
In my case, replacing from a backup (bootable image created with rpi-clone, it's very convenient and never gave me any problems) did it.
Update to v8.9.2 worked flawlessly.

MichaIng, 2022-11-09:
@nicgravel 
Were you able to repair the drive?

nicgravel, 2022-11-09:
unfortunately no. Had to reinstall everything but now I have a USB key to save a full backup of the system...just in case...
I tested it and everything works well after a restore.

MichaIng, 2022-11-10:
Makes sense to keep full backups or at least backups of those data you need to restore on a newly setup system, especially when using an SD card as system drive.

Okay, I'll mark this as closed then.

closed 2022-11-10 by MichaIng