MichaIng/DietPi #5795: Nextcloud + MariaDB service failed

oar-globant, 2022-10-10:
#### Details:
- Date           | Mon Oct 10 16:49:23 BST 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `systemctl restart mariadb`
- Exit code      | 1
- Software title | DietPi-Services

#### Additional Information (if applicable)
- Software title | NextCloud && MariaDB
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? I don't have sure
- Bug report     | 395ca0f5-4a55-48db-b1cf-8b3441c09242

#### Steps to reproduce

1. Try to access Nextcloud show issue: 
```
Internal Server Error
The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the server log.
```

2. Doing investigation more deeply I found one service are not running, is mariadb
3. Showing this error:
```
mariadb.service - MariaDB 10.5.15 database server
│      Loaded: loaded (/lib/systemd/system/mariadb.service; disabled; vendor preset: enabled)
│     Drop-In: /etc/systemd/system/mariadb.service.d
│              └─dietpi-services_edit.conf
│      Active: failed (Result: exit-code) since Mon 2022-10-10 16:54:02 BST; 22s ago
│        Docs: man:mariadbd(8)
│              https://mariadb.com/kb/en/library/systemd/
│     Process: 11332 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld
│ (code=exited, status=0/SUCCESS)
│     Process: 11333 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION
│ (code=exited, status=0/SUCCESS)
│     Process: 11335 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`cd
│ /usr/bin/..; /usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment
│ _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
│     Process: 11383 ExecStart=/usr/sbin/mariadbd $MYSQLD_OPTS $_WSREP_NEW_CLUSTER
│ $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
│    Main PID: 11383 (code=exited, status=1/FAILURE)
│      Status: "MariaDB server is down"
│         CPU: 254ms
│
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] InnoDB: The innodb_system
│ data file 'ibdata1' must be writable
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] Plugin 'InnoDB' init function
│ returned error.
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] Plugin 'InnoDB' registration
│ as a STORAGE ENGINE failed.
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [Note] Plugin 'FEEDBACK' is disabled.
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] Could not open mysql.plugin
│ table: "Unknown storage engine 'Aria'". Some plugins may be not loaded
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] Failed to initialize plugins.
│ Oct 10 16:54:02 DietPi mariadbd[11383]: 2022-10-10 16:54:02 0 [ERROR] Aborting
```

#### Expected behaviour
- Normally start into reboot or something and should be able to access Nextcloud.

#### Actual behaviour
- Not possible to access Nextcloud.

#### Extra details
- Important notes I have 2 HDD's into my NASpi, one used to Nextcloud storage (sdb) and another to Media service (sda)
- Also other services like, Jellyfin, Navidrome, DietpiDashboard still working

**lsblk:**
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0   3.6T  0 disk 
└─sda1        8:1    0   3.6T  0 part /mnt/jelly_media
sdb           8:16   0 931.5G  0 disk 
└─sdb1        8:17   0 931.5G  0 part /mnt/cloud
mmcblk0     179:0    0  59.5G  0 disk 
├─mmcblk0p1 179:1    0   128M  0 part /boot
└─mmcblk0p2 179:2    0  59.3G  0 part /
```

oar-globant, 2022-10-10:
**Also:**
readlink /var/lib/mysql
`/mnt/dietpi_userdata/mysql`
readlink -f /var/lib/mysql
`/mnt/cloud/dietpi_userdata/mysql`

Joulinar, 2022-10-10:
Your external disk sdb1 is within r/o mode and the system is not able to access the disk

`Errcode: 30 "Read-only file system`

oar-globant, 2022-10-10:
Hahahahah Thanks @Joulinar !
**Solved**  in first time setup Dietpi I made config for mount hard drivers manually, then after years I noticed its possible manage this by dietpi-config soo I moved and prbly forget something.

Close case! Thanks mate!

closed 2022-10-10 by oar-globant