MichaIng/DietPi #5796: Restore backup from Raspberry PI on Odroid N2+

sonnatager, 2022-10-10:
Hi,

I have a Raspberry PI 4 and want to move to a Odroid N2+. I don't want to initialize and configure all the services on more on the Odroid N2+. Can i restore a backup from the Raspberry PI on the Odroid to get all the configs?

Services installed:
[image]

Raspberry PI:
[image]

Joulinar, 2022-10-10:
I don't think this is going to work. Both are 2 different SBC running to different base images, kernel versions aso.

Joulinar, 2022-10-10:
> Should i mount the harddrive on the odroid before or after the setup of the nextcloud?

definitely. 

You could run the uninstall of NC on the RPi, as this will create a database dump. This once could be used on the N2 during install to create the original system automatically.

sonnatager, 2022-10-11:
Definitely mount it before?

With uninstall, you mean uninstall via dietpi-software?

Joulinar, 2022-10-11:
yes, first run `dietpi-software uninstall` on the original system. This will create the database dump. Usually, it should be created on the external disk already. Once done, move the disk onto your new device, mount it on the very same position und try to install NC. Theoretically your dump should be considered by the install process. Important is to have the dump located at `/mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-database-backup.sql` before installation starts.

sonnatager, 2022-10-11:
Thank you for your quick responses  :smile:. I will try this

closed 2022-10-11 by sonnatager

MichaIng, 2024-12-01:
@Phil1988 
You require elevated permissions as well to store the file, not only to access the database. Easiest is probably to run a subshell as root:
```sh
sudo bash -c 'mysqldump nextcloud > /mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-database-backup.sql'
```

Background is that a shell redirect `>` runs as originating user. So `sudo mysqldump nextcloud` is the command, which works as you run `mysqldump` via `sudo` with `root` permissions, but the shell redirect redirects the output of the `sudo` command, and is not part/passed through to the `sudo` command. Hence the following does not work either, as it calls `echo` (unnecessarily) with `root` permissions, but the redirect not:
```sh
sudo echo test > /root/test
```
For such, you will find constructs like this:
```sh
echo test | sudo tee /root/test
```
Here `echo` is called as originating user, but the output is written via `sudo`-elevated `tee`. In our case with the database, however both, the database access as well as the file write requires elevated permissions, hence easiest to wrap both into a subshell.

Phil1988, 2024-12-01:
Seems I suceeded with this:
`sudo mysqldump -u root -p nextcloud > /home/dietpi/dietpi-nextcloud-database-backup.sql`
It generated a sql file with this content at the beginning:
```
-- MariaDB dump 10.19  Distrib 10.11.6-MariaDB, for debian-linux-gnu (x86_64)
--
-- Host: localhost    Database: nextcloud
-- ------------------------------------------------------
-- Server version	10.11.6-MariaDB-0+deb12u1
```

Do you think this was OK or should I go your route?

MichaIng, 2024-12-01:
> Do you think this was OK or should I go your route?

If it shall be restored automatically on a new system on Nextcloud install, it must be stored to `/mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-database-backup.sql`.

Phil1988, 2024-12-01:
Yes I know that.
I just downloaded it manually and will restore it to that location.

So I would do that:
1. uninstall nextxloud on the old system.
this will create
`/mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-installation-backup`
2. copy the file backup to the new system to
`/mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-installation-backup`
3. copy the mysql backup to the new system to
`/mnt/dietpi_userdata/nextcloud_data/dietpi-nextcloud-database-backup.sql`
4. install nextcloud on the new system via `dietpi-software`
5. hope that everything will be restored

Is this correct?
One question to the`dietpi-nextcloud-installation-backup`:
My cloud consumes 58% of space is the dietpi `dietpi-nextcloud-installation-backup` a compressed backup of all the files or something similar?
I fear it could abort the backup while uninstalling it due to insufficient space.

MichaIng, 2024-12-01:
An uninstall will btw create the database backup as well :) , at least if the database server succeeds to start. Verify it, but in case you could then just copy `/mnt/dietpi_userdata/nextcloud_data` to the new system as a whole, before installing Nextcloud there.

`dietpi-nextcloud-installation-backup` is only the backup if the Nextcloud PHP scripts and assets, not the data. The data is in its parent directory `/mnt/dietpi_userdata/nextcloud_data`, which is left untouched. So basically the database dump and the Nextcloud scripts (which includes the config) are all moved into that data directory. And if that one is found on a new installation, it will check for and in case restore the backup automatically.