MichaIng/DietPi #5797: Google Cloud Platform Integration

JonVojtush, 2022-10-11:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- I want to use Diet Pi to host a server on Google Cloud Platform's Compute Engine (e# virtual machine). After doing the manual installation which requires a fresh install of debian, I download and run the install script. Although in the process diet pi removes all pre-existing software packages, some of which are required by Google to manage their VM instances (Google Cloud SDK). In addition, I am not able to SSH into the VM at all because it looks like the expected user was deleted. (If the host name changes that would probably also cause an issue)

#### Describe the solution you'd like:

- I'm thinking that if we add Google's required packages to a whitelist of packages to ignore if they are already installed, that should work.
- For the pre-existing users, during the installation script, an option can be given to keep or remove all/select pre-existing users and SSH keys.
- the same applies for the host name, I need to double check to see if the option is given to retain the existing host name.

MichaIng, 2022-10-12:
Can you share a list of packages which are definitely required?

There is a `dietpi.txt` option to add an SSH key for the root user. We actually plan to remove the dietpi user as well, and instead allow to create a custom login user at first boot, which could then imply the option to add an SSH key for it. However, for first login, root is required anyway, and SSH should work with root user + password, isn't it?

The `dietpi-installer` is currently not designed or aimed to pre-configure an image, but to create a minimal system without any customisation. These are to be added via `dietpi.txt` or manually after first boot instead, which implies custom login users and the hostname. We could think of adding customisation options to `dietpi-imager`, but that's an entirely different and large topic, to be handled correctly as well by first boot scripts etc. For now I'd concentrate on adding everything that DietPi can boot on GCP until the default local + SSH login abilities, leaving SSH key and hostname to be done via `dietpi.txt` as it is now.

JonVojtush, 2022-10-24:
I imagine that the [Google-Cloud-SDK](https://cloud.google.com/sdk/#section-2) & its dependencies would suffice. 

Some sort of [Custom Image](https://cloud.google.com/compute/docs/images/building-custom-os) appears to be their recommended method. I found [Packer](https://www.packer.io/downloads) may make that process easier.

Lol, you guys building something specific would be my convenient option 😄

closed 2025-05-11 by nomadicGopher