MichaIng/DietPi #5798: Unable to auto-setup when armbian repo is down

ngrigoriev, 2022-10-12:
It is not for the first time...Is there a way to specify an alternative repo URL? Right now http://apt.armbian.com/region/NA (NA mirror) works just fine, but the original one constantly fails with HTTP 500. I think it is pretty bad situation when it is impossible to provision a system because of that :(

#### Details:                                                                                                                                                                                             
- Date           | Tue Oct 11 20:59:59 EDT 2022                                                                                                                                                           
- DietPi version | v7.9.3 (MichaIng/master)                                                                                                                                                               
- Image creator  | DietPi Core Team                                                                                                                                                                       
- Pre-image      | Armbian                                                                                                                                                                                
- Hardware       | PINE H64 (aarch64) (ID=45)                                                                                                                                                             
- Kernel version | `Linux pineh64-05 5.10.60-sunxi64 #21.08.1 SMP Wed Aug 25 18:29:57 UTC 2021 aarch64 GNU/Linux`                                                                                         
- Distro         | bullseye (ID=6)                                                                                                                                                                        
- Command        | `apt-get -q update`                                                                                                                                                                    
- Exit code      | 255                                                                                                                                                                                    
- Software title | DietPi-Update                                                                                                                                                                          
#### Steps to reproduce:                                                                                                                                                                                  
                                                                                                                                                               
1. ...                                                                                                                                                                                                    
2. ...                                                                                                                                                                                                    
#### Expected behaviour:                                                                                                                                                                                  
                                                                                                                                                                              
- ...                                                                                                                                                                                                     
#### Actual behaviour:                                                                                                                                                                                    
                                                                                                                                                                               
- ...                                                                                                                                                                                                     
#### Extra details:                                                                                                                                                                                       
                                                                                                                                    
- ...                                                                                                                                                                                                     
#### Additional logs:                                                                                                                                                                                     
```                                                                                                                                                                                                       
Err:1 http://apt.armbian.com bullseye InRelease                                                                                                                                                           
  500  Internal Server Error [IP: 130.185.239.78 80]                                                                                                                                                      
Get:2 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]                                                                                                                        
Get:3 https://debian.mirror.rafal.ca/debian bullseye InRelease [116 kB]                                                                                                                                   
Get:4 https://debian.mirror.rafal.ca/debian bullseye-updates InRelease [44.1 kB]                                                                                                                          
Get:5 https://debian.mirror.rafal.ca/debian bullseye-backports InRelease [49.0 kB]                                                                                                                        
Get:6 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [187 kB]                                                                                                               
Get:7 https://debian.mirror.rafal.ca/debian bullseye/non-free arm64 Packages [72.9 kB]                                                                                                                    
Get:8 https://debian.mirror.rafal.ca/debian bullseye/contrib arm64 Packages [41.0 kB]                                                                                                                     
Get:9 https://debian.mirror.rafal.ca/debian bullseye/main arm64 Packages [8071 kB]                                                                                                                        
Get:10 https://debian.mirror.rafal.ca/debian bullseye-updates/main arm64 Packages [4160 B]                                                                                                                
Get:11 https://debian.mirror.rafal.ca/debian bullseye-backports/non-free arm64 Packages [7096 B]                                                                                                          
Get:12 https://debian.mirror.rafal.ca/debian bullseye-backports/main arm64 Packages [339 kB]                                                                                                              
Get:13 https://debian.mirror.rafal.ca/debian bullseye-backports/contrib arm64 Packages [3840 B]                                                                                                           
Fetched 8983 kB in 3s (2582 kB/s)                                                                                                                                                                         
[... cut ...]                                                                                           
W: Failed to fetch http://apt.armbian.com/dists/bullseye/InRelease  500  Internal Server Error [IP: 130.185.239.78 80]                                                                                    
W: Some index files failed to download. They have been ignored, or old ones used instead.                                                                                                                 
```

ngrigoriev, 2022-10-12:
It seems I was able to unblock the situation by doing this:

1. Mounting the SD card
2. Changing the URL in etc/apt/sources.list.d/armbian.list to "http://apt.armbian.com/region/NA"
3. (important, not all mirrors seem to be the same) adding 'Acquire::Retries "5";' to etc/apt/apt.conf.d/97dietpi

MichaIng, 2022-10-12:
Interesting idea with the retries. Better use an own config for this, since `97dietpi` is overwritten on DietPi updates:
```sh
echo 'Acquire::Retries "5";' > /etc/apt/apt.conf.d/99retries
```
It's a pity that some of the Armbian mirrors are so unreliable.

Not sure how to deal with this best. I'm not keen to simply ignore unreachable APT repositories, as this often requires attention, e.g. when outdated repos or suites have been taken down and need to be updated. Also ignoring failure in the update can lead to follow-up issues when actual packages are picked and tried to be installed from the existing (and then outdated) list files, which makes it only harder to debug and error-exiting in a middle of an installation very unclean.

Adding a special case for Armbian repos only? Not sure, probably it's better to keep collecting such reports, and forward to Armbian if not resolved in shorter time, as if bugs their users the same way.

What I can think of is adding a mirror selection for the Armbian repo, region and/or an explicit mirror. The list can be taken from:
- https://apt.armbian.com/mirrors
- https://apt.armbian.com/mirrors.json

Could be a `dietpi.txt` setting as well.

jahway603, 2022-10-12:
I encountered this same error yesterday on a Rock64 SBC.

ngrigoriev, 2022-10-13:
I am looking at dietpi/dietpi-software and around line 16216 there is:

```
		# Full package upgrade on first run installs: https://github.com/MichaIng/DietPi/issues/3098
		if [[ ! -f '/boot/dietpi/.skip_distro_upgrade' ]]
		then
			G_AGUP
			# Do not repeat APT update in Run_Installations()
			SKIP_APT_UPDATE=1
			G_AGDUG
```

So, it seems that adding this dot-file will disable the auto-update and, thus, will produce the repeatable installations. That can be upgraded later to the level needed.

MichaIng, 2022-10-13:
This flag file is for skipping the "dist-upgrade" only, which assures that kernel upgrades are done once on x86 where they would be othweise skipped. Regular "apt-get upgrades" are still always done.

Repeatable first run installs are conflicting our current aim to always skip up-to-date systems with latest security patches etc, though I see that this sometimes might not be wanted, at least not forced. Currently there are also sometimes package installs done as part of the first run setup, like marching microcode installs based on whether an Intel or AMD CPU is used on x86 systems. It wound require a larger rework to allow this, but has other benefits, like allowing an offline setup, which is a request as well: https://github.com/users/MichaIng/projects/3

So the plan could then be to combine an offline mode with reproducible mode, as one allows/implies the other.