MichaIng/DietPi #5799: Heimdall dashboard - a docker compose package for listing and linking your various web ports and URLs easily

KalixtShawxo, 2022-10-12:
### Creating a software request

#### Formal software information
- Software title | Heimdall Dashboard
- Short description | Heimdall is a software aimed at being a dashboard for all your web applications and is a perfect solution for your browser’s start page.
- Official URL (if available) | found it at https://pimylifeup.com/raspberry-pi-heimdall/
- GitHub URL (if available) | 
- (Official) install documentation | same as above

#### Are there similar/alternative software titles available with DietPi-Software?

- none directly

#### What makes your requested software better than the above solutions, if available?
Provides an easy way to have a central webpage for all your various web apps that have different ports or address URLS. 
- ...

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- it is a Docker-Compose package and could be easily automated into a DietPi Software feature

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. install docker-compose
2. install Heimdall package

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- I don't know a thing about BASH coding, I just thought this would be a nice addition to DietPi. I have it on a few of my PI systems now.

Joulinar, 2022-10-12:
usually, we don't add Docker container to our repository as they could be created easily yourself.

KalixtShawxo, 2022-10-12:
ok. it was just a thought since I recently added it to a couple of my DietPi systems that have multiple ports to remember. (Pi-Hole, dashboard, NetData, Webmin, Folding@home, etc) that is seemed like something others could benefit from as a ready made install.

MichaIng, 2022-10-12:
Btw, we plan such feature for the DietPi-Dashboard: https://github.com/ravenclaw900/DietPi-Dashboard/issues/169

KalixtShawxo, 2022-10-12:
> Btw, we plan such feature for the DietPi-Dashboard: [ravenclaw900/DietPi-Dashboard#169](https://github.com/ravenclaw900/DietPi-Dashboard/issues/169)

Awesome! I look forward to it. I like all the features of your Dashboard, it's a must install on everything.  :-)

KalixtShawxo, 2022-10-12:
Guess I'll close this then, since Michalng is already working on something for his DietPi-dashboard.

closed 2022-10-12 by KalixtShawxo