MichaIng/DietPi #5803: Home Assistant missing dependency

kennyeni, 2022-10-14:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version
bullseye 1
- Kernel version
Linux DietPi 5.15.61+ #1579 Fri Aug 26 11:08:59 BST 2022 armv6l GNU/Linux
- SBC model
RPi B (armv6l)
- Power supply used
Unknown
- SD card used
SanDisk Ultra

#### Additional Information (if applicable)
- Software title
Home Assistant
- Was the software title installed freshly or updated/migrated?
Fresh
- Can this issue be replicated on a fresh installation of DietPi?
I think so
- Bug report ID
dcc23b27-5d77-4407-8886-fe1b3d7e5b98

#### Steps to reproduce
1. Try to install Home Assistant via dietpi-install
2. ...

#### Expected behaviour
- Installation completes

#### Actual behaviour
- It fails since it's expecting rustup toolchain

#### Extra details

- Tried with rustup.sh, but the PATH would not be updated, so installation has to be retried (at least once, still running since it takes a couple hours)

Joulinar, 2022-10-14:
Could you share the full error you are facing? Maybe you can scroll up a bit on your SSH terminal?

kennyeni, 2022-10-14:
```
[  OK  ] DietPi-Software | chown -R homeassistant:homeassistant pyenv-master
[  OK  ] DietPi-Software | mv pyenv-master /home/homeassistant/.pyenv
[ INFO ] DietPi-Software | Installing Python with Home Assistant module into pyenv, please wait...
Downloading Python-3.9.13.tar.gz...
-> https://www.python.org/ftp/python/3.9.13/Python-3.9.13.tgz
Installing Python-3.9.13...
WARNING: The Python lzma extension was not compiled. Missing the lzma lib?
Installed Python-3.9.13 to /home/homeassistant/.pyenv/versions/3.9.13

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Requirement already satisfied: pip in ./.pyenv/versions/3.9.13/lib/python3.9/site-packages (22.0.4)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-22.2.2-py3-none-any.whl (2.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.0/2.0 MB 1.2 MB/s eta 0:00:00
Requirement already satisfied: setuptools in ./.pyenv/versions/3.9.13/lib/python3.9/site-packages (58.1.0)
Collecting setuptools
  Downloading https://www.piwheels.org/simple/setuptools/setuptools-65.4.1-py3-none-any.whl (1.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 771.5 kB/s eta 0:00:00
Collecting wheel
  Downloading https://www.piwheels.org/simple/wheel/wheel-0.37.1-py2.py3-none-any.whl (35 kB)
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 58.1.0
    Uninstalling setuptools-58.1.0:
      Successfully uninstalled setuptools-58.1.0
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-22.2.2 setuptools-65.4.1 wheel-0.37.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple/
Collecting homeassistant
  Downloading https://www.piwheels.org/simple/homeassistant/homeassistant-2022.10.3-py3-none-any.whl (19.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 19.7/19.7 MB 1.6 MB/s eta 0:00:00
Collecting awesomeversion==22.9.0
  Downloading https://www.piwheels.org/simple/awesomeversion/awesomeversion-22.9.0-py3-none-any.whl (12 kB)
Collecting httpx==0.23.0
  Downloading https://www.piwheels.org/simple/httpx/httpx-0.23.0-py3-none-any.whl (84 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 84.8/84.8 kB 360.6 kB/s eta 0:00:00
Collecting astral==2.2
  Downloading https://www.piwheels.org/simple/astral/astral-2.2-py2.py3-none-any.whl (30 kB)
Collecting typing-extensions<5.0,>=4.4.0
  Downloading https://www.piwheels.org/simple/typing-extensions/typing_extensions-4.4.0-py3-none-any.whl (26 kB)
Collecting async-timeout==4.0.2
  Downloading https://www.piwheels.org/simple/async-timeout/async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
Collecting python-slugify==4.0.1
  Downloading https://www.piwheels.org/simple/python-slugify/python_slugify-4.0.1-py2.py3-none-any.whl (7.0 kB)
Collecting voluptuous-serialize==2.5.0
  Downloading https://www.piwheels.org/simple/voluptuous-serialize/voluptuous_serialize-2.5.0-py3-none-any.whl (3.4 kB)
Collecting ifaddr==0.1.7
  Downloading https://www.piwheels.org/simple/ifaddr/ifaddr-0.1.7-py2.py3-none-any.whl (10 kB)
Collecting pyyaml==6.0
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0-cp39-cp39-linux_armv6l.whl (45 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.3/45.3 kB 243.7 kB/s eta 0:00:00
Collecting jinja2==3.1.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.1/133.1 kB 409.5 kB/s eta 0:00:00
Collecting orjson==3.7.11
  Downloading orjson-3.7.11.tar.gz (946 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 946.6/946.6 kB 1.9 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
[... 27 lines cut ...]
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
/boot/dietpi/dietpi-bugreport: line 138: aplay: command not found
[FAILED] DietPi-Services | ● lighttpd.service - Lighttpd Daemon
     Loaded: loaded (/lib/systemd/system/lighttpd.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-10-13 18:12:30 PDT; 13h ago
   Main PID: 7250 (code=exited, status=1/FAILURE)
        CPU: 1min 40.594s

Warning: journal has been rotated since unit was started, output may be incomplete.
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: dcc23b27-5d77-4407-8886-fe1b3d7e5b98

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...
```

Joulinar, 2022-10-14:
ok it might be the `orjson` wheel. Too bad, pre-compiled wheels failed to be created on PiWheels as well. https://www.piwheels.org/project/orjson/ 

I guess due to mussing Rust

MichaIng, 2022-10-14:
Will be solved by: https://github.com/piwheels/piwheels/pull/322

Looks like I need to fork and push the fix here as well to get this forward 😄.

Joulinar, 2022-10-14:
Yes PyPi is shipping ARMv7. 

@MichaIng do you know if PiWheels is going to built ARMv6 wheels for that?

MichaIng, 2022-10-14:
There is an issue about it open and it fails for the same reason why synapse-matrix fails (too old Rust compiler on the Debian Buster repo used for the builds), so will be solved by the same PR linked above.

MichaIng, 2022-10-26:
Synapse v1.70.0 has been released: https://github.com/matrix-org/synapse/releases/tag/v1.70.0
- aarch64 wheel for Python 3.7 and up is available: https://pypi.org/project/matrix-synapse/#files
- piwheels ARMv6/7 wheels are pending. after the changed Rust implementation in their CI, there are a lot of rebuilds queued, I guess, so it might take a while: https://piwheels.org/project/matrix-synapse/

Marking this as closed.

closed 2022-10-26 by MichaIng