MichaIng/DietPi #5807: v8 Image Firstrun Setup Fails when booting from USB.

booblla, 2022-10-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8, G_DIETPI_VERSION_SUB=9, G_DIETPI_VERSION_RC=2, G_GITBRANCH='master', G_GITOWNER='MichaIng'
- Distro version | bussleye 0
- Kernel version | Linux PiServer 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 3A
- SD card used | NA - ArgonOne M.2 SATA III Enclosure

#### Steps to reproduce
1. Flash image to USB drive.
2. Assign drive letter as Windows detects drive, but doesn't assign a drive letter.
3. Copy dietpi.txt to drive
4. Boot from USB drive and wait for error message.
5. Select Ok.
6. First run script partially runs and updates and installs some software.

#### Expected behaviour
First run should fully setup and install all software that is included in the dietpi.txt file.

#### Actual behaviour
It gets part the way through, produces an error and doesn't setup properly.

#### Extra details
I have tried to reproduce the issue with the DietPi_RPi-ARMv7-Bullseye image, but it works as expected when booting from USB.
The DietPi_RPi-ARMv8-Bullseye image works as expected when booting from SD Card.

[dietpi.txt](https://github.com/MichaIng/DietPi/files/9793155/dietpi.txt)
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/9793156/dietpi-firstboot.log)
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/9793157/dietpi-firstrun-setup.log)
[dietpi-ramlog.log](https://github.com/MichaIng/DietPi/files/9793158/dietpi-ramlog.log)
[dietpi-update.log](https://github.com/MichaIng/DietPi/files/9793159/dietpi-update.log)
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/9793160/fs_partition_resize.log)
[dietpi.txt](https://github.com/MichaIng/DietPi/files/9793164/dietpi.txt)

Joulinar, 2022-10-16:
Can you be more specific on what exactly the issue is? At least provide the error message. At least first run setup seems to complete correctly. DietPi Update is applied, and you can select software, which got installed afterwards.

MichaIng, 2022-10-16:
Ah, the automated first run install failed, leading to an interactive repetition which succeeded (since only the SSH server was switched, not the list of additional software installed). Jep, would be helpful to know which software install failed with which error message. At least there is none in `dietpi.txt` which is currently known to fail on aarch64.

booblla, 2022-10-16:
> Can you be more specific on what exactly the issue is? At least provide the error message. At least first run setup seems to complete correctly. DietPi Update is applied, and you can select software, which got installed afterwards.

[image]

Just before this error displays, a bunch of red (I'm assuming) error messages flash up for half a second. But I'm guessing that they are in the logs.

Raspberry Pi's and Linux are just a hobby for me so I can teach myself a few new things, so my explanations/examples etc. may be vague at times.

Joulinar, 2022-10-16:
If you are connected via SSH, you should be able to scroll up a bit to check for the error message.

booblla, 2022-10-16:
I'm just running the setup again to replicate everything again and after logging in after the OS update and reboot, I get these screens that I obviously never get when I do an SD Card only install. No idea if they are helpful or not, but here they are.

[image]
[image]
[image]
[image]
[image]

Joulinar, 2022-10-17:
these are the normal install screens you get on every fresh installation. They are part of the normal setup and expected. What we need is an error message.

MichaIng, 2022-10-22:
@booblla 
Please repeat the steps. On the DietPi-Software menu, select `Browse Software` and select all the software titles that you added to `dietpi.txt` for automatic installation. Then hit `Install`. That way we should see which software title fails any why.

closed 2022-11-23 by MichaIng