MichaIng/DietPi #5808: dietpi-update   : Run now to update DietPi from v8.1.2 to v8.9.2 makes device unbootable

ajssoliveira, 2022-10-16:
### Creating a bug report/issue

#### Required Information
8.1.2
- DietPi version | `cat /boot/dietpi/.version`
- G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
G_LIVE_PATCH_STATUS[1]='not applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- bullseye

- Kernel version | `uname -a`
- Linux phonetika 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Odroid C4/HC4 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
- Full power ICP30-120-2000 12V 2.0A
- SD card used | (EG: SanDisk ultra)
- Adata 16GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...dietpi-update
2. ...update completes. When you reboot, the device will not boot
3. Fresh installation of 8.9.2 succeeds

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2022-10-16:
Does the device boot on power cycle?

MichaIng, 2022-10-16:
This is an older C4 image with Meveric's vendor kernel and bootloader. There should have been a kernel upgrade to v4.9.277 applied + a U-Boot upgrade, which is flashed automatically in this case. It is from March, so if there was a general issue, I'd have expected a fix until now. But since also Meveric ships mainline kernel images + packages already, it's probably not tested much anymore.

Do you still have the old SD card and can do some debugging on it, plugging it into another Linux system? Else I'd probably not invest much time, since we use Armbian's mainline kernel + bootloader for quite a while now.

ajssoliveira, 2022-10-17:
Greetings

Indeed I have 2 such cards Shall I put the image of one of the cards on 
my web server so you may download it? Or shall I conduct the 
verification on an Ubuntu jammy box?

Please advise

All the best

Antonio J. de Oliveira
***@***.***

Joulinar, 2022-10-17:
If you have a spare SD card, you could try our current image available on our download page.

ajssoliveira, 2022-10-17:
Thanks. I did tried to download 8.9.2, it installed flawlessly on 2 
different cards. The upgrade behaviour was the same on 2 different 
upgrades on two Odroid C4. Alles kaput, no boot... I had 11 more to 
upgrade and I used the emergency breaks, no upgrade. The 2 boxes were 
with me, the othere ones were remotely located performing live duties, 
no way :-)

All the best

Antonio

Joulinar, 2022-10-17:
you mean the new installation is working on reboot but the currently running once not (as they require an upgrade)?

ajssoliveira, 2022-10-17:
I will try to explain:

I had 13 Odroid C4 for possible upgrade. Before doing 500km (site is 
high security and no remote access is still allowed) I decided to try 
the upgrade on 2 boxes I had with me (my probes). Both upgrades failed, 
I installed the new version on both probe boxes with success and no 
issues at all. So I decided not to upgrade the remote ones.
So, I confirm that the upgrade cannot be done and the fresh install 
works as a breeze.

Shall I put the upgraded card image on my server for you to look at?

All the best

Antonio

MichaIng, 2023-02-12:
Looks like we missed to reply here.

> Shall I put the upgraded card image on my server for you to look at?

If you still have them, that would be a good idea, at least to check whether all boot relevant files are there. If you have persistent system logs enabled, even better to check them.

Btw as far as I understood, since those boxes are remote, all we know is that network didn't came up, right?

In the meantime there is another kernel upgrade available to [v4.9.312](https://oph.mdrjr.net/meveric/pool/c4/l/linux-4.9.312-arm64/). No guarantee that it will work better, probably the kernel upgrade isn't even the issue, but if one C4 down at some point in time is okay, could be tested.

MichaIng, 2023-03-18:
I'll mark this as closed. Please reopen if the issue persists.

closed 2023-03-18 by MichaIng