MichaIng/DietPi #5810: Cloudshell doesn't find PiHole anymore

oradke, 2022-10-16:
Cloudshell relies on /run/pihole-FTL.port to get the current status. However, /run/pihole-FTL.port has been deprecated:

```
PORTFILE=/run/pihole-FTL.port[¶](https://docs.pi-hole.net/ftldns/configfile/#file_PORTFILE)
This option is deprecated as FTL does not write any port file anymore. Please parse pihole-FTL.conf if you need to check if a custom API port is set.
```
Source: https://docs.pi-hole.net/ftldns/configfile/

Joulinar, 2022-10-16:
Many thanks for reporting it. We will have a look.

MichaIng, 2022-10-16:
Ah right, we need to read the port from the config file now. Will be fixed with upcoming release.

oradke, 2022-10-16:
Cool, thanks.

oradke, 2022-10-17:
FYI, there no port listed in the config file by default! Might need to be activated first.

MichaIng, 2022-10-21:
Fixed with #5822 for release tomorrow.

closed 2022-10-21 by MichaIng