MichaIng/DietPi #5815: Bug occuring during install script of 3rd party debian based image system. DietPi >ONTO> Libre AML-S905X-CC i.e. Rasp Pi 3

zypltk, 2022-10-17:
#### Details:
- Date           | Mon Oct 17 12:56:26 EDT 2022
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=)
- Kernel version | `Linux DESKTOP-IRAKV5O 5.15.68.1-microsoft-standard-WSL2 #1 SMP Mon Sep 19 19:14:52 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
- Distro         |  (ID=)
- Command        | `7zr a -m0=lzma2 -mx=9 -spf -x!/var/log/journal/* .7z CMD_OUT.txt G_EXEC_ERROR_REPORT /var/log /boot/config.txt /boot/cmdline.txt /boot/boot.ini /boot/uenv.txt /boot/uEnv.txt /boot/armbianEnv.txt /boot/boot.cmd /boot/dietpi.txt /boot/dietpi /var/lib/dietpi /var/tmp/dietpi /run/dietpi /etc/bash.bashrc /etc/bashrc.d /root/.bashrc /etc/profile /etc/profile.d /root/.profile /etc/rc.local /etc/X11/xorg.conf /etc/X11/xorg.conf.d /etc/asound.conf /etc/network/interfaces /etc/network/interfaces.d /etc/wpa_supplicant/wpa_supplicant.conf /etc/fstab /etc/sysctl.conf /etc/sysctl.d /etc/init.d /etc/systemd/system /lib/systemd/system /usr/local/lib/systemd/system /etc/apt/sources.list /etc/apt/sources.list.d /etc/apt/preferences /etc/apt/preferences.d /etc/apt/apt.conf.d`
- Exit code      | 127
- Software title | DietPi-BugReport
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 858: 7zr: command not found
```

Joulinar, 2022-10-17:
What exactly you are trying to do? You are running our installer? Basically you are missing the 7zip executable

```
apt install p7zip
```

MichaIng, 2022-10-17:
Likely an error-handler before required APT packages are installed.

closed 2022-10-17 by zypltk