MichaIng/DietPi #5816: Bug occuring during install script of 3rd party debian based image system. DietPi >ONTO> Libre AML-S905X-CC i.e. Rasp Pi 3

zypltk, 2022-10-17:
#### Details:
- Date           | Mon Oct 17 17:23:06 UTC 2022
- Bug report     |
- DietPi version | v8.9.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=)
- Kernel version | `Linux raspberrypi 5.19.15-04011-g49dca1c53e48 #1 SMP PREEMPT Fri Oct 14 00:20:35 UTC 2022 aarch64 GNU/Linux`
- Distro         |  (ID=)
- Command        | `sed -i /^[[:blank:]]*start_x=/c\#start_x=1 /boot/config.txt`
- Exit code      | 4
- Software title | DietPi-Set_hardware
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
sed: couldn't open temporary file /boot/sedQlUrOy: Read-only file system
```

Joulinar, 2022-10-17:
Quite simple, your boot file system is not writable.

zypltk, 2022-10-17:
How do I boot the system writable? This has been a common issue I have been having with any OS on the device. Is it a grub option during boot or?

Joulinar, 2022-10-17:
Many you can share some more information on what hardware you are using, which image aso.

MichaIng, 2022-10-17:
Before running the installer, try:
```sh
mount -o remount,rw /boot
```
Not sure which kernel your custom OS uses, but note that for RPi (according to the hostname) the official RPi repository is added and the kernel + bootloader installed from there. So there will be nothing left of your custom image 😉. Of course the custom kernel can be reinstalled afterwards. Sadly Linux v5.15 has no yet reached the official firmware repository (hence no pre-compiled kernel image etc).

zypltk, 2022-10-17:
That is when I chose **Open subshell** and I "wget" the config.txt to the /boot/ directory and continued.

zypltk, 2022-10-17:
[This](https://www.diffchecker.com/qCmFD5QP) list of differences in the dietpi.txt file I used and the dietpi.txt file provided by the developers... may be the cause of some of my issues?

zypltk, 2022-10-17:
As well as this fact;
`#Modifcations to /boot/* files, will not be saved when powered on.
#Please ensure you edit from the DietPi-RamDisk location /DietPi/*`

I think I have been attempting to modify the files in the /boot/ directory when those will not be saved when rebooting. So I think I need to try to source the same missing files but place them in the /DietPi/ directory rather than the /boot/ directory

zypltk, 2022-10-17:
[This](https://github.com/Pieloaf/Debian-DietPi-Install-Script) is what I actually used for the the installation script...

I am now seeing [this](https://github.com/MichaIng/DietPi/issues/1285) and I think I am going to try it out and start from scratch.

Joulinar, 2022-10-17:
Maybe you should have a look into our official docs instead of using unknown 3rd party script https://dietpi.com/docs/hardware/#make-your-own-distribution

zypltk, 2022-10-18:
So I did indeed start over and I have ran into two (related) issues.

In pursuit of following your directions more accurately--

... I rebooted the Libre AML-S905X-CC device and then SSH'd to it from my Windows 11 system and my very first command was the one that [MichaIng](https://github.com/MichaIng) had suggested;
`mount -o remount,rw /boot`

I ran that, then punched in the installer script command found on your website's docs found [here](https://dietpi.com/docs/hardware/#script-execution);
`bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-installer')"`

At this point (a 'dpkg' command is finishing) and everything seems to be running smoothly.

#The list above of all previously cleaned up packages is found here...""
`"Removing linux-image-5.19.15-04011-g49dca1c53e48 (5.19.15-04011-g49dca1c53e48-1) ..."`

then suddenly, 💥BOOM❗ I get this; 

```
dpkg: error processing package linux-image-5.19.15-04011-g49dca1c53e48 (--remove):
 cannot remove '/boot/vmlinuz-5.19.15-04011-g49dca1c53e48': ❗Read-only file system❗
dpkg: too many errors, stopping
Errors were encountered while processing:
 linux-image-5.19.15-04011-g49dca1c53e48
Processing was halted because there were too many errors.
[  OK  ] DietPi-BugReport | mkdir /root/.ssh
/boot/dietpi/dietpi-bugreport: line 168: ((: < 7 : syntax error: operand expected (error token is "< 7 ")
[  OK  ] DietPi-BugReport | Added setting [ssh.dietpi.com]:29248 ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIJdEPlagpQ+RVHNOX3jkG1Bya7Oza1dAke8h8NszVW84 to end of file /root/.ssh/known_hosts
[FAILED] DietPi-BugReport | Sending bug report
 - Command: curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT .7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/
[FAILED] DietPi-BugReport | Sending bug report
 - Command: curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT .7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/
```

So there's a few things happening here.. but can we just address the damn read only mounted system lol. Like I said in the beginnning, I ran the `mount -o remount,rw /boot` command FIRST then executed the installer script.

How is the system going into a read only mount? I can't figure out this bug 🪲

zypltk, 2022-10-18:
I dropped down into a subshell to investigate and I ran this real quick to troubleshoot the network for any issues 
`sudo nano /etc/resolv.conf`

All set there. Then I ran `sudo apt autoremove` from the subshell myself, which gave this result;
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  linux-image-5.19.15-04011-g49dca1c53e48
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
1 not fully installed or removed.
After this operation, 114 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 67116 files and directories currently installed.)
Removing linux-image-5.19.15-04011-g49dca1c53e48 (5.19.15-04011-g49dca1c53e48-1) ...
dpkg: error processing package linux-image-5.19.15-04011-g49dca1c53e48 (--remove):
 cannot remove '/boot/vmlinuz-5.19.15-04011-g49dca1c53e48': Read-only file system
dpkg: too many errors, stopping
Errors were encountered while processing:
 linux-image-5.19.15-04011-g49dca1c53e48
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@raspberrypi:/tmp/DietPi-BugReport# sudo dpkg --configure -a
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for initramfs-tools (0.140) ...
W: /boot is ro mounted.
W: update-initramfs: Not updating /boot/initrd.img-5.19.16-04046-g062726298e8a
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for dictionaries-common (1.28.4) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...
Processing triggers for dbus (1.12.24-0+deb11u1) ...
Processing triggers for shared-mime-info (2.0-1) ...
Processing triggers for udev (247.3-7+deb11u1) ...
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1+deb11u1) ...
Processing triggers for sgml-base (1.30) ...
Processing triggers for mailcap (3.69) ...
Processing triggers for fontconfig (2.13.1-4.2) ...`

then I ran `mount -o remount,rw /boot` _once again_ and `sudo dpkg --configure -a` && `sudo apt update`

[...]

`Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.libre.computer/repo linux InRelease
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Hit:5 https://deb.debian.org/debian bullseye-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian bullseye InRelease
Fetched 48.4 kB in 4s (13.6 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.`

[...]

finally, I punched in the command that I saw was making the bug report fail
`curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT .7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/` once again, from the subshell.

This is where I remain presently;
`root@raspberrypi:/tmp/DietPi-BugReport# curl --key  --connect-timeout 8 --retry 1 --retry-delay 4 -sSvT .7z sftp://dietpi-survey:upload2dietpi@ssh.dietpi.com:29248/bugreport/
Warning: The file name argument '--connect-timeout' looks like a flag.
*   Trying 0.0.0.8:80...

* connect to 0.0.0.8 port 80 failed: Connection timed out
* Failed to connect to 8 port 80: Connection timed out
* Closing connection 0
curl: (28) Failed to connect to 8 port 80: Connection timed out
*   Trying 0.0.0.8:80...
* connect to 0.0.0.8 port 80 failed: Connection timed out
* Failed to connect to 8 port 80: Connection timed out
* Closing connection 1`

**THANK YOU VERY MUCH for you continued attention and assistance in these matters**
I appreciate ya! 🙏

Joulinar, 2022-10-18:
I would not call it a bug if your system it putting your file system into r/o mode. Have a look to kernel error messages. There needs to be a reason for this

`dmesg -l err,crit,alert,emerg`

zypltk, 2022-10-18:
ran that and got;

> [    2.309072] mmc0: error -22 whilst initialising SDIO card
[    3.445629] mmc0: Card stuck being busy! __mmc_poll_for_busy
[    3.527789] lima d00c0000.gpu: error -ENODEV: dev_pm_opp_set_regulators: no regulator (mali) found
[   66.369617] BTRFS error (device mmcblk1p2): bdev /dev/mmcblk1p2 errs: wr 0, rd 0, flush 0, corrupt 1, gen 0
[   66.369671] BTRFS error (device mmcblk1p2): bdev /dev/mmcblk1p2 errs: wr 0, rd 0, flush 0, corrupt 2, gen 0
[   66.370223] BTRFS error (device mmcblk1p2): bdev /dev/mmcblk1p2 errs: wr 0, rd 0, flush 0, corrupt 3, gen 0
[   66.370682] BTRFS error (device mmcblk1p2): bdev /dev/mmcblk1p2 errs: wr 0, rd 0, flush 0, corrupt 4, gen 0
[   66.371081] BTRFS error (device mmcblk1p2): bdev /dev/mmcblk1p2 errs: wr 0, rd 0, flush 0, corrupt 5, gen 0
[ 1696.086240] FAT-fs (mmcblk1p1): error, fat_get_cluster: invalid cluster chain (i_pos 2059339)
[ 1696.086278] FAT-fs (mmcblk1p1): Filesystem has been set read-only
[ 1696.086311] FAT-fs (mmcblk1p1): error, fat_free: invalid cluster chain (i_pos 2059339)

Joulinar, 2022-10-18:
Looks like data corruption.

zypltk, 2022-10-18:
> I would not call it a bug if your system it putting your file system into r/o mode. Have a look to kernel error messages. There needs to be a reason for this
> 
> `dmesg -l err,crit,alert,emerg`

Reason I used the term "bug" was due to the fact that the script itself chose "BugReport" to be the terminology used in this situation

MichaIng, 2022-10-18:
Ah, the root filesystem is Btrfs. Good to see that it is generally working. But yeah, filesystem errors, also in the FAT partition, which is in turn automatically mounted read-only to prevent further corruption.

zypltk, 2022-10-18:
> Looks like data corruption.

What does that mean? The SD card needs to be repaired and formatted, partitioned and re-flashed ??

Joulinar, 2022-10-18:
> script itself chose "BugReport"

This is our default term if something happen. We don't differentiate on the reason of the failure 😄

zypltk, 2022-10-18:
> > script itself chose "BugReport"
> 
> This is our default term if something happen. We don't differentiate on the reason of the failure 😄

Oh it's certainly user oriented lol but I just don't see the term appropriate off hand that would better replace it

zypltk, 2022-10-18:
😡🪱 lol

zypltk, 2022-10-18:
> Ah, the root filesystem is Btrfs. Good to see that it is generally working. But yeah, filesystem errors, also in the boot FAT filesystem, which is in turn automatically mounted read-only to prevent further corruption.

What would be your suggested route to take next?

Joulinar, 2022-10-18:
You are using a SD card? If yes try another one.

zypltk, 2022-10-18:
> @zypltk /me leaves for Walmart...

MichaIng, 2022-10-18:
Another thing you could try first is flashing a regular DietPi image and see whether this runs without errors. We cannot rule out that it's either the image you use or the way how DietPi handles Btrfs for rootfs (though we support Btrfs for non-root drives since a long time).

MichaIng, 2022-10-18:
`fsck`, at best running it from another system so you can interactively watch progress and repairs. But it is perfectly possible that it is dying hardware wise. SD cards are not really made for 24/7 usage as system drive in average, compared to USB sticks, eMMC, HDDs/SSDs.

zypltk, 2022-10-18:
> Another thing you could try first is flashing a regular DietPi image and see whether this runs without errors. We cannot rule out that it's either the image you use or the way how DietPi handles Btrfs for rootfs (though we support Btrfs for non-root drives since a long time).

When I attempt to correctly (following the Libre instructions) re-flash the SD drive using Win32DiskImager __OR__ attempting to repair it using `diskpart > list > select disk > clean > create partition primary > format FS=Fat` the device gets hung up at the "create parition primary" command and throws a timed out error message then gets stuck when I try to format it there after ...

**Sketchy right?** 

/me is _still_ leaving for Walmart...

zypltk, 2022-10-18:
Yea I know :/ I don't have the external hard drive I normally ran on my SBC systems...

👮‍♂️

...long story, haha...

zypltk, 2022-10-18:
It had its own power supply too... smh, D###IT