MichaIng/DietPi #5818: Unable to install Linux-Headers using Diet-Pi in VM

abishur, 2022-10-18:
### Creating a bug report/issue
I am running Diet-pi in a VM, I went to install the nvidia-grid drivers, but received an error message that I need to first install linux-headers.  Using apt install linux-headers returns "Package linux-headers has no installation candidate.

#### Required Information
- DietPi version | `8.9.2`
- Distro version | `bullseye`
- Kernel version | `5.10.0-18-amd64 #1 SMP Debian 5.10.140-1 (2022-09-02) x86_64 GNU/Linux`
- SBC model | `Virtual Machine (x86_64)`
- Power supply used | N/A VM running on server 120v powered
- SD card used | N/A VM Running on server

#### Additional Information (if applicable)
- Software title | nvidia-linux-grid-510_510.85.02_amd64,deb
- Was the software title installed freshly or updated/migrated? | install fresh from package
- Can this issue be replicated on a fresh installation of DietPi? yes, though you need a login for the licensing portal

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Using the Proxmox VM enter apt install linux-headers

#### Expected behaviour

- linux-headers package should install

#### Actual behaviour

- I get an error message saying that linux-headers is not available, but is referred to by another package

#### Extra details

- ...

MichaIng, 2022-10-18:
```sh
apt install linux-headers-amd64
```
🙂

abishur, 2022-10-18:
hmm... I did that and it's still complaining that linux-headers is not installed.  Did there used to be a dummy package that called itself "linux-headers"?

MichaIng, 2022-10-18:
What are you actually trying to install? Is `nvidia-linux-grid` a package name (not in Debian repo at least.

Ah, you might need to upgrade the Linux package itself as well, so the headers match:
```sh
apt install linux-image-amd64
```

closed 2022-11-23 by MichaIng