MichaIng/DietPi #5819: NanoPi R4S LAN interface lost on reboot, soft restart

timcoop3r, 2022-10-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.9.2
- Distro version | Bullseye
- Kernel version | 5.15.74-rockchip64
- SBC model | NanoPi R4S
- Power supply used | 5V 1.5A
- SD card used | Samsung Pro Endurance

#### Steps to reproduce

1. connect to device with Ethernet cable in WAN port
2. Cold boot device, eth0 (LAN) and eth1 (WAN) both present and working.
3. Perform ‘reboot’ command. 
4. Login to device, WAN port is now eth0, LAN port not detected

#### Expected behaviour

- Eth0 LAN and Eth1 WAN should be detected on every soft restart.

#### Actual behaviour

- Eth0 and eth1 only detected on cold starts. On soft restart only WAN is detected as eth0

#### Extra details
Issue is mentioned in FriendlyWRT release 2022-08-03 22.03-rc6
“Fixed the problem that the R4S/R4SE may not recognize the pcie device (lan port) after a soft reboot (small probability)”

timcoop3r, 2022-10-20:
After further reading have realised this is not a DietPi issue. So closing

closed 2022-10-20 by timcoop3r

MichaIng, 2022-10-20:
We use the Armbian kernel, not the vendor kernel used in FriendlyWRT. But it seems then both suffer from the same issue and Armbian might be able to implement the same fix?