MichaIng/DietPi #5821: Plex Media Server - Claiming Server not possible

faxesystem, 2022-10-21:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
bullseye 0
- Kernel version
Linux DietPi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
- SBC model
RPi 3 Model B (armv7l)
- Power supply used
5V 2.5A
- SD card used SanDisk ultra

#### Additional Information (if applicable)
- Software title
Plex Media Server
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Install Plex Media Server and claim the server after the installation in IPOFPI:32400/web
2. Do a fresh DietPi install and install Plex Media Server again go to IPOFPI:32400/web, login to Plex, no access message appears

#### Expected behaviour
Claiming the server with the Plex account during initial setup in web browser

#### Actual behaviour
After the login to the Plex Account I'm getting an error message that I don't have access to the server

I'm having a similar problem to https://github.com/MichaIng/DietPi/issues/5315. Installed Plex a few weeks ago and everything was working perfect. Had to do a fresh install on my Pi3B+ today and now Plex is not letting me claim the server. It is showing me the same screen as in the first post here after I log into my Plex Account. https://www.truenas.com/community/threads/plex-not-authorized-you-do-not-have-access-to-this-server.96858/ I once had this problem with a docker image and I had to enter a claim code into the config.

BUT I installed LXDE on my Pi as a last resort and opened up 127.0.0.1:32400/web and here I'm getting the expected configuration screen to add a new server to my Plex Online account. 

I than tried the option allowedNetworks="10.x.x.x/255.255.255.0" for my home network but this didn't help. 

It is very strange, that the very first installation a few months ago worked perfect and claiming it worked and now the second install on the same hardware and the same network does not (not counting the workaround of installing a GUI and using Firefox to claim under 127.0.0.1.). Also tried another fresh Diet install and it didn't work either. So something must be wrong here.

Joulinar, 2022-10-21:
It doesn't make sense to report the very same issue on forum and on GitHub. This will not speed up anything. It is just creating double effort on our site. Pls stick to one platform. Thx for understanding.

faxesystem, 2022-10-21:
Sorry, after I commented on the other bug report it was still shown as closed so I wasn't sure if it would become visible. I think I found the problem. The server has to be removed from the Plex Online Account before doing a new installation. This was never the case on my Unraid Docker but it seems to be for Dietpi. At least everything is working now.

closed 2022-10-21 by faxesystem