MichaIng/DietPi #5825: Error message when mounting new drive -

mernamer, 2022-10-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8, G_DIETPI_VERSION_SUB=9, G_DIETPI_VERSION_RC=2, G_GITBRANCH='master'
- Distro version | bullseye 0
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | DietPi-DeviceManager
- Was the software title installed freshly or updated/migrated? N/A
- Can this issue be replicated on a fresh installation of DietPi? Yes

- Bug report ID | 4b5b12e8-8485-4bfd-ac5c-59682a7100a0

#### Steps to reproduce

1. New drive
2. Formatted as Ext4 from Drive Manager (no error messages, appears to work), used whole drive option, not partitions, GPT partition table.
3. Attempt to mount via Drive Manager,

#### Expected behaviour

Drive mounts, is useable as normal.

#### Actual behaviour

Error message: mount: /mnt/piDrive: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error. Drive does not mount.

Sent as bug report, Reference code: 4b5b12e8-8485-4bfd-ac5c-59682a7100a0.

#### Extra details

Directory was created but empty before mount, doesn't appear to make a difference if I delete the directory and let Drive Manager create it. Also tried reformatting a few times, no change.

Cursory research on the error message suggests it could be the partition table wasn't created correctly, can't use mkfs.ext4 to create.

Also, full disclosure, I bought two of this type of drive, both are having the same issues with different versions of DietPi, I may have received bad drives, may need to return them?

Btw, to the team: Really appreciate what you've done here, never been inspired to enter bug reports for anything, was so used to having to scour the internet for an alternate solution, initially hated that DietPi takes away access to some of the tools I was used to using to fix stuff like this... **but I'm a convert. I use DietPi only at this point** and can't thank you all enough for making this happen.

MichaIng, 2022-10-23:
Many thanks for your report.

> initially hated DietPi takes away access to some tools I was used to using to fix stuff...

You can still use any tool you used before. The DietPi tools/scripts are optional wrappers for common CLI tools, `dd`, `parted` and `mkfs` in this case.

> Could be partition table not created correctly, but can't use mkfs.ext4 to create (per some cursory research).

`mkfs.ext4` is used to create the ext4 filesystem, not the partition. The partition is created before that via `parted`.

I just tested it, and in general it works fine. I'll have a look into your bug report.

Once question: Were those external drivers (2.5" I assume?) powered via dedicated PSU or via USB only?

mernamer, 2022-10-23:
>

mernamer, 2022-10-23:
> Many thanks for your report.
> 
> > initially hated DietPi takes away access to some tools I was used to using to fix stuff...
> 
> You can still use any tool you used before. The DietPi tools/scripts are optional wrappers for common CLI tools, `dd`, `parted` and `mkfs` in this case.
Yeah, I figured that out, thanks for reminding others reading that part. :)
> > Could be partition table not created correctly, but can't use mkfs.ext4 to create (per some cursory research).
> 
> `mkfs.ext4` is used to create the ext4 filesystem, not the partition. The partition is created before that via `parted`.
Got it, I only got that far on a few pages suggesting the partition table was an issue.
> I just tested it, and in general it works fine. I'll have a look into your bug report.
> 
> Once question: Were those external drivers (2.5" I assume?) powered via dedicated PSU or via USB only?
I tried PSU first, then tried plugging into USB only, only because they're thumb drives, not full external drives, thought it might have made a difference... it didn't.

mernamer, 2022-10-23:
> You can still use any tool you used before. The DietPi tools/scripts are optional wrappers for common CLI tools, `dd`, `parted` and `mkfs` in this case.
Thanks for this reminder to anyone reading that part, I figured that out at some point. :) 
> `mkfs.ext4` is used to create the ext4 filesystem, not the partition. The partition is created before that via `parted`.
Got it, I only got that far in the articles referencing the issue.
> One question: Were those external drivers (2.5" I assume?) powered via dedicated PSU or via USB only?
I was using PSU as a rule, tried once with just USB because they're thumb drives, not full external drives, and thought it might have made a difference, but it didn't.

MichaIng, 2022-10-23:
Your `dmesg` kernel log indeed shows physical drive access errors, a kernel backtrace and finally filesystem error when trying to sync the SCSI cache to disk. Check our:
```sh
dmesg -l emerg,alert,crit,err
```
> Got it, I only got that far in the articles referencing the issue.

No problem, common to mix up partitions and filesystems, I just wanted to clarify the terms, especially important when trying to diagnose false steps in this regards 😉. However, the steps are correct, but stuff is not synchronised to disk physically.

> because they're thumb drives

Ah okay, then it shouldn't make a difference.
__________
The USB port always returns error `-32`
```
hub 1-1.3:1.0: hub_ext_port_status failed (err = -32)
usb 1-1.3: Failed to suspend device, error -32
usb 1-1.3-port1: cannot reset (err = -32)
usb 1-1.3-port1: cannot disable (err = -32)
```
Later I see port 2 and `/dev/sdb`, did you have two USB drives attached at the same time? Otherwise it did probably detected the same drive a second time.

You did try all USB ports?
Do those drives work fine on other systems, e.g. when formatting there and trying to mount and use at the RPi?
And could you try when booting with the drive attached, instead of attaching it later?

mernamer, 2022-10-23:
Got a fix reformatting with ExFAT file system, thanks to a question above.

> Do those drives work fine on other systems[?]

I tried other RPi's, but this time tried Windows 10 device with Ext4 support installed, also could not read, reformatted to exFAT from the system format menu, tried again on the original DietPi installed device... and it worked!

> ...did you have two USB drives attached at the same time?

I did at one point, but unplugged the other to troubleshoot.

Ever have a drive just not support Ext4? I've never run into this on other thumb drives.

Joulinar, 2022-10-23:
Put the exFAT formatted drive back into DietPi and use Drive Manager to reformat it into ext4. How does it behave?

mernamer, 2022-10-24:
Success! Allowed a format to Ext4 from DriveManager and now working normally.

Thanks again @MichaIng and @Joulinar.

closed 2022-10-24 by mernamer

MichaIng, 2022-10-24:
Great that this worked. I wonder how it makes a difference, but obviously it does 😄.