MichaIng/DietPi #5826: [dietpi software] add other & dashboard section in categories

t3dium, 2022-10-23:
There's some software which can't currently be categorised properly in dietpi software, so an other section and id would be useful for contributes. Also a dashboard section for services like flame, homer, etc

MichaIng, 2022-10-24:
I'd put such into `System Stats & Management` (8) currently, but I agree it does not match so good for pure jump page/bookmark dashboards. Probably we can make an own category for them once we have 2 or 3 such applications.

t3dium, 2022-10-25:
> I'd put such into `System Stats & Management` (8) currently, but I agree it does not match so good for pure jump page/bookmark dashboards. Probably we can make an own category for them once we have 2 or 3 such applications.

flame and homarr could additionally be added as alternative dashboards

MichaIng, 2022-10-25:
Let's do not add new dashboards only to justify a new category, but add the new category if someone is explicitly requesting an additional dashboard/jump page 😉.

MichaIng, 2022-10-25:
> At the end we have our own dashboard

Jep, with jump page functionality planned, and, since it has a backend, it can be potentially configured from within the web interface 🙂.

MichaIng, 2026-11-09:
Closing this. As mentioned, new categories can be added on demand with new software titles, if there are overall enough which would fit well.

closed 2026-11-09 by MichaIng