MichaIng/DietPi #5830: Cannot had more than one No-Ip domains

champymarty, 2022-10-24:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version 
  buster 0

- Kernel version | `uname -a`
 Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
  RPi 4 Model B (aarch64)

- Power supply used 
  Standard raspberry pi 4 power supply

- SD card used  
 SanDisk ultra 64 Gb

#### Additional Information (if applicable)
- Software title
  dietpi-DDNS
- Was the software title installed freshly or updated/migrated?
  It comes with dietpi os
- Can this issue be replicated on a fresh installation of DietPi?
  Did not try
- Bug report ID 
  a8acf7cd-ed7b-488b-b36f-8df9df3e7050

#### Steps to reproduce

1. Add a DDNS domain for NoIp
2.  Try adding a second one after the first

#### Expected behaviour

- It should add bot domain

#### Actual behaviour

I get a error here: 
[image]

- ...

#### Extra details

- ...

Joulinar, 2022-10-24:
This is working as designed. Our script is not able to handle 2 domain atm.

champymarty, 2022-10-24:
oh, because when I do in domains, it says it can : "Please enter a comma-separated list of domains that shall point to this system: "

Joulinar, 2022-10-24:
ahhh sorry my bad. I was typing to fast, by bad. Issue is related to DNS as you are not able to resolve the NoIP update server address `ping dynupdate.noip.com`

can you check following

```
ping dynupdate.noip.com
cat /etc/resolv.conf
```

champymarty, 2022-10-24:
[image]
When I ping it I can access it

champymarty, 2022-10-24:
I have these DNS: 
[image]

ALso the other domain is getting updated, and I have no internet problems

MichaIng, 2022-10-24:
Generally adding two domains that way comma-separated should work: https://www.noip.com/de-DE/integrate/request
Can you simply retry? Probably there was a temporary network issue, since if `ping` is able to resolve, `curl` should be as well.

Btw, checking your bug report, you have `dhcpcd5` installed while using a static IP. If this is an old system, it is likely a remain of old Pi-hole dependencies, while nowadays, Pi-hole does not required `dhcpcd5` anymore, so you can remove it:
```sh
apt-mark auto dhcpcd5
apt autopurge
```
However, it is definitely disabled so not causing any issues in your case.

Joulinar, 2022-10-24:
btw, there is no need to create screen prints. You should be able to copy everything from SSH terminal directly. This is way more easier to handle 😃

champymarty, 2022-10-24:
I retried multiple time, at least 5 time. Always show the same result. I can try to remove dhcpcd5 put i think it was use with PiVPN, which I just reinstall

MichaIng, 2022-10-24:
Cam you try:
```sh
curl -I dynupdate.noip.com
```

champymarty, 2022-10-24:
I get :
root@DietPi:~# curl -I dynupdate.noip.com
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Mon, 24 Oct 2022 19:33:44 GMT
Content-Type: text/plain
Content-Length: 22
Connection: keep-alive
Location: http://www.noip.com
ETag: "62d50c84-16"

MichaIng, 2022-10-24:
Confusing, this works fine. Can you test:
```sh
curl -L -u username:password https://dynupdate.noip.com/nic/update?hostname=host1.ddns.net,host2.ddns.net
```
replacing credentials and hosts. This is the exact command `dietpi-ddns` runs, besides hiding connection progress.

champymarty, 2022-10-24:
i get this:
root@DietPi:~# curl -L -u username:password https://dynupdate.noip.com/nic/update?hostname=host1.ddns.net,host2.ddns.net
badauth

Joulinar, 2022-10-24:
> badauth

Did you adjust `username:password` with your own one?

champymarty, 2022-10-24:
it works ! It does:
nochg 24.201.119.96
nochg 24.201.119.96

But it does not work in dietpi-DDNS

Joulinar, 2022-10-24:
how does it behabe using just 1 domain instaed of 2?

champymarty, 2022-10-24:
yes the dimain get updated on no ip with one domain

MichaIng, 2022-10-26:
How is it possible that the same command works on console but does not work within the script 🤔. In `dietpi-ddns` is always results in this "Could not resolve host" error?

Can you test:
```sh
which curl
bash -c '. /boot/dietpi/func/dietpi-globals; G_INIT; curl -I dynupdate.noip.com'
bash -c '. /boot/dietpi/func/dietpi-globals; curl -I dynupdate.noip.com'
bash -c 'curl -I dynupdate.noip.com'
```

champymarty, 2022-10-27:
i get :
root@DietPi:~# which curl
/usr/bin/curl
root@DietPi:~# bash -c '. /boot/dietpi/func/dietpi-globals; G_INIT; curl -I dynupdate.noip.com'
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Thu, 27 Oct 2022 22:56:28 GMT
Content-Type: text/plain
Content-Length: 22
Connection: keep-alive
Location: http://www.noip.com
ETag: "62d50c84-16"

root@DietPi:~# bash -c '. /boot/dietpi/func/dietpi-globals; curl -I dynupdate.noip.com'
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Thu, 27 Oct 2022 22:56:34 GMT
Content-Type: text/plain
Content-Length: 22
Connection: keep-alive
Location: http://www.noip.com
ETag: "62d50c7b-16"

root@DietPi:~# bash -c 'curl -I dynupdate.noip.com'
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Thu, 27 Oct 2022 22:56:39 GMT
Content-Type: text/plain
Content-Length: 22
Connection: keep-alive
Location: http://www.noip.com
ETag: "62d50c84-16"

root@DietPi:~#

MichaIng, 2022-11-03:
So it's not the clean script or G_INIT environment or dietpi-globals. I've absolutely no idea how in `dietpi-ddns` the curl call can do anything different 🤔.

Probably to check every step:
```sh
bash -x /boot/dietpi/dietpi-ddns
```
trigger the issue, then check or paste the whole console output (will be much) for any possible reason.

champymarty, 2022-11-07:
I get : 
[ INFO ] DietPi-DDNS | Testing DDNS update...
+ unset -f Print
+ local result
++ curl -6or4 -sSfL -u raphaelstjean@yahoo.ca:**masked_pass** 'https://dynupdate.noip.com/nic/update?hostname=champymartyserver.ddns.net,champymartyserverpi.ddns.net'
+ result='curl: (6) Could not resolve host: dynupdate.noip.com'
+ G_DIETPI-NOTIFY 1 'DDNS update test failed, please check your input:\ncurl: (6) Could not resolve host: dynupdate.noip.com'
+ ainput_string=("$@")
+ local i ainput_string output_string grey green red reset yellow dietpi_green
+ [[ -t 0 ]]
+ output_string='\e[0m\r\e[J'
+ grey='\e[90m'
+ green='\e[32m'
+ red='\e[31m'
+ reset='\e[0m'
+ yellow='\e[33m'
+ dietpi_green='\e[38;5;154m'
+ [[ 1 != \-\2 ]]
+ [[ -w /tmp/dietpi-process.pid ]]
+ local 'bracket_l=\e[90m[\e[0m' 'bracket_r=\e[90m]\e[0m'
+ local 'ok=\e[90m[\e[0m\e[32m  OK  \e[90m]\e[0m ' 'failed=\e[90m[\e[0m\e[31mFAILED\e[90m]\e[0m '
+ ((  1 == -1  ))
+ ((  1 == -2  ))
+ ((  1 == 0  ))
+ ((  1 == 1  ))
+ output_string+='\e[90m[\e[0m\e[31mFAILED\e[90m]\e[0m '
+ ainput_string+=('\n')
+ Print 1
+ [[ 1 == 1 ]]
+ [[ -n DietPi-DDNS ]]
+ output_string+='\e[90mDietPi-DDNS |\e[0m '
+ (( i=1 ))
+ (( i<3 ))
+ output_string+='DDNS update test failed, please check your input:\ncurl: (6) Could not resolve host: dynupdate.noip.com'
+ (( i++ ))
+ (( i<3 ))
+ output_string+='\n'
+ (( i++ ))
+ (( i<3 ))
+ echo -ne '\e[0m\r\e[J\e[90m[\e[0m\e[31mFAILED\e[90m]\e[0m \e[90mDietPi-DDNS |\e[0m DDNS update test failed, please check your input:\ncurl: (6) Could not resolve host: dynupdate.noip.com\n\e[0m'
[FAILED] DietPi-DDNS | DDNS update test failed, please check your input:

MichaIng, 2022-11-08:
Found and fixed it: https://github.com/MichaIng/DietPi/commit/70ed79e

The `-6or4` argument, when choosing to not force any IP family, should have been unset before running `curl`. Unset was however `-4or6` due to number twist 😅. It wasn't recognised if the DDNS provider has an AAAA (IPv6) record, since `-6or4` are valid arguments, `-6` forcing IPv6, then `-or4` which writes the output to a file name `r4`. No-IP however has an A (IPv4) record only:
```
# getent ahosts dynupdate.noip.com
158.247.7.204   STREAM dynupdate.noip.com
158.247.7.204   DGRAM
158.247.7.204   RAW
```

closed 2022-11-12 by MichaIng

eikaramba, 2022-12-21:
super happy that i found this thread. i had the exact same problem and totally didn't understand why it wasn't working although i could ping the domain and everything. everyone else would have maybe just closed the ticket and blamed it on the user or something, but no michaelng keeps on delivering. just incredible! Sry just wanted to say how rare such dedication is nowadays.

TLDR: Same Problem, works now after 8.11 Release