MichaIng/DietPi #5831: Pi 3/4: How to disable Audio on HDMI and audio on Headphone but keep awareness for USB Audio?

DrCWO, 2022-10-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- After istalling ALSA I see ALSA Card 0 and Card 1 if I enter aplay -l by default.

#### Describe the solution you'd like:

- I want to disable both but keep ALSA active to recognize USB audio interfaces.

#### Describe alternatives you've considered:

- none

#### Additional context

- ...

MichaIng, 2022-10-25:
It should be possible to select `usb-dac` near the end of the list, even if there is no USB DAC connected currently. Onboard audio will then be disabled.

Joulinar, 2022-10-27:
@DrCWO does this answer your question or is there still something open?

DrCWO, 2022-10-27:
Anything is OK now thank you :-)

Joulinar, 2022-10-27:
ok I'm going to close the issue, Feel free to reopen if needed.

closed 2022-10-27 by Joulinar