MichaIng/DietPi #5832: Unable to relocate dietpi_userdata after external drive failure

MattChiswell, 2022-10-25:
System Information:

- DietPi version | 8.9.2
- Distro version | bullseye 1
- Kernel version | Linux 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | Official UK Pi 5.1V 3A
- SD card used | SanDisk Ultra 16GB

I have been using SanDisk USB drives on my RPi's as a backup drive for a long time, I also have the `userdata` setup to use these drives as well. Last week, one of these USB drives experienced an, as of yet, unknown failure and has been returned to SanDisk for investigation.

Whilst awaiting their conclusion, I had the need to replace the drive temporarily with another working device. Having done this, I launched `dietpi-drive_manager` and attempted to move the `userdata` location to my new device, I was unable to do this with both the old and new drives connected at the same time, because the old drive was not even detected by the OS anymore after failing. I now appear to be stuck in this loop, I am unable to connect the original drive that held the `userdata` because it has completely failed, I am unable to _move_ the folder to my new drive because the source doesn't exist.

I don't appear to be able to perform many administrative tasks (backups/updates etc) without being given the following error:

```
[FAILED] DietPi-Userdata validation 
DietPi was unable to verify the existence of the userdata directory (). 
Please ensure all previous external drives are connected and functional, before trying again.
Unable to continue, exiting.    
```

I can't say that any part of the system that is checking its location is bugged, because its literally doing exactly as it should, checking if the current set folder actually exists, the drive isn't connected, so it fails. I think the issue may lie with `dietpi-drive_manager` but once again, being that this is a very specific failure, I am unsure whether you would classify this as a bug. The root issue is that I can't seem to select a new location for the `userdata` folders when the original location is no longer available (and never will be due to hardware failure), this fails with:

```
[FAILED] DietPi-Set_userdata | Please provide a source () and target (/mnt/dietpi_userdata) directory for input.
[FAILED] DietPi-Set_userdata | Exited with error
```

Source being empty because the drive is not connected. Is there anyway to manually set the directory and circumvent the userdata validation just once? As soon as it recognises the new folder is present will it forget the old location?

Joulinar, 2022-10-25:
Have a look to `/mnt`. There should be a symbolic link to the old disk. Simply remove that link and create an empty new user data folder

MattChiswell, 2022-10-25:
Perfect, solved the issue.

Appreciate the immediate response!

closed 2022-10-25 by MattChiswell

MichaIng, 2022-10-25:
Hmm, shall we add an option when this error message appears to continue with an empty userdata dir? But it should be made clear that in case of installed software, databases etc which do use the userdata dir, their startup usually fails until either data is restored or they are reinstalled to start over.

Joulinar, 2022-10-25:
I would not do this. Users need to be alarmed if the external drive is missing. If we would offer such an option to create an empty user data directory, users could just accept without reading the text box with care or being aware on the consequence.

MichaIng, 2022-10-25:
You are right.