MichaIng/DietPi #5834: OpenCV install failing on dependency

AkaMrBill, 2022-10-26:
I have been trying to install OpenCV 4.4 or 4.5 a few times. Each time I attempt to install the pre-requisites, the process fails when "sudo apt-get install libgtk2.0-dev" is run. The error is "The following packages have unmet dependencies:
 libgtk2.0-dev : Depends: gir1.2-gtk-2.0 (= 2.24.33-2) but 2.24.33-2+rpt1 is to be installed
E: Unable to correct problems, you have held broken packages"

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `Bullseye 0`
- Kernel version | `Linux DietPi 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)'
- Power supply used | 5v 4A
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | OpenCV 4.4
- Attempted fresh install
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
Followed step by step instructions at https://qengineering.eu/install-opencv-4.4-on-raspberry-pi-4.html
1. Attempted to install Dependencies
2. When "sudo apt-get install libgtk2.0-dev libcanberra-gtk*" is run, the following message is displayed.
"The following packages have unmet dependencies:
 libgtk2.0-dev : Depends: gir1.2-gtk-2.0 (= 2.24.33-2) but 2.24.33-2+rpt1 is to be installed
E: Unable to correct problems, you have held broken packages"

#### Expected behaviour
 libgtk2.0-dev and libcanberra-gtk* should be installed

#### Actual behaviour
The following packages have unmet dependencies:
 libgtk2.0-dev : Depends: gir1.2-gtk-2.0 (= 2.24.33-2) but 2.24.33-2+rpt1 is to be installed
E: Unable to correct problems, you have held broken packages.

#### Extra details
Sorry, this is outside my scope.
Maybe changing is pre-installed. Or maybe another dependancy is needed?
- ...

MichaIng, 2022-10-26:
You have LXDE installed, right? Can you test this:
```sh
cat << '_EOF_' > /etc/apt/preferences.d/dietpi-lxde
Package: openbox* obconf* libob* pcmanfm* libfm* gtk-* libgtk* libgail* gir1.2-gtk-* lx*
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
_EOF_
apt update
apt install libgtk2.0-dev
```

AkaMrBill, 2022-10-26:
I certainly can try, but I'm not familiar enough with what you are asking
to do this right.

What I need to know is, are these separate commands, or a group? What is
the goal of doing these?

Bill K.

MichaIng, 2022-10-26:
You can copy and paste them as is into your SSH console. The first 5 lines are one command with a so called here document, the three lines before the stop word.

Basically it's changing the content of the existing `/etc/apt/preferences.d/dietpi-lxde` file. You can also do this manually, replace `gir1.2-gtk-3.0` with `gir1.2-gtk-*`.

AkaMrBill, 2022-10-26:
While waiting for your reply, I realized what your command did. So, I
edited the file manually.

That seems to have got me past the issue. I am currently proceeding with
the rest of the install!

Thanks,
Bill K.

Joulinar, 2022-10-26:
Well you are trying to install a package `libjasper-dev` that simply did not exist.

MichaIng, 2022-10-26:
Jep, not part of the Debian repo (and neither present in RPi repo or Raspbian): https://packages.debian.org/search?keywords=libjasper

It seems to have been last shipped with Debian Jessie: http://raspbian.raspberrypi.org/raspbian/pool/main/j/jasper/

The guide is not that good, it also contains a step to install Python 2 modules and related warning that this isn't available on Bullseye. Later Python 3.7 environment is setup, while Bullseye ships Python 3.9. And it pulls an old OpenCV version, 4.6.0 has been released already.

Did you try this: https://lindevs.com/install-precompiled-opencv-on-raspberry-pi/
https://github.com/prepkg/opencv-raspberrypi/releases

Sadly there seems to be no official build instructions, clarifying the dependencies, so you would need to trial&error when following the guide, skipping `libjasper-dev` using Python 3.9, and in case the built process should tell you which development headers are missing.

AkaMrBill, 2022-10-26:
Yep, so by just removing it from the command,  I was able to finish the
install.

The guide I used did mention NOT installing Python 2 due to this being on a
Bullseye build, so those steps I also skipped.
As of now, it looks like I have success!

Should I close out the issue on github?

Bill K.

Joulinar, 2022-10-26:
I'm going to close the issue for now, feel free to reopen if needed

closed 2022-10-26 by Joulinar

MichaIng, 2022-10-27:
Okay great that it worked that way.