MichaIng/DietPi #5839: Error message while creating image file

DrCWO, 2022-10-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version |
bullseye 0

- Kernel version | 
Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

- SBC model | 
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 15W)

- SD card used | SanDisk ultra

#### Steps to reproduce

1. I have the image of Dietpi on a SD-Card inserted with a SD-Card-reader in USB of Pi
2. export SKIexport SKIP_ARCHIVE=1
3. bash -c "$(curl -sSfL 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.build/images/dietpi-imager')"

#### Expected behaviour

- runs without error message

#### Actual behaviour

- I see this error message after starting the command above and have no idea if it still produces a valid image file
```
[  OK  ] DietPi-Imager | cd /root
**bash: line 71: ((: == 1 : syntax error: operand expected (error token is "== 1 ")**
[ INFO ] DietPi-Imager | Checking for required APT packages: parted fdisk zerofree p7zip
[ INFO ] DietPi-Imager | Unmounting all filesystems below selected /dev/sda ...
```

#### Extra details

I think it's time to thank you for DietPi and the great support you offer. I reported some issues in the past and I am very happy of your quick response time and also for answering all my sometimes silly questions. THANK YOU 👍

DrCWO, 2022-10-27:
Despite this error message I kept it running but after a short time I saw this in the mask and gave up...

```
 - Command: fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt                                                     │
        │  - Exit code: 1                                                                                                      │
        │  - DietPi version: v8.10.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                  │
        │  - Image creator: DietPi Core Team                                                                                   │
        │  - Pre-image: from scratch                                                                                           │
        │  - Error log:       
```

DrCWO, 2022-10-27:
Just to let you know: Same with
`https://raw.githubusercontent.com/MichaIng/DietPi/master/.build/images/dietpi-imager`

DrCWO, 2022-10-27:
One more info. If I do `export SHRINK_ONLY=0` before starting the script I don't get the syntax error.

**But I still get the error message of fstrim 👎**

Joulinar, 2022-10-27:
Can you scroll up on the SSH session and check for the real error message please

DrCWO, 2022-10-27:
[ INFO ] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt, please wait...
[FAILED] DietPi-Imager | fstrim -v --quiet-unsupported /tmp/DietPi-Imager_mnt

Joulinar, 2022-10-27:
Number of partitions depends on SBC model. For Raspberry Pi it's standard to have 2 partition. BootFS located on vfat and RootFS on ext4. On a NanoPi R5S, we actually have up to 8 partition because of the FriendlyElec base image 🙄

MichaIng, 2022-10-27:
I'm not sure why on some drives `fstrim` fails, even with `--quiet-unsupported`, but I assured now that this failure does not halt the script, but it will just continue: https://github.com/MichaIng/DietPi/commit/8e11fc4

DrCWO, 2022-10-27:
My script stopped exactly at this point and I fell back to the dialog screen that I aborted...

DrCWO, 2022-10-27:
I retested and you are right, it continues running, sorry for my wrong comment. 
When I saw the error I stopped it as I did not know if the created image will be OK after the error message...

Joulinar, 2022-10-27:
> Will it be OK even with the reported error?

Yes, should be ok

MichaIng, 2022-10-28:
Yes, the error is still shown for transparency, just not halting the script.

closed 2022-11-08 by MichaIng