MichaIng/DietPi #5840: Navidrome update overwrites settings

pedrom34, 2022-10-27:
#### Required Information
- DietPi version | 8.10.2
- Distro version | bullseye
- Kernel version | 5.15.74-rockship64
- SBC model | Rock64

#### Additional Information (if applicable)
- Software title | Navidrome
- Was the software title installed freshly or updated/migrated? yes
- Can this issue be replicated on a fresh installation of DietPi? I do not know

#### Steps to reproduce

1. Update Navidrome by dietpi-software reinstall 204

#### Expected behaviour

- Keep navridrome.toml untouched

#### Actual behaviour

- Settings in dietpi_userdata/navidrome/navidrome.toml are overwritten.

Joulinar, 2022-10-27:
Many thanks for report. 

Indeed we recreate `/mnt/dietpi_userdata/navidrome/navidrome.toml` on reinstall. Something we need to check brefore reinstall.

MichaIng, 2022-10-27:
Solved for next release.

pedrom34, 2022-10-27:
thank you very much for your reactivity 😄