MichaIng/DietPi #5842: WebIOPi Installation Error

lutfor-diu, 2022-10-28:
#### Details:
- Date           | Fri Oct 28 10:41:14 +06 2022
- DietPi version | v8.10.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux DietPi 5.15.61+ #1579 Fri Aug 26 11:08:59 BST 2022 armv6l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y --allow-change-held-packages install g++ make automake`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libtirpc-dev : Depends: libtirpc3 (= 1.3.1-1) but 1.3.1-1+deb11u1 is to be installed
E: Unable to correct problems, you have held broken packages.

Joulinar, 2022-10-28:
Did you have installed something like a desktop in parallel?

lutfor-diu, 2022-10-28:
> Did you have installed something like a desktop in parallel?

No

lutfor-diu, 2022-10-28:
DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | Build-Essential will be installed
[ INFO ] DietPi-Software | Python 3 RPi.GPIO will be installed
[  OK  ] DietPi-Software | Checking IPv4 network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : qbittorrent
[  OK  ] DietPi-Services | unmask : filebrowser
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : filebrowser
[  OK  ] DietPi-Services | stop : qbittorrent
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 http://mirror.nus.edu.sg/raspbian/raspbian bullseye InRelease
Get:2 https://pkgs.tailscale.com/stable/debian bullseye InRelease
Get:3 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:4 https://archive.raspberrypi.org/debian bullseye/main armhf Packages [307 kB]
Fetched 336 kB in 15s (22.0 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Build-Essential: GNU C/C++ compiler, development libraries and headers

[ INFO ] DietPi-Software | APT install for: g++ make automake, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libtirpc-dev : Depends: libtirpc3 (= 1.3.1-1) but 1.3.1-1+deb11u1 is to be installed
E: Unable to correct problems, you have held broken packages.
[FAILED] DietPi-Software | APT install for: g++ make automake
 - Command: apt-get -y --allow-change-held-packages install g++ make automake
RPi B (armv6l) | IP: 192.168.31.2 | Use up/down buttons to scroll text─────────┐
│ APT install for: g++ make automake
│  - Command: apt-get -y --allow-change-held-packages install g++ make
│ automake
│  - Exit code: 100
│  - DietPi version: v8.10.2 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 |
│ DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: from scratch
│  - Error log:
│ Reading package lists...
│ Building dependency tree...
│ Reading state information...
│ Some packages could not be installed. This may mean that you have
│ requested an impossible situation or if you are using the unstable
│ distribution that some required packages have not yet been created

lutfor-diu, 2022-10-28:
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- **Reference code: b53b5f4a-1d91-43ba-9310-ff59c74d13fe**

Joulinar, 2022-10-28:
Can you share following

```
apt show libtirpc3
```

lutfor-diu, 2022-10-28:
> Can you share following
> 
> ```
> apt show libtirpc3
> ```

root@DietPi:~# apt show libtirpc3
Package: libtirpc3
Version: 1.3.1-1+deb11u1
Status: install ok installed
Priority: optional
Section: libs
Source: libtirpc
Maintainer: Josue Ortega <josue@debian.org>
Installed-Size: 201 kB
Depends: libc6 (>= 2.28), libgssapi-krb5-2 (>= 1.17), libtirpc-common (>= 1.3.1-1+deb11u1)
Breaks: nfs-common (<< 1:1.2.8-7), nfs-kernel-server (<< 1:1.2.8-7)
Homepage: http://sourceforge.net/projects/libtirpc
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: transport-independent RPC library
 This package contains a port of Sun's transport-independent RPC library to
 Linux. The library is intended as a replacement for the RPC code in the GNU C
 library, providing among others support for RPC (and in turn, NFS) over IPv6.

N: There is 1 additional record. Please use the '-a' switch to see it

MichaIng, 2022-10-28:
Based on the Raspbian online package list, both versions should match, including the dependency: http://raspbian.raspberrypi.org/raspbian/dists/bullseye/main/binary-armhf/

Please try:
```sh
/boot/dietpi/func/dietpi-set_software apt clean
apt update
apt install g++ make automake
```

lutfor-diu, 2022-10-29:
> Based on the Raspbian online package list, both versions should match, including the dependency: http://raspbian.raspberrypi.org/raspbian/dists/bullseye/main/binary-armhf/
> 
> Please try:
> 
> ```shell
> /boot/dietpi/func/dietpi-set_software apt clean
> apt update
> apt install g++ make automake
> ```

It works, after changing the main respiratory region to default one. Thank you so much

Joulinar, 2022-10-29:
ok going to close this one, feel free to reopen if needed.

closed 2022-10-29 by Joulinar

Korpip, 2022-11-22:
Installing this on both a pi 4 and a pi 3b+ yeilds similar results.
Nov 21 19:25:19 DoBox systemd[1]: webiopi.service: Main process exited, code=exited, status=1/FAILURE
Nov 21 19:25:19 DoBox systemd[1]: webiopi.service: Failed with result 'exit-code'.

Joulinar, 2022-11-22:
Pls create a new issue as yours might be unrelated. Thx